esphome/issues #4885: ESP32 mit MCP2515 extended_ID wird nicht empfangen

Grisu74, 2023-09-16:
### The problem

Hallo,

esphome mit esp32 und MCP2515 kann keine extended_id empfangen
Senden geht und Standard_id funktioniert auch.

Wo ist das Problem?

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.2

### What platform are you using?

ESP32

### Board

ESP32 Dev Kit C V2

ssieb, 2023-09-17:
You didn't include any info.  What have you tried?  What isn't working?

Grisu74, 2023-09-17:
Hello,

here is my code.

I want to control a charger via CanBus.
I use an ESP32 with MCP2515 and the MCP library. This is connected to CanHack. Communication with the charger works. If I integrate ESP32 and MCP2515 into the ESPHOME bus system, everything works except for receiving extended IDs. Sending works!! StandardID works.

`captive_portal:

spi:
  id: McpSpi
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19
 

switch:
  - platform: template
    name : "CanBus" 
    id: "Switch_CanBus"
    turn_on_action: 
      then:
        - canbus.send: 
               data: [0X20, 0x00]
               canbus_id: my_mcp2515
               use_extended_id: true
               can_id: 0x000C0103
               #remote_transmission_request: True
              
        - delay: 
            seconds: 2
    
    turn_off_action: 
      then:
        - canbus.send: 
               data: [0X30, 0x00]
               canbus_id: my_mcp2515
               use_extended_id: true
               can_id: 0x000C0103
               #remote_transmission_request: True 
        - delay: 
            seconds: 2  
      

#time:
#  - platform: homeassistant
#    id: homeassistant_time
#    on_time:
#      - seconds: /1
#        then:
#          - canbus.send:
#               data: [0X20, 0x00]
#               canbus_id: my_mcp2515
#               use_extended_id: True
#               can_id: 0x000C0103
               #remote_transmission_request: True

  

canbus:
  - platform: mcp2515
    id: my_mcp2515
    spi_id: McpSpi
    cs_pin: GPIO5
    clock: 8MHZ
    bit_rate: 250KBPS 
    use_extended_id: True
    can_id: 0x000C0103 #Controller 
    
    on_frame: 
    - can_id: 0x000C0003
      #can_id_mask: 0x1fffffff   
      use_extended_id: true
      then: 
      - logger.log: "test"
      - lambda: |-
          std::string b(x.begin(), x.end());
          ESP_LOGD("can extended id 0x123", "%s", &b[0] );

    - can_id: 0x321
      #can_id_mask: 0x1fffffff   
      use_extended_id: False
      then: 
      - logger.log: "test"
      - lambda: |-
          std::string b(x.begin(), x.end());
          ESP_LOGD("can id 0x321", "%s", &b[0] );   `

SpenZerX, 2024-01-09:
LOOPBACK mode gives no echo on tx with extended id ...  Please fix.