esphome/issues #4886: SHT3x now enables heater by default

jkl1337, 2023-09-16:
### The problem

Issue #5161 seems to enable the sensor heater by default despite it's title: `cv.Optional(CONF_HEATER_ENABLED, default=True): cv.boolean`, and later `if (!this->write_command(heater_enabled_ ? SHT3XD_COMMAND_HEATER_ENABLE : SHT3XD_COMMAND_HEATER_DISABLE)) {`

### Which version of ESPHome has the issue?

2023.10.0-dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

jasonlaguidice, 2023-09-28:
I just updated to 2023.9 and with no other changes the temperature sensor in my AirGradient box using the same sensor reads 15 *F higher. This may be the cause for me too and I will test.

jasonlaguidice, 2023-09-28:
Returning back reporting the same thing - manually setting `heater_enabled: false` corrects the problem and normal sensor operation is restored.

closed 2023-09-29 by kbx81

alexruffell, 2023-10-12:
No wonder all my sensors were off!