esphome/issues #4888: Integer as variable

NEOhidra, 2023-09-17:
### The problem

Assigning an integer as a value to a variable leads to an error, at validation and the installation fails.

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

d1_mini

### Example YAML snippet

```yaml
## test-variable.yaml
esphome:
  name: test-variable

esp8266:
  board: d1_mini

<<: !include { file: sections/common.yaml, vars: { web_server_version: 1 } }
<<: !include sections/wifi.yaml

## sections/common.yaml
ota:
 password: !secret ota_password
 id: my_ota

web_server:
  port: 80
  version: ${web_server_version}
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.8.2
INFO Reading configuration /config/test-variable.yaml...
Failed config

web_server: [source /config/sections/common.yaml:11]
  port: 80
  
  Unknown value '1', did you mean '1'?.
  version: 1
```

ssieb, 2023-09-17:
Try adding this to your config:
```yaml
external_components:
  - source: github://pr#5405
    components: [ web_server ]
```

NEOhidra, 2023-09-17:
Seems to work.

closed 2023-09-20 by jesserockz