esphome/issues #4889: Deep Sleep: KEEP_AWAKE pin mode does not work properly

shifuoogway, 2023-09-18:
### The problem

[https://esphome.io/components/deep_sleep.html#deep-sleep-esp32-wakeup-pin-mode](url) describes "KEEP_AWAKE: Keep the ESP32 awake while the wakeup pin is still active. Or in other words: defer the activation of the deep sleep until the wakeup pin is no longer active."

It appears that the implementation rather than defer deep sleep actually restarts run_duration while pin is high causing unwarranted wifi activity and signal delay. 

### Which version of ESPHome has the issue?

2023.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32D

### Component causing the issue

N/A

### Example YAML snippet

```yaml
deep_sleep:
  wakeup_pin: 14
  wakeup_pin_mode: KEEP_AWAKE
  run_duration: 120s
  sleep_duration: 300s
```

### Anything in the logs that might be useful for us?

```txt
This happens at the end of each run_duration cycle, i.e. each 120s while the pin is held high 

[I][deep_sleep:116]: Beginning Deep Sleep
[I][deep_sleep:118]: Sleeping for 300000000us
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:262]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
```

shifuoogway, 2023-09-18:
The problem becomes greater as run_duration is reduced. A 30s run duration means that the unit is reawakened and wifi reconnected once every 30s

David6390, 2023-12-12:
I experienced the same behaviour.

Also as a note here, if `sleep_duration` is not defined, then `wakeup_pin_mode: KEEP_AWAKE` work as intended. 
I think it should also work the same way when `sleep_duration` is defined.