esphome/issues #4890: CL Clamp/ADS1115

ToViNi, 2023-09-18:
### The problem

Setting up 2 ADS1115 with 4 ct clamps sensors each, and using continuous_mode: on gives one sensor with wrong value.

With continuous_mode on, one random sensor will show a value on a sensor with no current.

Example.  4A on sensor A2 gives a readout on sensor A1, even when there is no current in the other tree sensors.

A1:  1.1 A (Wrong)
A2:  4.0 A
A3:  0.0 A
A4:  0.0 A

Restart the ESP, and the (low) value moves to another sensor.

This shows that there is something leftover from last sensor reading somewhere.

### Which version of ESPHome has the issue?

2023.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.2

### What platform are you using?

ESP32

### Board

dev

### Component causing the issue

ADS1115

### Example YAML snippet

```yaml
ads1115:
  - address: 0x48
    id: ads1115a
    continuous_mode: on

sensor:
  - platform: ads1115
    ads1115_id: ads1115a 
    multiplexer: A0_GND
    gain: 4.096
    id: a1
    accuracy_decimals: 3
    icon: "mdi:gauge"
    state_class: "measurement"    

  - platform: ct_clamp
    sensor: a1
    id: ca1
    name: Current A1
    accuracy_decimals: 3
    update_interval: 2s
    sample_duration: 0.2s

..... and more sensors like the one
```

ToViNi, 2023-09-19:
Debug gave me the answer.   To long sample duration or to short update intervals gives very unstable sample count.  One sensor gets 60 samples, another only 7 samples.   Testing shows that with 2 ADS1115 = 8 channels, update intervals must be minimum 5s for 0.25s sample duration.   This gave me logging like this line: Raw AC Value: 0.000A after 179 different samples (716 SPS) and the values are good. (I2S speed 800kHz)

ssieb, 2023-09-19:
I think the reason is that there is no scheduling to make sure that only one input is getting sampled at a time.  So what happens is that ct clamps on multiple channels can try to sample at the same time and it doesn't work.  For now, I suggest disabling the update interval and use an `interval:` to run updates for each channel in sequence with an appropriate delay between each one.

ToViNi, 2023-09-19:
> I think the reason is that there is no scheduling to make sure that only one input is getting sampled at a time. .......

You made my day!   Excellent idea.   The number of samples was very random.   Now, with interval/sequenced  updates, is VERY consistent:   

13:53:57 | [D] | [main:715] | Start measure sequence
13:53:57 | [D] | [ct_clamp:041] | 'Current A1' - Raw AC Value: 0.000A after 216 different samples (720 SPS)
13:53:57 | [D] | [ct_clamp:041] | 'Current A2' - Raw AC Value: 0.002A after 220 different samples (733 SPS)
13:53:58 | [D] | [ct_clamp:041] | 'Current A3' - Raw AC Value: 0.000A after 214 different samples (713 SPS)
13:53:58 | [D] | [ct_clamp:041] | 'Current A4' - Raw AC Value: 0.001A after 210 different samples (700 SPS)
13:53:58 | [D] | [ct_clamp:041] | 'Current C1' - Raw AC Value: 0.001A after 214 different samples (713 SPS)
13:53:59 | [D] | [ct_clamp:041] | 'Current C2' - Raw AC Value: 0.000A after 219 different samples (730 SPS)
13:53:59 | [D] | [ct_clamp:041] | 'Current C3' - Raw AC Value: 0.002A after 214 different samples (713 SPS)
13:53:59 | [D] | [ct_clamp:041] | 'Current C4' - Raw AC Value: 0.001A after 218 different samples (726 SPS)

ssieb, 2023-09-19:
Leave this issue open for scheduling support in the voltage sampler.

ssieb, 2023-09-19:
Can you try this with your original config?  Also have one of the channels without a ct_clamp to test if regular updates work too.
```yaml
external_components:
  - source: github://ssieb/esphome@vsampler
    components: [ voltage_sampler, ct_clamp, ads1115 ]
    refresh: 1min
```

ToViNi, 2023-09-20:
Your code works great.    Original components:

01:52:59	[D]	[ct_clamp:041]	
'Current A1' - Raw AC Value: 0.000A after 14 different samples (46 SPS)
01:52:59	[D]	[ct_clamp:041]	
'Current C1' - Raw AC Value: 0.001A after 25 different samples (83 SPS)
01:53:03	[D]	[ct_clamp:041]	
'Current C3' - Raw AC Value: 0.000A after 213 different samples (710 SPS)
01:53:03	[D]	[ct_clamp:041]	
'Current C4' - Raw AC Value: 0.001A after 223 different samples (743 SPS)
01:53:03	[D]	[ct_clamp:041]	
'Current A4' - Raw AC Value: 0.000A after 215 different samples (716 SPS)
01:53:04	[D]	[ct_clamp:041]	
'Current A3' - Raw AC Value: 0.000A after 79 different samples (263 SPS)
01:53:04	[D]	[ct_clamp:041]	
'Current A2' - Raw AC Value: 0.000A after 59 different samples (196 SPS)
01:53:04	[D]	[ct_clamp:041]	
'Current C2' - Raw AC Value: 0.001A after 112 different samples (373 SPS)
01:53:04	[D]	[ct_clamp:041]	

Your code:

01:55:54	[D]	[ct_clamp:049]	
'Current A2' - Raw AC Value: 0.000A after 222 different samples (740 SPS)
01:55:54	[D]	[ct_clamp:049]	
'Current C4' - Raw AC Value: 0.001A after 206 different samples (686 SPS)
01:55:55	[D]	[ct_clamp:049]	
'Current A4' - Raw AC Value: 0.001A after 224 different samples (746 SPS)
01:55:55	[D]	[ct_clamp:049]	
'Current C1' - Raw AC Value: 0.001A after 205 different samples (683 SPS)
01:55:56	[D]	[ct_clamp:049]	
'Current A3' - Raw AC Value: 0.000A after 218 different samples (726 SPS)
01:55:56	[D]	[ct_clamp:049]	
'Current C2' - Raw AC Value: 0.000A after 213 different samples (710 SPS)
01:55:56	[D]	[ct_clamp:049]	
'Current C3' - Raw AC Value: 0.001A after 229 different samples (763 SPS)
01:55:56	[D]	[ct_clamp:049]	
'Current A1' - Raw AC Value: 0.001A after 221 different samples (736 SPS)
    
Tried one and one channel.   All works great and do not affect any other channel. (at least not in this test).
Values used:  Interval 5s , 300mS samples.

Denicolomanuel, 2024-02-06:
Hello,
I used your code and it works great. i only have some noise time to time. but when i look at the esp log, i have some warning/informations : 
```
[13:22:07][D][ct_clamp:048]: 'Courant P1' - Raw AC Value: 0.003A after 458 different samples (458 SPS)
[13:22:07][D][sensor:093]: 'Courant P1': Sending state 0.08755 A with 2 decimals of accuracy
***[13:22:08][W][component:214]: Component ct_clamp.sensor took a long time for an operation (0.67 s).***
***[13:22:09][W][component:215]: Components should block for at most 20-30ms.***
[13:22:20][D][ct_clamp:048]: 'Courant P2' - Raw AC Value: 0.003A after 472 different samples (472 SPS)
[13:22:20][D][sensor:093]: 'Courant P2': Sending state 0.10274 A with 2 decimals of accuracy
***[13:22:21][W][component:214]: Component ct_clamp.sensor took a long time for an operation (0.61 s).***
***[13:22:22][W][component:215]: Components should block for at most 20-30ms.***
[13:22:23][D][ct_clamp:048]: 'Courant P3' - Raw AC Value: 0.003A after 478 different samples (478 SPS)
[13:22:24][D][sensor:093]: 'Courant P3': Sending state 0.09837 A with 2 decimals of accuracy
***[13:22:25][W][component:214]: Component ct_clamp.sensor took a long time for an operation (0.60 s).***
***[13:22:26][W][component:215]: Components should block for at most 20-30ms.***
[13:22:27][D][ct_clamp:048]: 'Courant P1' - Raw AC Value: 0.003A after 471 different samples (471 SPS)
[13:22:27][D][sensor:093]: 'Courant P1': Sending state 0.07020 A with 2 decimals of accuracy
***[13:22:28][W][component:214]: Component ct_clamp.sensor took a long time for an operation (0.67 s).***
***[13:22:29][W][component:215]: Components should block for at most 20-30ms.***
[13:22:29][D][ads1115:233]: 'adc_sensor3': Got Voltage=1.572625V
[13:22:30][D][sensor:093]: 'adc_sensor3': Sending state 1.57263 V with 3 decimals of accuracy
***[13:22:31][W][component:214]: Component ads1115.sensor took a long time for an operation (1.09 s).***
***[13:22:32][W][component:215]: Components should block for at most 20-30ms.***
```
i don't know if it's a problem or not and how to solve it.
any idea?

here is my code:

```
esphome:
  name: tableau-electrique-general
  friendly_name: Tableau Electrique général

esp8266:
  board: nodemcu

external_components:
  - source: github://ssieb/esphome@vsampler
    components: [ voltage_sampler, ct_clamp, ads1115 ]
    refresh: 1min

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: XXX

ota:
  password: XXX

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: XXX
    password: XXX

captive_portal:

# configuration du I2C
i2c:
  sda: 4
  scl: 5
  scan: True
  frequency: 200000

# configuration de l ADC
ads1115:
  address: 0x48
  continuous_mode: True

# configuration des capteurs de courant et pinces
sensor:
  # sensors source
  - platform: ads1115
    multiplexer: 'A1_GND'
    gain: 4.096
    name: "adc_sensor1"
    id: "adc_sensor1"
  - platform: ct_clamp
    sensor: adc_sensor1
    name: "Courant P1"
    sample_duration : 1s
    update_interval : 20s
    filters:
      - calibrate_linear:
          # Measured value of 0 maps to 0A
          - 0 -> 0
          # Known load: 4.0A
          # Value shown in logs: 0.1333A
          - 0.3 -> 8.3 

  - platform: ads1115
    multiplexer: 'A2_GND'
    gain: 4.096
    name: "adc_sensor2"
    id: "adc_sensor2"
  - platform: ct_clamp
    sensor: adc_sensor2
    name: "Courant P2"
    sample_duration : 1s
    update_interval : 20s
    filters:
      - calibrate_linear:
          # Measured value of 0 maps to 0A
          - 0 -> 0
          # Known load: 4.0A
          # Value shown in logs: 0.1333A
          - 0.27 -> 8.2

  - platform: ads1115
    multiplexer: 'A3_GND'
    gain: 4.096
    name: "adc_sensor3"
    id: "adc_sensor3"
  - platform: ct_clamp
    sensor: adc_sensor3
    name: "Courant P3"
    sample_duration : 1s
    update_interval : 20s
    filters:
      - calibrate_linear:
          # Measured value of 0 maps to 0A
          - 0 -> 0
          # Known load: 4.0A
          # Value shown in logs: 0.1333A
          - 0.27 -> 8.2
```
Have a nice day.

Manu

ssieb, 2024-02-06:
I don't know why it would be taking so long.  I'll try to take a look.

nobthenob, 2024-04-19:
> Can you try this with your original config? Also have one of the channels without a ct_clamp to test if regular updates work too.
> 
> ```yaml
> external_components:
>   - source: github://ssieb/esphome@vsampler
>     components: [ voltage_sampler, ct_clamp, ads1115 ]
>     refresh: 1min
> ```

Hello, I experienced the same kind of "random crosstalking" with 4 ct clamps connected to one ADS1115 and was happy when I found the issue described here.
I tried your code and after first tests, it seems to solve the problem,  channels are now independent. Thank you very much!

Can you can help me with the following questions:
Where can I find the doc on the voltage_sampler component? I am interested to understand how it works and what are limitations, in order to tweak my ct clamp parameters for fast update rates.
thx, nob

ssieb, 2024-04-19:
The `voltage_sampler` is an interface that adc devices can provide.  The `ct_clamp` component tries to read as fast as possible or at least every 1ms.

nobthenob, 2024-04-21:
Hello Samuel,
I have now monitored the behaviour of my 4 channel CT Clamp sensor a while and played around with the update_interval and sample_duration values. 
So far all 4 channel are working independently, as they should :)

Nevertheless, I still see that the number of samples (SPS) of the 4 channels differ very much, like what ToViNi described in the first place: Typically one channel shows significantly lower SPS values, leading to small data errors.
Interestingly, the channel that show this significanty lower SPS values is changing with every restart.
In *average* I get about 200 SPS per channel with a ESP8266 D1 mini.

Should I open a new issue on that to provide code and logs?
with best regards, nob

rwalker777, 2025-03-12:
This issue still exists in 2025.2.2... can we get this merged?  @ssieb Any reason why it's not merged in?

rwalker777, 2025-03-12:
> Hello Samuel, I have now monitored the behaviour of my 4 channel CT Clamp sensor a while and played around with the update_interval and sample_duration values. So far all 4 channel are working [...]

It only samples different values, so if it gets zero 300 times, it will show a low SPS.

jkoljo, 2025-03-29:
I am also struggling with the same issue. No fix found for now. The code above may have helped a bit but it might be placebo as well. Unfortunately CT Clamp + ADS1115 is not currently good enough to start replicating my proto setup in the electrical panel.