esphome/issues #4891: teleinfo: bad crc

alucryd, 2023-09-18:
### The problem

Using teleinfo floods the log with crc errors, even though data seems to be fine and my entities are correctly updated in HA. I haven't found a way to debug the offending tag as even debug logging doesn't show anything more verbose than `[20:11:42][E][teleinfo:038]: bad crc: got 40 except 89`. My Linky is in standard mode, I am using a Denky D4 as my board of choice. I tried both Arduino and ESP-IDF frameworks, both are affected.

Is there a way to display the whole TIC data so I can more easily debug the issue? Unfortunately I don't speak C++, I tried to add more logs, but my changes only managed to break logs altogether as pointers are like dark magic to me.

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.09.2

### What platform are you using?

ESP32

### Board

denky_d4

### Component causing the issue

teleinfo

### Example YAML snippet

```yaml
esphome:
  name: linky
  friendly_name: Linky

esp32:
  board: denky_d4

logger:
  baud_rate: 0
  level: INFO

api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

time:
  - platform: homeassistant
    timezone: "Europe/Paris"
    id: homeassistant_time

binary_sensor:
  - platform: status
    name: "Status"

button:
  - platform: restart
    name: "Restart"

uart:
  rx_pin: GPIO8
  baud_rate: 9600
  parity: EVEN
  data_bits: 7
  stop_bits: 1

light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    chipset: SK6812
    pin: GPIO14
    num_leds: 1
    rmt_channel: 0
    name: "Status LED"
    restore_mode: ALWAYS_OFF

teleinfo:
  update_interval: 5s
  historical_mode: false

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
  - platform: uptime
    name: "Uptime"
  - platform: teleinfo
    tag_name: "EAST"
    name: "Total Power"
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    icon: mdi:flash
    accuracy_decimals: 2
    filters:
      - multiply: 0.001
  - platform: teleinfo
    tag_name: "EASF01"
    name: "Total Power (Peak)"
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    icon: mdi:flash
    accuracy_decimals: 2
    filters:
      - multiply: 0.001
  - platform: teleinfo
    tag_name: "EASF02"
    name: "Total Power (Off-Peak)"
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    icon: mdi:flash
    accuracy_decimals: 2
    filters:
      - multiply: 0.001
  - platform: teleinfo
    tag_name: "IRMS1"
    name: "Effective Current"
    unit_of_measurement: A
    device_class: current
    state_class: measurement
    icon: mdi:current-ac
  - platform: teleinfo
    tag_name: "URMS1"
    name: "Effective Voltage"
    unit_of_measurement: V
    device_class: voltage
    state_class: measurement
    icon: mdi:flash-triangle
  - platform: teleinfo
    tag_name: "PREF"
    name: "Maximum Apparent Power (Reference)"
    state_class: measurement
    device_class: apparent_power
    unit_of_measurement: VA
    icon: mdi:information
    filters:
      - multiply: 1000
  - platform: teleinfo
    tag_name: "PCOUP"
    name: "Maximum Apparent Power (Current)"
    state_class: measurement
    device_class: apparent_power
    unit_of_measurement: VA
    icon: mdi:information
    filters:
      - multiply: 1000
  - platform: teleinfo
    tag_name: "SINSTS"
    name: "Apparent Power"
    unit_of_measurement: VA
    state_class: measurement
    device_class: apparent_power
    icon: mdi:gauge
  - platform: teleinfo
    tag_name: "SMAXSN"
    name: "Apparent Power (Peak)"
    unit_of_measurement: VA
    state_class: measurement
    device_class: apparent_power
    icon: mdi:gauge
  - platform: teleinfo
    tag_name: "CCASN"
    name: "True Power"
    unit_of_measurement: W
    state_class: measurement
    device_class: power
    icon: mdi:gauge
  - platform: teleinfo
    tag_name: "CCASN-1"
    name: "True Power (N-1)"
    unit_of_measurement: W
    state_class: measurement
    device_class: power
    icon: mdi:gauge
  - platform: teleinfo
    tag_name: "UMOY1"
    name: "Average Voltage"
    unit_of_measurement: V
    device_class: voltage
    state_class: measurement
    icon: mdi:flash-triangle
    
text_sensor:
  - platform: wifi_info
    ip_address:
      name: IP Address
    mac_address:
      name: Mac Address
  - platform: teleinfo
    tag_name: "PRM"
    name: "PRM"
    icon: mdi:information
```
```

### Anything in the logs that might be useful for us?

```txt
[20:11:34][I][app:102]: ESPHome version 2023.8.3 compiled on Sep 18 2023, 20:11:04
[20:11:36][E][teleinfo:038]: bad crc: got 46 except 64
[20:11:42][E][teleinfo:038]: bad crc: got 40 except 89
[20:11:46][E][teleinfo:038]: bad crc: got 57 except 89
[20:11:51][E][teleinfo:038]: bad crc: got 37 except 54
[20:11:57][E][teleinfo:038]: bad crc: got 40 except 89
[20:12:01][E][teleinfo:038]: bad crc: got 57 except 89
[20:12:06][E][teleinfo:038]: bad crc: got 37 except 93
[20:12:12][E][teleinfo:038]: bad crc: got 62 except 56
[20:12:16][E][teleinfo:038]: bad crc: got 38 except 66
[20:12:22][E][teleinfo:038]: bad crc: got 34 except 80
[20:12:27][E][teleinfo:038]: bad crc: got 65 except 59
[20:12:31][E][teleinfo:038]: bad crc: got 38 except 66
[20:12:37][E][teleinfo:038]: bad crc: got 34 except 80
[20:12:41][E][teleinfo:038]: bad crc: got 57 except 89
[20:12:46][E][teleinfo:038]: bad crc: got 60 except 76
```
```

### Additional information

Note that the pinout of the denky_d4 is bogus in esphome, had to patch `boards.py` to be able to use pin 8 for RX.

```
diff --git a/esphome/components/esp32/boards.py b/esphome/components/esp32/boards.py
index 61cb8cdc..bc2c466e 100644
--- a/esphome/components/esp32/boards.py
+++ b/esphome/components/esp32/boards.py
@@ -235,6 +235,10 @@ ESP32_BOARD_PINS = {
         "SDA": 5,
         "SS": 15,
     },
+    "denky_d4": {
+        "RX": 8,
+        "LED": 14
+    },
     "esp-wrover-kit": {},
     "esp32-devkitlipo": {},
     "esp32-evb": {
 ```

alucryd, 2023-09-18:
Managed to get a little further by editing `teleinfo.cpp`, in case of a bad crc I allowed the code to continue and print the offending tag and value, here's what I got:
```
[20:31:25][E][teleinfo:038]: bad crc: got 37 except 95
[20:31:25][E][teleinfo:144]: tag: EASD02
[20:31:25][E][teleinfo:162]: Invalid timestamp for tag 
[20:31:30][E][teleinfo:038]: bad crc: got 69 except 65
[20:31:30][E][teleinfo:144]: tag: EASD02
[20:31:30][E][teleinfo:162]: Invalid timestamp for tag 
[20:31:35][E][teleinfo:038]: bad crc: got 38 except 68
[20:31:35][E][teleinfo:144]: tag: EASD02
[20:31:35][E][teleinfo:173]: value: 00
[20:31:40][E][teleinfo:038]: bad crc: got 34 except 82
[20:31:40][E][teleinfo:144]: tag: EASD02
[20:31:40][E][teleinfo:173]: value: 000223970
[20:31:40][E][teleinfo:176]: Invalid value for tag EASD02
[20:31:45][E][teleinfo:038]: bad crc: got 72 except 68
[20:31:45][E][teleinfo:144]: tag: EASD02
[20:31:45][E][teleinfo:162]: Invalid timestamp for tag 
[20:31:50][E][teleinfo:038]: bad crc: got 38 except 68
[20:31:50][E][teleinfo:144]: tag: EASD02
[20:31:50][E][teleinfo:173]: value: 00
[20:31:55][E][teleinfo:038]: bad crc: got 34 except 82
[20:31:55][E][teleinfo:144]: tag: EASD02
[20:31:55][E][teleinfo:173]: value: 000223970
[20:31:55][E][teleinfo:176]: Invalid value for tag EASD02
[20:31:59][E][teleinfo:038]: bad crc: got 57 except 91
[20:31:59][E][teleinfo:144]: tag: EASD02
[20:31:59][E][teleinfo:173]: value: 0005957
[20:32:05][E][teleinfo:038]: bad crc: got 60 except 84
[20:32:05][E][teleinfo:144]: tag: EASD02
[20:32:05][E][teleinfo:173]: value: PAS DE          MESSAGE         
[20:32:10][E][teleinfo:038]: bad crc: got 34 except 82
[20:32:10][E][teleinfo:144]: tag: EASD02
[20:32:10][E][teleinfo:173]: value: 000223970
[20:32:10][E][teleinfo:176]: Invalid value for tag EASD02
[20:32:15][E][teleinfo:038]: bad crc: got 57 except 91
[20:32:15][E][teleinfo:144]: tag: EASD02
[20:32:15][E][teleinfo:173]: value: 0005957
[20:32:20][E][teleinfo:038]: bad crc: got 60 except 84
[20:32:20][E][teleinfo:144]: tag: EASD02
[20:32:20][E][teleinfo:173]: value: PAS DE          MESSAGE         
```

The offending tag is consistently `EASD02`, and values are all over the place for some reason, sometimes it even gets the value of the MSG1 tag.

Toniob, 2023-10-14:
Hi,

I've got the same issue. And we are not the only one : #4690 .

Toniob, 2023-10-16:
I’ve checked in home assistant database and sometimes, I’ve got wrong values too. I’ve got 1927 values in total, and here is the list of wrong values : 

```
+--------------+-------------------+
| state        | last_updated_ts   |
+--------------+-------------------+
| 0            | 1697305907.114415 |
| 0            | 1697306027.156993 |
| 0            | 1697306087.189611 |
| 0            | 1697306147.217716 |
| 0            |  1697320836.95614 |
| 0            | 1697338837.057877 |
| 22402736     | 1697355357.147739 |
| 0            | 1697389925.056315 |
| 131320283136 | 1697423585.919117 |
| 0            | 1697454305.314354 |
+--------------+-------------------+
```

If I check the previous state value of the 9th line, I’ve got 13132028. It seems some other value was appended to it. But the crc check was ok in some way.

alucryd, 2023-11-28:
I've got the v1.3.

Toniob, 2023-11-28:
I’ve got the 1.1 version. Did you try to change the trimmer resistor value ?

alucryd, 2023-11-29:
Yeah, v1.3 has a potentiometer, I tried the whole range. Below a certain value it's completely unusable, passed that value it works, except for this particular bug.

Toniob, 2024-01-20:
I finally got rid of the wrong values by adding a filter :
```
  - platform: teleinfo
    tag_name: "EASD02"
    name: "Heures Pleines"
    unit_of_measurement: "Wh"
    icon: mdi:gauge
    device_class: "energy"
    state_class: "total_increasing"
    filters:
     - lambda: |-
         float MAX_DIFFERENCE = 10000.0;
         static float last_value = NAN;
         if (isnan(last_value) || std::abs(x - last_value) < MAX_DIFFERENCE)
           return last_value = x;
         else
           return {};
```

Maybe not optimal but it works

roma6868, 2025-02-11:
I also have the same problem. 
In the beginning, I used an ESP8266 for 1 year and never had any issues with the teleinformation data. Then, due to a lack of GPIO pins, I switched to the ESP32-WROOM-32, and since then, I've been having this random CRC issue, which also distorts the statistics in Home Assistant

However, I used the same .yaml file in both cases (except, of course, for changing the GPIO pins).

alucryd, 2026-04-27:
> This one ? https://github.com/esphome/esphome/pull/3855
> Sure it s not hardware issue right? I m fine merging it if thats the case.

That's the one. It's affecting a variety of devices, including one that was designed specifically for use with teleinfo, so a hardware issue seems unlikely.