esphome/issues #4892: Blinking when and after transitioning

iBobik, 2023-09-18:
### The problem

My CT light blinks like on the video. A lot while transitioning, less after it. Mostly only brightness change.

Was not significant with PWM frequency about 8 kHz, very significant with 40 kHz.

With version 2023.8.3 it was more signifiant, also color temperature was changing in blinks.

### Which version of ESPHome has the issue?

2023.10.0-dev, Docker image e61ccdc9d9ee

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Shelly RGBW2, H801

### Component causing the issue

light, cwww, esp8266_pwm

### Example YAML snippet

```yaml
light:
  - platform: cwww
    id: light_1
    name: $human_name
    cold_white: output_blue
    warm_white: output_red
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2400 K
    constant_brightness: true
    restore_mode: ALWAYS_OFF
    default_transition_length: 100ms

output:
  - platform: esp8266_pwm
    pin: GPIO15
    frequency: 40000 Hz
    max_power: 0.8
    id: output_red

  - platform: esp8266_pwm
    pin: GPIO12
    frequency: 40000 Hz
    max_power: 0.8
    id: output_blue
```

### Anything in the logs that might be useful for us?

```txt
Video: https://www.dropbox.com/scl/fi/yeth1vpuvf16l1r7cg3jr/esphome-e61ccdc9d9ee.mov?rlkey=ctpt1w59n2iwiy25175dd7mlj&dl=0
```

### Additional information

Transitions are send from Adaptive lightning in HASS.

bkaufx, 2023-09-21:
40khz is pretty extreme for esp8266.  You probably need to keep it at 1000hz or less.

iBobik, 2023-09-21:
> 40khz is pretty extreme for esp8266. You probably need to keep it at 1000hz or less.

I read examples with even 100 kHz. Also Meanwell power sources does not recommend such slow PWM because it beeps.

iBobik, 2024-01-20:
Just updated to ESPHome version 2023.12.8 and issue is still there.20. 1.