esphome/issues #4893: ESP32-C3

git-peterk, 2023-09-18:
### The problem

I was trying to compile a very minimalistic firmware to test ESPHome with my new ESP32-C3 boards.
I've tried 4 different boards so far, but none of them are able to connect to my Wi-Fi network. They all give the exact same error as shown in the log: `Event: Disconnected ssid='SMITH_HA' bssid=[redacted] reason='Auth Expired'`
I tried to get the boards closer to the AP, using various methods to power the boards (PC's USB port, power bank, a wall power adapter) without any success.
I also tried setting the board to AP mode, that way I can connect to the board using my phone.

I have many ESP32-D0 based boards that work flawlessly, but they all use the Arduino framework, this was the 1st time I was trying to use ESP-IDF, as I don't have access to the UART pins on the devices, so I could only use the USB_SERIAL_JTAG port. I also tried compiling a firmware with the Arduino framework (although I can't see the device logs that way), but the boards did not connect to the AP that way either.

```
INFO ESPHome 2023.8.3
INFO Reading configuration /config/esphome/esphome-presence.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-presence (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5

...

Linking /data/esphome-presence/.pioenvs/esphome-presence/firmware.elf
RAM:   [=         ]   7.9% (used 25900 bytes from 327680 bytes)
Flash: [====      ]  43.6% (used 800128 bytes from 1835008 bytes)
Building /data/esphome-presence/.pioenvs/esphome-presence/firmware.bin
Creating esp32c3 image...
Successfully created esp32c3 image.
esp32_create_combined_bin(["/data/esphome-presence/.pioenvs/esphome-presence/firmware.bin"], ["/data/esphome-presence/.pioenvs/esphome-presence/firmware.elf"])
Wrote 0xd48a0 bytes to file /data/esphome-presence/.pioenvs/esphome-presence/firmware-factory.bin, ready to flash to offset 0x0
======================== [SUCCESS] Took 119.56 seconds ========================
INFO Successfully compiled program.
```

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.09.2

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitm-1

### Component causing the issue

wifi_esp32

### Example YAML snippet

```yaml
esphome:
[... 362 lines cut ...]
[D][esp-idf:000]: 

[D][esp-idf:000]: I (16009) wifi:
[D][esp-idf:000]: state: auth -> init (200)
[D][esp-idf:000]: 

[D][esp-idf:000]: I (16010) wifi:
[D][esp-idf:000]: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
[D][esp-idf:000]: 

[W][wifi_esp32:641]: Event: Disconnected ssid='SMITH_HA' bssid=[redacted] reason='Auth Expired'
[W][wifi:588]: Error while connecting to network.
[W][wifi:624]: Restarting WiFi adapter...
[D][esp-idf:000]: I (16024) wifi:
[D][esp-idf:000]: flush txq
[D][esp-idf:000]: 

[D][esp-idf:000]: I (16024) wifi:
[D][esp-idf:000]: stop sw txq
[D][esp-idf:000]: 

[D][esp-idf:000]: I (16025) wifi:
[D][esp-idf:000]: lmac stop hw txq
[D][esp-idf:000]:
```

HeMan, 2023-09-21:
Do you happen to be using an Ubiquiti network? I've had similar problems with my network and ESPs.

xdecock, 2023-09-25:
Same here managed ro work around it by changing the power to 

```yaml
wifi:
  output_power: 8.5
```

mclever, 2023-10-02:
I'm seeing the same thing on my ESP-WROOM-32's after the most recent update. Even with a basic config it can't connect to WiFi with auth expired errors. 

Tried the output_power change and it didn't make a difference.

Going to try and revert to backup of esphone add-on and re-flash.

aeonSolutions, 2023-12-17:
I've came across the same problem when coding my ESP32 base libraries ( see here : https://github.com/aeonSolutions/aeonlabs-ESP32-C-Base-Firmware-Libraries )

You have to add this code after running WiFi.softAP or WiFi.begin for the Wifi to work.

WiFi.setTxPower(WIFI_POWER_8_5dBm); 

see https://www.wemos.cc/en/latest/tutorials/c3/get_started_with_arduino_c3.html#wifi

ssieb, 2023-12-17:
Isn't that the same as the `output_power:` option?

rikki78, 2023-12-22:
> I'm seeing the same thing on my ESP-WROOM-32's after the most recent update. Even with a basic config it can't connect to WiFi with auth expired errors.
> 
> Tried the output_power change and it didn't make a difference.
> 
> Going to try and revert to backup of esphone add-on and re-flash.

Did reverting make a difference?

Particpant, 2024-01-07:
I'm seeing an inability to connect from some brand new ESP-WROOM32 boards. Same WiFi AP (and config) is just fine with a bunch of sonoff 8266 devices that are on an older ESPHome.

ThomDietrich, 2024-01-22:
I can confirm this issue.
I am working with the Esp32-c3-supermini devboard and @HeMan I am indeed on a Ubiquiti Unifi network.

The issue was resolved via:

```yaml
wifi:
  output_power: 8.5
```

clydebarrow, 2024-01-24:
> I am working with the Esp32-c3-supermini

What colour is the antenna? I have two types of these, one works, one doesn't - seems to be that the antenna is useless.

Good one on the left (red antenna) bad on the right.

[image]

ThomDietrich, 2024-01-24:
> seems to be that the antenna is useless.

Actually I can confirm these are trash :-( Setting `output_power` did *something* but the node is constantly dropping from the network.

nielsnl68, 2024-01-24:
I did have the same issues with the Lolin D1 c3 mini. In the end i stopped using them.

clydebarrow, 2024-01-24:
> ```
> [D][wifi:485]:     Channel: 6
> [D][wifi:486]:     RSSI: -82 dB
> ```

There is a clue here. My good C3 reports -49dBm in the same location as the bad one reports -81. That's a huge difference.

clydebarrow, 2024-01-24:
Just for kicks I removed the antenna from the bad one and soldered a 1/2 wavelength piece of wire in its place. That brought the reported  RSSI up quite a bit (to -50 or so) but connections still fail.

[image]

ThomDietrich, 2024-01-24:
Great initiative, sadly no success. I will show mine the way to the trash can.
Off topic but btw I am super happy with the "M5Stack M5Stamp C3U". Small form factor, a bit of eye candy, great wifi performance, and a company I deem supporting as they develop cool stuff. Only downside: $8 vs $3 per module

clydebarrow, 2024-01-24:
There's clearly something more than just a bad antenna going on. I have 7 and one of those will connect to WiFi but only with a strong signal, the rest never do. 

So unless there is anything new, this issue should be closed IMHO.

jalr, 2024-02-21:
I also tried @clydebarrow's modification, but in my case it made the problem worse and I ended up with no connection instead of a rare one like before.

I looked at the design of the filter on my board and it turns out that the LNA pin of the ESP is connected to a capacitor and one side of the chip antenna is then connected to the other side of that capacitor. I'm not that familiar with RF stuff, but this is different from the CLC circuit described in the [RF tuning section of the ESP Hardware Design Guidelines](https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32c3/schematic-checklist.html#rf-tuning)

Thinking I could neglect the capacitance for now as these are quite low, I wrapped a botched 4 turn air coil around the antenna wire (~1.8mm diameter). I removed the capacitor and replaced it with the coil. Then, I scraped off the PCB to expose the ground and soldered the antenna wire shield there. The second pad of the chip antenna is not grounded!
I have used an antenna from a discarded wifi router and also tried a diy biquad antenna, and both work quite well.

[image]

malexmave, 2024-02-22:
For me, when I have the exact same device and flash it from one of the pre-built binaries off the website (e.g., as a bluetooth proxy), it works fine. When I flash it using esphome in my HomeAssistant, it has the connection problems described here. So I doubt that this is a hardware issue, at least on my devices. I tried the software workarounds described here, and none of them seem to work for me. Is there a reason why so many people here seem to be sure that it is a hardware issue, and not a regression somewhere in the stack?

aeonSolutions, 2024-02-26:
> I also tried @clydebarrow's modification, but in my case it made the problem worse and I ended up with no connection instead of a rare one like before.
> 
> I looked at the design of the filter on my board and it turns out that the LNA pin of the ESP is connected to a capacitor and one side of the chip antenna is then connected to the other side of that capacitor. I'm not that familiar with RF stuff, but this is different from the CLC circuit described in the [RF tuning section of the ESP Hardware Design Guidelines](https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32c3/schematic-checklist.html#rf-tuning)
> 
> Thinking I could neglect the capacitance for now as these are quite low, I wrapped a botched 4 turn air coil around the antenna wire (~1.8mm diameter). I removed the capacitor and replaced it with the coil. Then, I scraped off the PCB to expose the ground and soldered the antenna wire shield there. The second pad of the chip antenna is not grounded! I have used an antenna from a discarded wifi router and also tried a diy biquad antenna, and both work quite well.
> 
> ![ESP32-C3_antenna_mod](https://private-user-images.githubusercontent.com/8084342/306661573-b34feeda-ca51-4b4f-84ad-10fdb16aa389.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg5NDQzMjksIm5iZiI6MTcwODk0NDAyOSwicGF0aCI6Ii84MDg0MzQyLzMwNjY2MTU3My1iMzRmZWVkYS1jYTUxLTRiNGYtODRhZC0xMGZkYjE2YWEzODkuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjZUMTA0MDI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzczMjFkMDM1OGNmZTgxYzM2MDRlYjNmMzhkYzA1MzJkOWI2NzY4N2U5MjU1NGJlMWU4NWIzN2JlODdjYzg5ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.o8J6E9iHNcxPwsjcIlWbGaDu1U2i3VwIl1pLRVWERlU)

if your board is using an onboard antenna the layout is correct as it is . 
If you're using an external antenna connected to a IPEX connector the CLC circuit is recommended but not mandatory.

Gelisob, 2024-03-12:
Can confirm same issue with latest (at the time of posting) ESPHOME and this esp32 c3 supermini with black ceramic antenna.
I got them with idea to replace ceramic with wired pcb antenna but was curious to check signal dBm difference before changing.
It kept failing with auth expired and other messages when having original black antenna. Closer to router with stronger signal after much spamming got in. I kept looking at numbers in esphome terminal so I could see what wifi networks and with what strength it saw all the time. Then I changed to pcb antenna and esp immediately saw more networks and dBm to my wifi went from -70 to -45. But now it never was able to join. So there must be something more going on with those black ceramic antenna units. Avoid until resolved. 

Same sketch onto red antenna esp32 c3 and instantly joined, no issues. 

Tried removing capacitor like @jalr did and actually soldering ground plane into ground i cleaned but no help. Do not know enough about antennas probably. But i'm quite sure that if i make this antenna change on red one - it will continue working normally.
[image]
[image]

HeMan, 2024-03-12:
Can you people confirm that you are not running Ubiquiti networks?

clydebarrow, 2024-03-12:
> Can you people confirm that you are not running Ubiquiti networks?

I cannot confirm that - i.e. I am using Ubiquiti. And I agree with the other comments that there is more to it than just antenna/signal strength. Any chance the chips are non-genuine?

HeMan, 2024-03-12:
The reason I asked is that I think it some kind of Ubiquiti-issue. I have a few different ESPs of varying types that have been working perfect for some years that suddenly got `Disconnected reason='Auth Expired'` with no change. They started working some days later and been working since.

rikki78, 2024-03-12:
I'm not using Ubiquiti and have the same issue.

Gelisob, 2024-03-12:
> Can you people confirm that you are not running Ubiquiti networks?

i'm not sure what you mean by that. Some specific network type? 
Mine is just common Telia home internet, hosted by an asus wifi router. 
And keep in mind, same sketch/location/network - red antenna unit - all works. This alone should be clear sign that it's some hardware issue. Add to that the fact, that from terminal I can see that it scans for networks and antenna change does change it's coverage and signal strengths.

ThomDietrich, 2024-03-12:
@HeMan I own a Ubiquiti Unifi wifi network but I think it is pretty clear by now that this is a simple bad hardware issue.

HeMan, 2024-03-13:
> @HeMan I own a Ubiquiti Unifi wifi network but I think it is pretty clear by now that this is a simple bad hardware issue.

Yes, it's pretty clear that I was barking up the wrong tree.

randybb, 2024-03-13:
Such problem on "Ubiquity" is only when people have no clue what they are doing and enable "Band Steering" for 2.4 GHz only devices. So please, don't blame Ubiquity :)

Gelisob, 2024-03-13:
maybe we can discuss what the hardware issue could be instead. Because - the device does see networks and tries connecting. Does esp32 have some even more core level firmware that uploading esphome sketch does not change, that we could flash over? Bootloader or something.. to take from "red" and put on "black" ?

malexmave, 2024-03-13:
Do all of you have the same issues when you flash a [pre-built ESPHome project](https://esphome.io/projects/) for something like a bluetooth proxy? Because, for me, the same device works with a pre-built image, but fails with the errors described here when compiling it myself.

clydebarrow, 2024-03-13:
> Do all of you have the same issues when you flash a [pre-built ESPHome project](https://esphome.io/projects/) for something like a bluetooth proxy?

Yes.

So, since this is clearly a hardware problem, not an ESPHome software problem, I'm closing this issue. Comments can still be added if there is any new information.

closed 2024-03-13 by clydebarrow