esphome/issues #4894: ESP Home continuously failing after unpacking Installing platformio/framework-espidf @ ~3.40405

Krispkiwi, 2023-09-18:
### The problem

I've been attempting to add a bluetooth proxy to ESPHome however when I press the install, I can never get it to complete the job. It keeps failing at "Installing platformio/framework-espidf @ ~3.40405." after reaching 100% on unpacking, the window falls to a "retry" state.

When pressing retry I then get the error "MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package"

I've been able to replicate this on multiple HA installs. Reinstalling ESP Home fixes the missing package just to fall back to the original issue again.

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9

### What platform are you using?

ESP8266

Krispkiwi, 2023-09-21:
Anyone have any ideas?

ys112, 2023-09-22:
+1 here facing the same issue installing platformio/framework-arduinoespressif32 @ ~3.20005.0, I got stuck at unpacking too. No luck with reinstalling and updating my pi.

Krispkiwi, 2023-09-22:
> +1 here facing the same issue installing platformio/framework-arduinoespressif32 @ ~3.20005.0, I got stuck at unpacking too. No luck with reinstalling and updating my pi.

Glad I'm not the only one, I've asked a few places and no one seems to have any idea what's going on.

Krispkiwi, 2023-09-23:
> +1 here facing the same issue installing platformio/framework-arduinoespressif32 @ ~3.20005.0, I got stuck at unpacking too. No luck with reinstalling and updating my pi.

Just doing some testing via windows powershell esphome now, it might have worked?? Just getting a second opinion because I've never done it this way before.

ys112, 2023-09-24:
@Krispkiwi you mean manually flashing using windows powershell?

Krispkiwi, 2023-09-28:
> @Krispkiwi you mean manually flashing using windows powershell?
> 
> I tried downgrading add-on versions and esp32 platform versions but seems like something is wrong with the installing of the framework.

Okay this approach has proved to be problematic. I'm not able to update the board the standard way after the latest esphome update.

ys112, 2023-09-28:
Currently I am using window's cmd prompt to compile and OTA. ESPHome addon in HA just would not go past the installing error for some reason even with the newer updates...

wrobelda, 2023-09-28:
Same issue here. A simple "Install" -> "Manual download" fails immediately with: 

```
INFO ESPHome 2023.9.1
INFO Reading configuration /config/esphome/blmproxy.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing blmproxy (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package
```

Krispkiwi, 2023-09-28:
> Same issue here. A simple "Install" -> "Manual download" fails immediately with:
> 
> ```
> INFO ESPHome 2023.9.1
> INFO Reading configuration /config/esphome/blmproxy.yaml...
> INFO Generating C++ source...
> INFO Compiling app...
> Processing blmproxy (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32@5.4.0)
> --------------------------------------------------------------------------------
> MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package
> ```
> 
> Meanwhile the ESP8266 compilation works fine:
> 
> ```
> INFO ESPHome 2023.9.1
> INFO Reading configuration /config/esphome/d1mini.yaml...
> INFO Generating C++ source...
> INFO Compiling app...
> Processing d1mini (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
> --------------------------------------------------------------------------------
> HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
> ```
> 
> So it would appear there's something wrong with some of the packages?
> 
> EDIT: I can build `esp32-c3-devkitm-1` just fine, but not `esp32doit-devkit-v1` nor `esp32dev`, so it appears that only the ESP32 base variant is affected.
> 
> EDIT2: switching to ESP-IDF framework worked for me, though, so at least there's a workaround.

Excuse my ignorance, how did you switch frameworks?

wrobelda, 2023-09-28:
> Excuse my ignorance, how did you switch frameworks?

https://esphome.io/components/esp32.html#esp-idf-framework

ssieb, 2023-09-29:
> MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

@wrobelda that's a different issue.  re-installing the addon should fix that one.

ys112, 2023-10-02:
Interestingly I am able to go pass the installing framework issue today and got an error of 
sh: 1: xtensa-esp32-elf-g++: not found which was fixed by disabling 64bit on pi4

Krispkiwi, 2023-10-02:
Mines still failing today.

Krispkiwi, 2023-10-02:
> > Excuse my ignorance, how did you switch frameworks?
> 
> https://esphome.io/components/esp32.html#esp-idf-framework

Drat looks like my framework is already esp-idf

DukemFIVE, 2023-10-19:
same here, also tried with the esp-idf and ardunio framework

thzzlan, 2023-11-23:
I have the same problem. ESPHome fails after unpacking the framework and then HA is extremely slow for a while. The supervisor seems to crash as well but it restarts after some time. I tried the esp-idf and the arduino framework, but both caused ESPHome to fail unpacking.

matthewliut, 2023-11-27:
RIP still a problem...

matthewliut, 2023-11-27:
Really?? IL give that a try for sure. I already tried using a vm as well but when compiling it, the supervisor crashed on the vm as well 😭. (It did get to the crash point alot faster tho LOLL)

matthewliut, 2023-11-28:
> How much RAM do you have?

I have a pi4 b with 4gb of ram!

matthewliut, 2023-11-28:
> To shed some hope, you could try using the HA app on your phone to update instead of a PC browser. When I used a PC browser, the compiling always failed at certain points with intensive resource usage on pi4. But when I used the HA app on my phone, I was able to successfully compile and OTA. Weird but it worked.

Holy fuck this worked first try. Took like 3 minutes too loll. Finally I can make my stupid bed sensor 🤣

NikolasRupp, 2023-12-04:
Hi has anyone found another solution for this. Having the same Problem

flashvenom, 2023-12-15:
> Hi has anyone found another solution for this. Having the same Problem

Same here. In a similar thread, the user changed the SD card and that fixed it, so perhaps I need to use a bigger/better SD card on my raspi?

flashvenom, 2023-12-22:
Just wanted to update for @NikolasRupp and future readers here. **It would appear it was the SD card.** I had 30% free, but it was also a very cheap Microcenter micrSD card. Backed up everything, installed HA fresh on a SanDisk sdcard, restored from backup.  I also has a few minor upgrades to HA Core, so I did that as well which may have fixed the problem I guess...

But everything is much better now. HA loads way faster and I seem to have way less errors... EspHome is compiling as intended...

alexcf, 2024-04-30:
For those who have this issue, delete the packages folder in your platformio folder. 
https://github.com/esphome/issues/issues/4272

ELKER54, 2024-05-11:
Even after deleting the packages folder I still cannot install it. Keep getting the error:
InvalidJSONFile: Could not load broken JSON: /root/.platformio/packages/framework-arduinoespressif32/.piopm

nacree, 2025-01-24:
I'm still encountering this same issue. I had some issues connecting to my ESP device, it could not read logs over wifi or locally, so ended up reinstalling the addon, but now I cannot even build it locally.

And unfortunately does not work with mobile app either.. "Unpacking" somehow slowly gets into 100% and almost freezes the whole home assistant. Would feel weird that SD card started to do this suddenly, when it worked still yesterday when I had my first touch with ESPHome.

ssieb, 2025-11-27:
What are you running HA on?  Compiling requires a lot of resources, particularly CPU and RAM.