esphome/issues #4897: Error with last command : incorrect Header

ayubn, 2023-09-19:
### The problem

hi i  hv installed an ld2410 on esp32 using esphome version 2023.2.1 
it start working fine but after some time it start throwing error and stop detection  

[ld2410:177] | Error with last command : incorrect Header

some time its 177 and some time 236 etc 

### Which version of ESPHome has the issue?

2023.2.1 

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
uart:
  tx_pin: GPIO17
  rx_pin: GPIO5
  baud_rate: 256000
  parity: NONE
  stop_bits: 1
  
  

ld2410:
  timeout: 1s
  max_move_distance: 1.50m
  max_still_distance: 1.50m
  g0_move_threshold: 40  # 0m / 0'
  g0_still_threshold: 10 # 0m / 0'
  g1_move_threshold: 40  # 0 - 0.75m / 0 - 2.46'
  g1_still_threshold: 10 # 0 - 0.75m / 0 - 2.46'
  g2_move_threshold: 40  # 0.75 - 1.5m / 2.46' - 4.92'
  g2_still_threshold: 10 # 0.75 - 1.5m / 2.46' - 4.92'
  # g3_move_threshold: 40  # 1.5 - 2.25m / 4.92' - 7.38'
  # g3_still_threshold: 10 # 1.5 - 2.25m / 4.92' - 7.38'
  # g4_move_threshold: 40  # 2.25 - 3m' / 7.38' - 9.84'
  # g4_still_threshold: 40 # 2.25 - 3m' / 7.38' - 9.84'
  # g5_move_threshold: 40  # 3 - 3.75 / 9.84' - 12.30'
  # g5_still_threshold: 40 # 3 - 3.75 / 9.84' - 12.30'
  # g6_move_threshold: 30  # 3.75 - 4.5m / 12.30' - 14.76'
  # g6_still_threshold: 15 # 3.75 - 4.5m / 12.30' - 14.76'
  # g7_move_threshold: 30  # 4.5 - 5.25m / 14.76' - 17.22'
  # g7_still_threshold: 15 # 4.5 - 5.25m / 14.76' - 17.22'
  # g8_move_threshold: 30  # 5.25 - 6m / 17.22' - 19.68'
  # g8_still_threshold: 15 # 5.25 - 6m / 17.22' - 19.68'

sensor:
    - platform: ld2410
      moving_distance:
        name: Moving Distance
      still_distance:
        name: Still Distance
      moving_energy:
        name: Move Energy
      still_energy:
        name: Still Energy
      detection_distance:
        name: Detection Distance

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
```

### Anything in the logs that might be useful for us?

```txt
[ld2410:177] | Error with last command : incorrect Header

some time its 177 and some time 236 etc
```

ayubn, 2023-09-19:
One more issue ld2410 start fake detection when my ac is turn on anyone have idea abt that or anyone can help me where should i post this question

dflvunoooooo, 2023-11-15:
I too have the `[ld2410:336]: Error with last command : incorrect Header` error in the logs.

klimentiy23, 2024-01-22:
[17:08:01][C][ld2410:077]:   Firmware Version : 2.04.23022511
[17:08:01][E][ld2410:336]: Error with last command : incorrect Header

MrKetsvil, 2024-01-31:
if (buffer[0] != 0xFD || buffer[1] != 0xFC || buffer[2] != 0xFB || buffer[3] != 0xFA) {  // check 4 frame start bytes
     ESP_LOGE(TAG, "**Error with last command : incorrect Header**");
     return true;

https://esphome.io/api/ld2410_8cpp_source

neuquenfr, 2024-04-08:
Hi. Any solution found? I have the same problem on several of my LD2410C sensors, too. Thanks!

hamgang, 2024-05-03:
同样遇到这个问题，目前没有找到解决办法。模块是LD2410C。

urgemore, 2024-05-24:
I don't know if this will be helpful to anyone, but I just started encountering this problem last night with my Skreek Sensor 1U, which uses the LD2410. I saw the error message this morning and decided to try reinstalling the same latest firmware it was running. That seems to have cleared the error.

Pseudomizer, 2025-05-12:
Year 2025 and I got the same error. After I renamed the device in the ESP Home Builder and updated it wirelessly.