esphome/issues #4898: SGP30 never finishes baseline calibration

Pitel, 2023-09-19:
### The problem

I bought SGP30 sensor, connected it, and after 3 days, it still says: "Baseline reading not available for: 43029s"

Sure, the timer goes down, but when I check it the next day, still no baseline.

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

sgp30

### Example YAML snippet

```yaml
sensor:
  - platform: dht
    pin: D5
    model: DHT22
    temperature:
      id: "temperature"
      name: "temperature"
      accuracy_decimals: 1
      filters:
        - delta: 0.2
      on_value:
        http_request.post:
          url: "http://192.168.XXXXX/t"
          body: !lambda |-
            char buf[8];
            sprintf(buf, "%.1f", x);
            return buf;
    humidity:
      id: "humidity"
      name: "humidity"
      accuracy_decimals: 1
      filters:
        - delta: 0.2
      on_value:
        http_request.post:
          url: "http://192.168.XXXXX/h"
          body: !lambda |-
            char buf[8];
            sprintf(buf, "%.1f", x);
            return buf;
  - platform: sgp30
    eco2:
      name: "eCO2"
      accuracy_decimals: 0
      filters:
        - throttle_average: 60s
        - delta: 10
      on_value:
        http_request.post:
          url: "http://192.168.XXXXX/eco2"
          body: !lambda |-
            char buf[8];
            sprintf(buf, "%d", (int) x);
            return buf;
    tvoc:
      name: "TVOC"
      accuracy_decimals: 0
      filters:
        - throttle_average: 60s
        - delta: 10
      on_value:
        http_request.post:
          url: "http://192.168.XXXX/tvoc"
          body: !lambda |-
            char buf[8];
            sprintf(buf, "%d", (int) x);
            return buf;
    compensation:
      temperature_source: "temperature"
      humidity_source: "humidity"
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.8.3
INFO Reading configuration /home/pitel/esphome/dht.yaml...
INFO Starting log output from dht.local using esphome API
INFO Successfully connected to dht.local
[14:49:48][I][app:102]: ESPHome version 2023.8.3 compiled on Sep 18 2023, 13:03:23
[14:49:48][C][wifi:543]: WiFi:
[14:49:48][C][wifi:379]:   Local MAC: XXXXXX
[14:49:48][C][wifi:380]:   SSID: 'R2D2'
[14:49:48][C][wifi:381]:   IP Address: 192.168.XXXX
[14:49:48][C][wifi:382]:   BSSID: XXXXXXXXXX
[14:49:48][C][wifi:384]:   Hostname: 'dht'
[14:49:48][C][wifi:386]:   Signal strength: -74 dB ▂▄▆█
[14:49:48][C][wifi:390]:   Channel: 1
[14:49:48][C][wifi:391]:   Subnet: 255.255.255.0
[14:49:48][C][wifi:392]:   Gateway: 192.168.XXXXX
[14:49:48][C][wifi:393]:   DNS1: 192.168.XXXXXX
[14:49:48][C][wifi:394]:   DNS2: 0.0.0.0
[14:49:48][D][sgp30:291]: Got eCO2=400.0ppm TVOC=8.0ppb
[14:49:48][D][sgp30:171]: External compensation data received: Humidity 68.90%
[14:49:48][D][sgp30:181]: External compensation data received: Temperature 25.80°C
[14:49:48][D][sgp30:198]: Calculated Absolute humidity: 16.593 g/m³ (0x1097)
[14:49:48][D][sgp30:156]: Baseline reading not available for: 42799s
[14:49:48][C][logger:301]: Logger:
[14:49:48][C][logger:302]:   Level: DEBUG
[14:49:48][C][logger:303]:   Log Baud Rate: 115200
[14:49:48][C][logger:305]:   Hardware UART: UART0
[14:49:48][C][i2c.arduino:053]: I2C Bus:
[14:49:48][C][i2c.arduino:054]:   SDA Pin: GPIO4
[14:49:48][C][i2c.arduino:055]:   SCL Pin: GPIO5
[14:49:48][C][i2c.arduino:056]:   Frequency: 50000 Hz
[14:49:48][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[14:49:48][I][i2c.arduino:069]: Results from i2c bus scan:
[14:49:48][I][i2c.arduino:075]: Found i2c device at address 0x58
[14:49:48][C][dht:017]: DHT:
[14:49:48][C][dht:018]:   Pin: GPIO14
[14:49:48][C][dht:024]:   Model: DHT22 (or equivalent)
[14:49:48][C][dht:027]:   Update Interval: 60.0s
[14:49:48][C][dht:029]:   Temperature 'temperature'
[14:49:48][C][dht:029]:     Device Class: 'temperature'
[14:49:48][C][dht:029]:     State Class: 'measurement'
[14:49:48][C][dht:029]:     Unit of Measurement: '°C'
[14:49:48][C][dht:029]:     Accuracy Decimals: 1
[14:49:48][C][dht:030]:   Humidity 'humidity'
[14:49:48][C][dht:030]:     Device Class: 'humidity'
[14:49:48][C][dht:030]:     State Class: 'measurement'
[14:49:48][C][dht:030]:     Unit of Measurement: '%'
[14:49:48][C][dht:030]:     Accuracy Decimals: 1
[14:49:49][C][sgp30:229]: SGP30:
[14:49:49][C][sgp30:230]:   Address: 0x58
[14:49:49][C][sgp30:250]:   Serial number: 31483098
[14:49:49][C][sgp30:256]:   Baseline: No baseline configured
[14:49:49][C][sgp30:258]:   Warm up time: 43200s
[14:49:49][C][sgp30:260]:   Update Interval: 1.0s
[14:49:49][C][sgp30:261]:   eCO2 sensor 'eCO2'
[14:49:49][C][sgp30:261]:     Device Class: 'carbon_dioxide'
[14:49:49][C][sgp30:261]:     State Class: 'measurement'
[14:49:49][C][sgp30:261]:     Unit of Measurement: 'ppm'
[14:49:49][C][sgp30:261]:     Accuracy Decimals: 0
[14:49:49][C][sgp30:261]:     Icon: 'mdi:molecule-co2'
[14:49:49][C][sgp30:262]:   TVOC sensor 'TVOC'
[14:49:49][C][sgp30:262]:     Device Class: 'volatile_organic_compounds_parts'
[14:49:49][C][sgp30:262]:     State Class: 'measurement'
[14:49:49][C][sgp30:262]:     Unit of Measurement: 'ppb'
[14:49:49][C][sgp30:262]:     Accuracy Decimals: 0
[14:49:49][C][sgp30:262]:     Icon: 'mdi:radiator'
[14:49:49][C][sgp30:265]: Store baseline: YES
[14:49:49][C][sgp30:267]:   Compensation:
[14:49:49][C][sgp30:268]:     Temperature Source: 'temperature'
[14:49:49][C][sgp30:268]:       Device Class: 'temperature'
[14:49:49][C][sgp30:268]:       State Class: 'measurement'
[14:49:49][C][sgp30:268]:       Unit of Measurement: '°C'
[14:49:49][C][sgp30:268]:       Accuracy Decimals: 1
[14:49:49][C][sgp30:269]:     Humidity Source: 'humidity'
[14:49:49][C][sgp30:269]:       Device Class: 'humidity'
[14:49:49][C][sgp30:269]:       State Class: 'measurement'
[14:49:49][C][sgp30:269]:       Unit of Measurement: '%'
[14:49:49][C][sgp30:269]:       Accuracy Decimals: 1
[14:49:49][C][sntp:053]: SNTP Time:
[14:49:49][C][sntp:054]:   Server 1: 'cz.pool.ntp.org'
[14:49:49][C][sntp:055]:   Server 2: ''
[14:49:49][C][sntp:056]:   Server 3: ''
[14:49:49][C][sntp:057]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[14:49:49][C][mdns:112]: mDNS:
[14:49:49][C][mdns:113]:   Hostname: dht
[14:49:49][C][ota:093]: Over-The-Air Updates:
[14:49:49][C][ota:094]:   Address: dht.local:8266
[14:49:49][C][ota:097]:   Using Password.
[14:49:49][C][api:138]: API Server:
[14:49:49][C][api:139]:   Address: dht.local:6053
[14:49:49][C][api:141]:   Using noise encryption: YES
[14:49:49][C][http_request:014]: HTTP Request:
[14:49:49][C][http_request:015]:   Timeout: 5000ms
[14:49:49][C][http_request:016]:   User-Agent: ESPHome
[14:49:49][C][http_request:017]:   Follow Redirects: 1
[14:49:49][C][http_request:018]:   Redirect limit: 3
[14:49:49][D][sgp30:291]: Got eCO2=400.0ppm TVOC=2.0ppb
[14:49:49][D][sgp30:171]: External compensation data received: Humidity 68.90%
[14:49:49][D][sgp30:181]: External compensation data received: Temperature 25.80°C
[14:49:49][D][sgp30:198]: Calculated Absolute humidity: 16.593 g/m³ (0x1097)
[14:49:49][D][sgp30:156]: Baseline reading not available for: 42799s
[14:49:50][D][sgp30:291]: Got eCO2=400.0ppm TVOC=0.0ppb
[14:49:50][D][sgp30:171]: External compensation data received: Humidity 68.90%
[14:49:50][D][sgp30:181]: External compensation data received: Temperature 25.80°C
[14:49:50][D][sgp30:198]: Calculated Absolute humidity: 16.593 g/m³ (0x1097)
[14:49:50][D][sgp30:156]: Baseline reading not available for: 42798s
```

xdecock, 2023-09-24:
Could you add an [Uptime sensor](https://esphome.io/components/sensor/uptime.html) ? To make sure there's no reboot?

```yaml
sensor:
  - platform: uptime
    name: Uptime Sensor
```

Pitel, 2023-09-26:
It looks like it reboots. I'll try better power supply. I'm using some cheap old chinese phone charger now.
[image]

Pitel, 2023-09-30:
Better power supply (Raspberry Pi power adapter capable of 5 V 2,5 A) didn't help.

Is it somehow possible to detect the reboot and get some logs or something, even when it happens randomly?

xdecock, 2023-09-30:
I'd log over usb and write that to a file but i hope there's a more power
efficient way to solve this

Pitel, 2023-10-01:
Keeping it running all day with attached logging over WiFi has finally resulted in acquiring the baseline without reboots. It's like Schroedinger's sensor or something. 🤡

Pitel, 2023-10-02:
I found a cause of the reboot!

`api` component has `reboot_timeout` parameter, which defaults to 15 minutes. So the unit reboots every 15 minutes, unless there is an API connection, for example logging.

I noticed there are more similar reboot components for other components (wifi, mqtt), with similar default values. I understand why they're here, but they make it really hard to run the inital 12 hours calibration of SGP30 sensor.

**Maybe this should be mentioned in documentation!**