esphome/issues #4899: Update needed for newer AHT20 boards from Adafruit

dechandler, 2023-09-19:
### The problem

They fixed this for their library 2.0.5 - it looks like the fix is just waiting a bit in the calibration phase.
https://github.com/adafruit/Adafruit_AHTX0/releases

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

Wemos Mini D1 (clone)

### Component causing the issue

aht10

### Anything in the logs that might be useful for us?

```txt
[15:16:58][I][app:029]: Running through setup()...
[15:16:58][I][i2c.arduino:183]: Performing I2C bus recovery
[15:16:58][C][aht10:030]: Setting up AHT10...
[15:16:58][E][aht10:033]: Communication with AHT10 failed!
[15:16:58][E][component:113]: Component aht10.sensor was marked as failed.
```

Unfocused, 2024-01-24:
There were some fixes made to the AHT10 component since you posted this:
- https://github.com/esphome/esphome/pull/5198

@dechandler, could you confirm whether this is still an issue for you when using ESPHome 2023.12.0 or newer?

dechandler, 2024-01-25:
That worked, thank you for letting me know!

For the record, it's with 2023.12.9 and a D1 mini clone, and I had to add `variant: aht20` to the sensor block (which I hadn't seen before).

closed 2024-01-25 by dechandler