esphome/issues #4901: ESP32+esp32_ble_tracker+Mqtt --> Backtrace:0x40083829:0x3ffd7dc00x4009491d:0x3ffd7de0 Crash !

zibous, 2023-09-21:
### The problem

If esp32_ble_tracker and mqtt is enabled, application crashes with backtrace.
API Component (w/o mqtt) is working !

Version **2022.12.8** Feb 10 2023 is working !

Compiling with PIP, Docker - does not make a difference. 

### Which version of ESPHome has the issue?

2023.9.0b2,  2023.10.0-dev, 2023.8.3 ...

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4 and any other ESP32 Board

### Component causing the issue

MQTT

### Example YAML snippet

```yaml
## ---------------------------------------------------
# ESP32 MQTT TESTCASE
## ---------------------------------------------------
substitutions:
  ## device settings
  device_name_short: "mqtt-test"
  device_description: "Testdevice: ESP32 MQTT Testcase"
  projectname: "MQTT-TEST.ESP32"
  appversion: "2.0.0"
  domain: !secret domain

## ---------------------------------------------------
## BOARD
## ---------------------------------------------------
esp32:
  board: az-delivery-devkit-v4

## ---------------------------------------------------
## ESPHOME APPLICATION
## ---------------------------------------------------
esphome:
  name: ${device_name_short}
  friendly_name: ${device_name_short}
  comment: ${device_description}
  project:
    name: ${projectname}
    version: ${appversion}
  build_path: ./build/${device_name_short}
  platformio_options:
    platform: espressif32
    board: az-delivery-devkit-v4
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
  on_boot:
    then:
      - logger.log:
          level: WARN
          tag: "system"
[... 503 lines cut ...]
WARNING Decoded 0x400f0c52: loopTask(void*) at /Users/petsie1612/.platformio/packages/framework-arduinoespressif32@3.20005.220925/cores/esp32/main.cpp:42
[15:43:31]
[15:43:31]
[15:43:31]
[15:43:31]
[15:43:31]ELF file SHA256: 0000000000000000
[15:43:31]
[15:43:31]Rebooting...
[15:43:31]ets Jun  8 2016 00:22:57
[15:43:31]
[15:43:31]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:43:31]configsip: 0, SPIWP:0xee
[15:43:31]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:43:31]mode:DIO, clock div:2
[15:43:31]load:0x3fff0030,len:1184
[15:43:31]load:0x40078000,len:13132
[15:43:31]load:0x40080400,len:3036
[15:43:31]entry 0x400805e4
```

### Additional information

Linking .pioenvs/mqtt-test/firmware.elf
RAM:   [=         ]  10.3% (used 55072 bytes from 532480 bytes)
Flash: [========= ]  94.2% (used 1728593 bytes from 1835008 bytes)  ???

zibous, 2023-09-22:
# Device

```
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
Auto-detected Flash size: 4MB
```

## Testcases

<table>
	<tr>
		<th>MQTT</th>
		<th>API</th>
		<th>ESP32_BLE_TRACKER</th>
		<th>RAM</th>
		<th>FLASH</th>
		<th>STATE</th>		
	</tr>
	<tr>
		<td>✔︎ yes</td>
		<td>✘ no</td>
		<td>✔︎ yes</td>
		<td>10.30%</td>
		<td><b>94.20%</b></td>
		<td><b>✘ ERROR !!</b></td>		
	</tr>
		<td>✔︎ yes</td>
		<td>✘ no</td>
		<td>✘ no</td>
		<td>8.00%</td>
		<td>56.60%</td>
		<td>✔︎ WORKING</td>		
	</tr>
	</tr>
		<td>✘ no</td>
		<td>✔︎ yes</td>
		<td>✔︎ yes</td>
		<td>10.20%</td>
		<td><b>90.50%</b></td>
		<td>✔︎ WORKING</td>		
	</tr>
	</tr>
		<td>✘ no</td>
		<td>✔︎ yes</td>
		<td>✘ no</td>
		<td>7.90%</td>
		<td>52.90%</td>
		<td>✔︎ WORKING</td>		
	</tr>
</table>

Result: mqtt + esp32_ble_tracker do not work !!!

michaelscl, 2023-12-23:
I have the same chip.
Today I accidentally found out that my ESP has been malfunctioning for a few months. The problem is the same.The kernel crashes and ESP reboots constantly. I had to do a downgrade to 2022.12.8 where everything really works. What can I do to get the bug fixed in the latest version 2023.12.3 ?

```
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
Auto-detected Flash size: 4MB

esp32:
  board: "denky32"
```

ssieb, 2023-12-23:
If it's the same error as pasted earlier, it's running out of RAM.

michaelscl, 2023-12-24:
Today, I found some more information and set it up:
```
esp32:
  board: esp32dev
  framework:
    type: arduino
```
With this setting it is now OK.

michaelscl, 2023-12-24:
Today, I found some more information and set it up:
```
esp32:
  board: esp32dev
  framework:
    type: arduino
```
With this setting it is now OK. ESP is not resetting.

But it confirms that the combination of MQTT + BLE does not work. ESP does not connect to MQTT.
```

[I][wifi:587]: WiFi Connected!
[C][web_server:123]: Setting up web server...
[C][sntp:028]: Setting up SNTP...
[C][mqtt:037]: Setting up MQTT...
[D][mqtt:180]: Resolving MQTT broker IP address...
[D][mqtt:213]: Resolved broker IP address to 10.140.1.40
[I][mqtt:234]: Connecting to MQTT...
[D][esp-idf:000]: E (29005) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned 0
[D][esp-idf:000]: E (29006) MQTT_CLIENT: MQTT connect failed
[W][mqtt:323]: MQTT Disconnected: TCP disconnected.
```

michaelscl, 2023-12-24:
Today, I found some more information and set it up:
```
esp32:
  board: esp32dev
  framework:
    type: arduino
```
With this setting it is now OK. ESP is not resetting!

And if someone has a problem with BLE + MQTT, you need to disable web_server.

Now everything is working.

davidsarkany, 2024-03-02:
Same issue with 2024.2.1.
I don't have problem with 2023.4.1.
```
esphome:
  name: esp

esp32:
  board: lolin32_lite
  framework:
    type: arduino
    
captive_portal:

logger:

api:
  password: !secret ApiPassword

wifi:
  ssid: !secret WifiSsid
  password: !secret WifiPassword
  ap:
    ssid: !secret FallbackWifiSsid
    password: !secret FallbackWifiPassword
  
mqtt:
  broker: !secret MqttBroker
  username: !secret MqttUsername
  password: !secret MqttPassword
  port: !secret MqttPort
 
ota:
  password: !secret OtaPassword

esp32_ble_tracker:

sensor:
  - platform: atc_mithermometer
    mac_address: "XX:XX:XX:XX:XX:XX"
    temperature:
      id: livingroom_temperature
      name: "Livingroom temperature"
      filters:
        - throttle: 1s
        - heartbeat: 30s
        - debounce: 0.1s
      on_value:
        - mqtt.publish:
            topic: "sensor/livingroom_temperature"
            payload: !lambda |-
              return to_string(id(livingroom_temperature).state);
    humidity:
      id: livingroom_humidity
      name: "Livingroom humidity"
      filters:
        - throttle: 1s
        - heartbeat: 30s
        - debounce: 0.1s
      on_value:
        - mqtt.publish:
            topic: "sensor/livingroom_humidity"
            payload: !lambda |-
              return to_string(id(livingroom_humidity).state);
    battery_level:
      id: livingroom_battery
      name: "Livingroom battery"
      filters:
        - throttle: 1s
        - heartbeat: 30s
        - debounce: 0.1s
      on_value:
        - mqtt.publish:
            topic: "sensor/livingroom_battery"
            payload: !lambda |-
              return to_string(id(livingroom_battery).state);

  - platform: atc_mithermometer
    mac_address: "XX:XX:XX:XX:XX:XX"
    temperature:
      id: bed_room_temperature
      name: "Bed room temperature"
      filters:
        - throttle: 1s
        - heartbeat: 30s
        - debounce: 0.1s
      on_value:
        - mqtt.publish:
            topic: "sensor/bed_room_temperature"
            payload: !lambda |-
              return to_string(id(bed_room_temperature).state);
    humidity:
      id: bed_room_humidity
      name: "Bed room humidity"
      filters:
        - throttle: 1s
        - heartbeat: 30s
        - debounce: 0.1s
      on_value:
        - mqtt.publish:
            topic: "sensor/bed_room_humidity"
            payload: !lambda |-
              return to_string(id(bed_room_humidity).state);
    battery_level:
      id: bed_room_battery
      name: "Bed room battery"
      filters:
        - throttle: 1s
        - heartbeat: 30s
        - debounce: 0.1s
      on_value:
        - mqtt.publish:
            topic: "sensor/bed_room_battery"
            payload: !lambda |-
              return to_string(id(bed_room_battery).state);

  - platform: atc_mithermometer
    mac_address: "XX:XX:XX:XX:XX:XX"
    temperature:
      id: outdoor_temperature
      name: "Outdoor temperature"
      filters:
        - throttle: 1s
        - heartbeat: 30s
        - debounce: 0.1s
      on_value:
        - mqtt.publish:
            topic: "sensor/outdoor_temperature"
            payload: !lambda |-
              return to_string(id(outdoor_temperature).state);
    humidity:
      id: outdoor_humidity
      name: "Outdoor humidity"
      filters:
        - throttle: 1s
        - heartbeat: 30s
        - debounce: 0.1s
      on_value:
        - mqtt.publish:
            topic: "sensor/outdoor_humidity"
            payload: !lambda |-
              return to_string(id(outdoor_humidity).state);
    battery_level:
      id: outdoor_battery
      name: "Outdoor battery"
      filters:
        - throttle: 1s
        - heartbeat: 30s
        - debounce: 0.1s
      on_value:
        - mqtt.publish:
            topic: "sensor/outdoor_battery"
            payload: !lambda |-
              return to_string(id(outdoor_battery).state);

button:
  - platform: wake_on_lan
    id: "start_pc_button"
    name: "Start PC"
    target_mac_address: XX:XX:XX:XX:XX:XX
```

ssieb, 2024-03-04:
@davidsarkany the answers have already been given.