esphome/issues #4904: ESP8266 not functioning when UART connected. Connecting after power up works fine.

DracoTomes, 2023-09-24:
### The problem

Hello,

I'm setting up an ESP8266 that is connected via UART. The ESP is a NodeMCU from AZ-Delivery, it's currently on a breadboard.
GPIO13 and PGIO15 are used for the UART bus. When the MCU receives power while the GPIOs are connected it will not work completely. It is not outputting to hardware UART / USB cable and will not do any automations. It seems GPIO15 is also pulled high since the TX LED on the RS-485 Transceiver will light up. The only way I can interact with the device is to flash a new firmware OTA, even though the usual ESPHome API is not reachable.
If I unplug the cables on GPIO13 and GPIO15 the device will boot normally.
Afterwards I plug in the cables and the device will function using this UART bus for modbus commands... until the next reboot.

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

UART

### Example YAML snippet

```yaml
esphome:
  name: "test-board"
  friendly_name: Test Board

esp8266:
  board: nodemcuv2

logger:
  level: VERBOSE
  baud_rate: 0

uart:
  tx_pin: GPIO015
  rx_pin: GPIO013
  baud_rate: 9600

modbus:
  id: modbus1

modbus_controller:
  - id: test
    address: 11
    modbus_id: modbus1
```

randybb, 2023-09-24:
If GPIO15 is pulled HIGH during booting, then it will boot from SDIO, which is not what you want. You are using GPIO15 as output, but probably that modbus module has a pullup resistor (like this one https://protosupplies.com/product/max485-ttl-to-rs-485-interface-module/) which is stronger that the pulldown resistor (12k) on your NodeMCU board.

DracoTomes, 2023-09-24:
That checks out. Given this limitation in the [documentation](https://esphome.io/components/uart.html), I don't have another choice of pins to use though:

> The ESP8266 has two UARTs; the second of which is TX-only. Only a limited set of pins can be used. UART0 may use either tx_pin: GPIO1 and rx_pin: GPIO3, or tx_pin: GPIO15 and rx_pin: GPIO13. UART1 must use tx_pin: GPIO2. Any other combination of pins will result in use of a software UART.

Could I possibly add another resistor to pull the pin down again?

randybb, 2023-09-24:
Remove TX pullup resistor on that board, or use a different module without pullups. Or, ideally, switch to esp32. esp8266 is way too limited.

DracoTomes, 2023-09-24:
> Or, ideally, switch to esp32

It seems I've been trying to delay the inevitable for too long.

closed 2023-09-24 by DracoTomes

ssieb, 2023-09-24:
It's only 9600 baud, so the software uart should work fine.  But switching to ESP32 would be better.