esphome/issues #4905: ili9xx/ST7796 display needs 4.77s when using framework esp-idf/recommended instead 0,17s with arduino/recommended

fsiek, 2023-09-24:
### The problem

Experimenting with a ESP32 Wroom32, 240Mhz, 520kB Ram, 4MB flash, connected to a ST7796 with 320x480.

Due to heap memory restrictions I am experimenting using esp-idf instead of arduino as it gives me ~20kB larger maximum heap block for the display buffer.

Even esp-idf gives me a larger possible display area, the display is running very slow with the esp-idf framework. Setting the data_rate for the display in the yaml file has no effect.

It look like the SPI data rate is not set correctly to the possible 40MHz.

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.9.1

### What platform are you using?

ESP32

### Board

ESP32-D0WD-V3 (revision v3.1)

### Component causing the issue

Display ST7796

### Example YAML snippet

```yaml
esphome:
  name: display1
  friendly_name: Display1

esp32:
  board: az-delivery-devkit-v4
  framework: 
    type: esp-idf
    version: recommended
      
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "........"

ota:
  password: "........."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
[... 323 lines cut ...]
[21:36:08][I][wifi:274]: WiFi Connecting to 'WLAN1'...
[21:36:08][D][esp-idf:000]: I (7539) wifi:
[21:36:08][D][esp-idf:000]: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
[21:36:08][D][esp-idf:000]:
[21:36:08]
[21:36:08][D][esp-idf:000]: I (7547) wifi:
[21:36:08][D][esp-idf:000]: state: init -> auth (b0)
[21:36:08][D][esp-idf:000]:
[21:36:08]
[21:36:12][W][component:204]: Component ili9xxx.display took a long time for an operation (4.77 s).
[21:36:12][W][component:205]: Components should block for at most 20-30ms.
[21:36:13][D][esp-idf:000]: I (13061) wifi:
[21:36:13][D][esp-idf:000]: state: auth -> init (200)
[21:36:13][D][esp-idf:000]:
[21:36:13]
[21:36:13][D][esp-idf:000]: I (13063) wifi:
[21:36:13][D][esp-idf:000]: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
[21:36:13][D][esp-idf:000]:
[21:36:13]
```

### Additional information

I changed display.py to work with color_palette: GRAYSCALE in YAML according to recommendation from clydebarrow:
https://github.com/esphome/esphome/pull/5406 but the problem shows up with color_palette: IMAGE_ADAPTIVE, too.

xdecock, 2023-09-24:
The beta version of esphome fixes most of it due to this pr: https://github.com/esphome/esphome/pull/5311

To get the full 40Mhz on esp32 i had to avoid defining the miso pin as there seem to be a limit to 26Mhz for full duplex connections.

fsiek, 2023-09-24:
Sorry, not speaking about factor 2 but 25. SPI seams to run at 1.6 Mhz instead of 40Mhz.

nielsnl68, 2023-09-24:
Hi could you run esphome in DEV mode and use PR#5406 to see of your issue will be fixed with that one.

fsiek, 2023-09-25:
OK, checked in dev mode and it looks much better now. This time I can see in the log, that the data is set. 
`[09:17:11][D][spi_device:354]: mode 0, data_rate 40000kHz`

This leads to a much better throughput:
`[09:17:12][W][component:204]: Component ili9xxx.display took a long time for an operation (0.33 s).`

There is still room for improvements but I guess this topic can be marked as resolved with the new SPI component code.

closed 2023-09-25 by fsiek