esphome/issues #4907: ESP32 C3: ble_tracker "breaks" device

pascaltippelt, 2023-09-25:
### The problem

I try to use cheap ESP32 C3 Mini-Boards for BLE presence. Board: https://de.aliexpress.com/item/1005005757810089.html

This works:

```yaml
esphome:
  name: test-c3
  friendly_name: TEST-C3

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "JBLol1aFnZRq9oWKYiw8qocyMdpEJZxK68CM71XeSYI="

ota:
  password: "cbd2f7da545bb77cec8295fb01642a5a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-C3 Fallback Hotspot"
    password: "A9ckFmL5GxIv"

captive_portal:
```

Resulting in this log over COM:

```bash
ESP-ROM:esp32c3-api1-20210207
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4205dcf8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
[   240][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   271][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[   271][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[  6087][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 4
[  6088][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[  6803][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: NOPE, BSSID: aa:aa:aa:aa:aa:aa, Reason: 202
[  6804][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  6811][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 202 - AUTH_FAIL
[  6817][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running
[  6824][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  6887][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: SSID: NOPE, BSSID: aa:aa:aa:aa:aa:aa, Channel: 1, Auth: WPA2_PSK
[  6894][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  6914][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: NOPE, BSSID: aa:aa:aa:aa:aa:aa Reason: 8
[  6924][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  6931][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[  6987][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: NOPE, BSSID: aa:aa:aa:aa:aa:aa, Channel: 1, Auth: WPA2_PSK
[  6989][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  7112][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:192.168.178.103
[  7112][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  7115][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.178.103, MASK: 255.255.255.0, GW: 192.168.178.1
[  8236][V][WiFiGeneric.cpp:454] _arduino_event_cb(): IF[0] Got IPv6: IP Index: 0, Zone: 2, fe80:0000:0000:0000:nope:nope:nope
[  8237][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 8 - STA_GOT_IP6
```

...but as soon as I add:

```yaml
esp32_ble_tracker:
```

The board stops connecting to wifi, somestimes doesn't even boot. Log:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038028c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
[   567][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   569][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[   574][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[  3951][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 4
[ 30585][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 38962][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 4
[ 60543][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 61144][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[ 61145][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[ 61149][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
[ 91156][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 91160][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[121170][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[121174][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: NOPE, BSSID: aa:aa:aa:aa:aa:aa, Channel: 1, Auth: WPA2_PSK
[156184][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[156192][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: NOPE, BSSID: aa:aa:aa:aa:aa:aa, Reason: 8
[156420][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[156427][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 8 - ASSOC_LEAVE
```
(Nothing happens after that for minutes, no connection to Home Assistant, WiFi ist NOT connected, says AP)

So I thought: Maybe esp-idf... Result: Bootloop.

```bash
ESP-ROM:esp32c3-api1-20210207
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd5810,len:0x1684
ets_loader.c 78 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd5810,len:0x1684
ets_loader.c 78 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd5810,len:0x1684
ets_loader.c 78 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd5810,len:0x1684
ets_loader.c 78 
```

### Which version of ESPHome has the issue?

 2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2023.9.3

### What platform are you using?

ESP32

### Board

ESP32-C3 "Super Mini"

### Component causing the issue

esp32_ble_tracker:

### Example YAML snippet

```yaml
yaml
esphome:
  name: test-c3
  friendly_name: TEST-C3

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "JBLol1aFnZRq9oWKYiw8qocyMdpEJZxK68CM71XeSYI="

ota:
  password: "cbd2f7da545bb77cec8295fb01642a5a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-C3 Fallback Hotspot"
    password: "A9ckFmL5GxIv"

captive_portal:
esp32_ble_tracker:
```

### Anything in the logs that might be useful for us?

```txt
[see above]
```

### Additional information

[see above]

ssieb, 2023-09-25:
When you switched to esp-idf, the flash mode was changed for some reason.  You'll need to add the platformio option to keep it at DIO.

xdecock, 2023-09-26:
Merge this in your esp-idf yaml:
```yaml
esphome:
  platformio_options:
    board_build.flash_mode: dio

logger:
  hardware_uart: USB_SERIAL_JTAG
```

Those will allow the board to boot and the log to be streamed via usb

AlecTBM, 2023-10-09:
esp32_ble_tracker have made all of my Nodemcu-32s go in a bootloop since 2023.07 or 2023.08, and have been unstable (ESP rebooting within 1 or 2 hours) since 2023.04 or 2023.05. This is on devices that was running fine for 1,5 year before. So I have not upgraded any devices that use esp32_ble_tracker past 2023.03. I am not sure if this is related as they are not ESP32-C3, but just a note.

pascaltippelt, 2023-10-10:
Hi @AlecTBM . I tried to reproduce my problem with an other esp module. It seems, as I had a defective module. You are right, it works as esprected. I'm sorry I disturbed you.

closed 2023-10-10 by pascaltippelt