esphome/issues #4908: Checksum error in received SML data with Landis E320

christof62, 2023-09-25:
### The problem

SML component not working with Landis E320.
Checksum error in received SML data.

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9

### What platform are you using?

ESP8266

### Board

d1_mini_pro

### Component causing the issue

SML

### Example YAML snippet

```yaml
esphome:
  name: smartmeter
  friendly_name: Smartmeter

esp8266:
  board: d1_mini_pro

# Enable logging
logger:
  level: DEBUG
  hardware_uart: UART1
  
# Enable Home Assistant API
api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

    
uart:      
  - id: uart_2
    rx_pin: RX
    baud_rate: 9600
    data_bits: 8
    parity: NONE
    stop_bits: 1 
    rx_buffer_size: 1024
    debug:
      direction: BOTH
      dummy_receiver: false
      after:
        delimiter: "\n"
      sequence:
        - lambda: UARTDebug::log_hex(direction, bytes, ' ');        

sml:  
  id: pv_sm
  uart_id: uart_2

sensor:
  - platform: sml
    name: "PV Roof Provided Energy"
    sml_id: pv_sm
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 1
    state_class: total_increasing
    device_class: energy
    filters:
      - multiply: 0.0001
```

### Anything in the logs that might be useful for us?

```txt
22:52:19][I][app:102]: ESPHome version 2023.8.3 compiled on Sep 25 2023, 22:51:27
[22:52:19][C][wifi:543]: WiFi:
[22:52:19][C][wifi:379]:   Local MAC: 5C:CF:7F:39:DC:E2
[22:52:19][C][wifi:380]:   SSID: [redacted]
[22:52:19][C][wifi:381]:   IP Address: 192.168.178.40
[22:52:19][C][wifi:382]:   BSSID: [redacted]
[22:52:19][C][wifi:384]:   Hostname: 'smartmeter'
[22:52:19][C][wifi:386]:   Signal strength: -70 dB ▂▄▆█
[22:52:19][C][wifi:390]:   Channel: 6
[22:52:19][C][wifi:391]:   Subnet: 255.255.255.0
[22:52:19][C][wifi:392]:   Gateway: 192.168.178.1
[22:52:19][C][wifi:393]:   DNS1: 192.168.178.1
[22:52:19][C][wifi:394]:   DNS2: 0.0.0.0
[22:52:19][C][logger:301]: Logger:
[22:52:19][C][logger:302]:   Level: DEBUG
[22:52:19][C][logger:303]:   Log Baud Rate: 115200
[22:52:19][C][logger:305]:   Hardware UART: UART1
[22:52:19][C][uart.arduino_esp8266:102]: UART Bus:
[22:52:19][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[22:52:19][C][uart.arduino_esp8266:106]:   RX Buffer Size: 1024
[22:52:19][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[22:52:19][C][uart.arduino_esp8266:109]:   Data Bits: 8
[22:52:19][C][uart.arduino_esp8266:110]:   Parity: NONE
[22:52:19][C][uart.arduino_esp8266:111]:   Stop bits: 1
[22:52:19][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[22:52:19][C][sml:092]: SML:
[22:52:19][C][sml_sensor:033]: SML 'PV Roof Provided Energy'
[22:52:19][C][sml_sensor:033]:   Device Class: 'energy'
[22:52:19][C][sml_sensor:033]:   State Class: 'total_increasing'
[22:52:19][C][sml_sensor:033]:   Unit of Measurement: 'kWh'
[22:52:19][C][sml_sensor:033]:   Accuracy Decimals: 1
[22:52:19][C][sml_sensor:037]:   OBIS Code: 1-0:1.8.0
[22:52:19][C][mdns:112]: mDNS:
[22:52:19][C][mdns:113]:   Hostname: smartmeter
[22:52:19][C][ota:093]: Over-The-Air Updates:
[22:52:19][C][ota:094]:   Address: smartmeter.local:8266
[22:52:19][C][ota:097]:   Using Password.
[22:52:19][C][api:138]: API Server:
[22:52:19][C][api:139]:   Address: smartmeter.local:6053
[22:52:19][C][api:141]:   Using noise encryption: YES
[22:52:20][D][uart_debug:114]: <<< 01 4C 47 5A 02 08 00 FF 65 00 1C 01 04 00 52 03 69 00 00 00 00 00 00 01 41 84 44 16 62 00 62 00 02 1B 1B 1B 1B 1A 01 0E E0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1B 1B 1B 1B 01 01 01 01 76 63 01 01 76 01 07 FF FF FF 00 01 4C 47 5A 00 03 CF FE FD 02 82 00 76 05 00 84 44 18 02 FF FF FF FF FF 0B 0A
[22:52:20][D][uart_debug:114]: <<< 01 0C 62 0A
[22:52:20][D][uart_debug:114]: <<< FF FF 72 62 01 65 00 01 01 01 01 01 01 04 4C 47 00 01 01 01 01 0B 0A
[22:52:20][W][component:204]: Component sml took a long time for an operation (0.07 s).
[22:52:20][W][component:205]: Components should block for at most 20-30ms.
[22:52:20][W][sml:116]: Checksum error in received SML data.
[22:52:21][D][uart_debug:114]: <<< 01 4C 47 00 02 08 00 FF 65 00 1C 01 04 24 35 69 00 00 00 00 00 00 15 00 84 44 19 62 00 62 00 00 1B 1B 1B 1B 1A 01 13 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1B 1B 1B 1B 01 01 01 01 06 AE AC 01 76 01 07 FF FF FF 00 28 4C 47 5A 00 03 CF FE FD 00 09 01 01 76 05 00 84 44 1B FF FF FF FF FF FF 0B 0A
[22:52:21][D][uart_debug:114]: <<< 01 00 62 0A
[22:52:21][D][uart_debug:114]: <<< FF FF 72 62 01 65 00 A6 01 01 01 01 01 04 4C 07 01 01 01 01 0B 0A
[22:52:21][W][component:204]: Component sml took a long time for an operation (0.07 s).
[22:52:21][W][component:205]: Components should block for at most 20-30ms.
[22:52:21][W][sml:116]: Checksum error in received SML data.
[22:52:22][D][uart_debug:114]: <<< 01 4C 07 00 02 08 00 FF 65 00 1C 01 79 52 03 69 00 00 00 00 00 00 76 05 00 84 44 1C 62 00 62 00 00 1B 1B 1B 1B 1A 01 1A FE 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1B 1B 1B 1B 01 01 01 01 72 63 01 01 76 01 07 FF FF 00 0A
[22:52:22][D][uart_debug:114]: <<< 01 4C 47 5A 00 03 CF FE 00 AC AC 0F 76 05 00 84 44 1E D0 FF FF FF FF FF 0B 0A
[22:52:22][D][uart_debug:114]: <<< 01 00 62 0A
[22:52:22][D][uart_debug:114]: <<< FF FF 72 62 01 65 C9 01 01 01 01 01 01 04 4C FF 01 01 01 01 0B 0A
[22:52:22][W][sml:116]: Checksum error in received SML data.
```

macmeck, 2023-10-05:
I have the same issue here. Seems to be related to #4805 which should have been fixed in 2023.8.1

Funny enough, the sensor seems to pick up a correct initial value as I can see the energy value in HomeAssistant when the sensor starts up. But this value never seems to get updated due to the checksum errors.

macmeck, 2023-10-10:
@christof62 , is your sensor powered by 3.3V or 5V?

macmeck, 2023-10-15:
covering the IR sender seems to do the trick for me

christof62, 2023-10-18:
> @christof62 , is your sensor powered by 3.3V or 5V?

It's powered by 3.3V.

christof62, 2023-10-18:
> covering the IR sender seems to do the trick for me

Thanks @macmeck, I will try this.