esphome/issues #4909: ESP32-C6 bluetooth not working

cjc7373, 2023-09-26:
### The problem

When I tried to enable `esp32_ble_tracker` in my esp32-c6-devkitc-1 board, I encountered some compile error, see the logs below.

### Which version of ESPHome has the issue?

2023.9.0b2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
# snippet of the full config
esp32:
  board: esp32-c6-devkitc-1
  # board: esp32dev
  framework:
    type: esp-idf
    platform_version: 6.4.0
    version: 5.1.1
  variant: esp32c6

esp32_ble_tracker:
  scan_parameters:
    active: true
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.9.0b2
INFO Reading configuration /config/esphome/esp32-c6.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.2.0
INFO Compiling app...
Processing esp32-c6 (board: esp32-c6-devkitc-1; framework: espidf; platform: platformio/espressif32@6.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C6 160MHz, 512KB RAM, 8MB Flash
 - framework-espidf @ 3.50101.230828 (5.1.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling .pioenvs/esp32-c6/src/esphome/components/api/api_connection.o
Compiling .pioenvs/esp32-c6/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/esp32-c6/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/esp32-c6/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/esp32-c6/src/esphome/components/api/api_server.o
Compiling .pioenvs/esp32-c6/src/esphome/components/api/list_entities.o
Compiling .pioenvs/esp32-c6/src/esphome/components/api/proto.o
Compiling .pioenvs/esp32-c6/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/esp32-c6/src/esphome/components/api/user_services.o
Compiling .pioenvs/esp32-c6/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/esp32-c6/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/esp32-c6/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/esp32-c6/src/esphome/components/bluetooth_proxy/bluetooth_connection.o
Compiling .pioenvs/esp32-c6/src/esphome/components/bluetooth_proxy/bluetooth_proxy.o
Compiling .pioenvs/esp32-c6/src/esphome/components/captive_portal/captive_portal.o
Compiling .pioenvs/esp32-c6/src/esphome/components/esp32/core.o
Compiling .pioenvs/esp32-c6/src/esphome/components/esp32/gpio.o
Compiling .pioenvs/esp32-c6/src/esphome/components/esp32/preferences.o
Compiling .pioenvs/esp32-c6/src/esphome/components/esp32_ble/ble.o
Compiling .pioenvs/esp32-c6/src/esphome/components/esp32_ble/ble_advertising.o
In file included from src/esphome/components/esp32_ble/ble.cpp:7:
src/esphome/components/esp32_ble/ble.cpp: In member function 'bool esphome::esp32_ble::ESP32BLE::ble_setup_()':
/data/cache/platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:263:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_drop_mode' [-Wmissing-field-initializers]
  263 | }
      | ^
src/esphome/components/esp32_ble/ble.cpp:62:40: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
   62 |       esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:263:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_low_tx_pwr' [-Wmissing-field-initializers]
  263 | }
      | ^
src/esphome/components/esp32_ble/ble.cpp:62:40: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
   62 |       esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:258:58: error: narrowing conversion of 'efuse_hal_chip_revision()' from 'uint32_t' {aka 'long unsigned int'} to 'uint8_t' {aka 'unsigned char'} [-Werror=narrowing]
  258 |     .version_num                = efuse_hal_chip_revision(),                            \
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~^~
src/esphome/components/esp32_ble/ble.cpp:62:40: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
   62 |       esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors
*** [.pioenvs/esp32-c6/src/esphome/components/esp32_ble/ble.o] Error 1
========================= [FAILED] Took 54.26 seconds =========================
```

HeMan, 2023-09-27:
One of those messages (`/data/cache/platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:258:58: error: narrowing conversion of 'efuse_hal_chip_revision()' from 'uint32_t' {aka 'long unsigned int'} to 'uint8_t' {aka 'unsigned char'} [-Werror=narrowing]`) seem to be an error in ESP-IDF for esp32c6. I'll create a PR for it.

HeMan, 2023-09-27:
It looks like all of them is due to bugs in ESP-IDF.

cjc7373, 2023-09-30:
Thanks, I applied your patch locally and the compile error has gone.

However, the bluetooth is still not working. And when I set log level to debug, I see the following lines:
```bash
[14:36:18][C][esp32_ble:243]: ESP32 BLE: bluetooth stack is not enabled
[14:36:18][C][esp32_ble_tracker:617]: BLE Tracker:
[14:36:18][C][esp32_ble_tracker:618]:   Scan Duration: 300 s
[14:36:18][C][esp32_ble_tracker:619]:   Scan Interval: 320.0 ms
[14:36:18][C][esp32_ble_tracker:620]:   Scan Window: 30.0 ms
[14:36:18][C][esp32_ble_tracker:621]:   Scan Type: ACTIVE
[14:36:18][C][esp32_ble_tracker:622]:   Continuous Scanning: True
```
Any clue why this happens?

luomanruo, 2023-11-21:
Thank you for bringing this to our attention, and we appreciate your patience. We have verified the issue you reported and provided a temporary solution. For this problem, we recommend modifying the esp_bt.h file by casting the return value of efuse_hal_chip_revision() to uint8_t as follows, matching the version_num structure member.
`.version_num                = (uint8_t)efuse_hal_chip_revision(), `
This is a temporary workaround, and we plan to address this issue in future releases.

stefan-schweiger, 2024-06-17:
@luomanruo can you be more specific what you mean by modifying the esp_bt.h file? Also maybe any update on a potential permanent fix for this?

mx5gr, 2024-06-29:
Issue still persists with ESP-IDF 5.2.2, PlatformIO 6.7.0 and ESPHome 2024.6.4:

Compiling .pioenvs/sens-1st-bedroom-ac/bootloader_support/src/esp_image_format.c.o
In file included from src/esphome.h:26,
from src/main.cpp:3:
src/esphome/components/esp32_ble/const_esp32c6.h:62:1: warning: missing initializer for member ‘esp_bt_controller_config_t::cca_drop_mode’ [-Wmissing-field-initializers]
62 | };
| ^
src/esphome/components/esp32_ble/const_esp32c6.h:62:1: warning: missing initializer for member ‘esp_bt_controller_config_t::cca_low_tx_pwr’ [-Wmissing-field-initializers]
src/esphome/components/esp32_ble/const_esp32c6.h:62:1: warning: missing initializer for member ‘esp_bt_controller_config_t::csa2_select’ [-Wmissing-field-initializers]

rvalitov, 2024-10-10:
My ESP32-C6 works great with Bluetooth with the following config:

```yaml
# Example config.yaml
esphome:
  name: bt-proxy
  friendly_name: Bluetooth Proxy
  platformio_options:
    board_build.flash_mode: dio 

esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  flash_size: 8MB
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_OPENTHREAD_ENABLED: n
      CONFIG_ENABLE_WIFI_STATION: y
      CONFIG_USE_MINIMAL_MDNS: y

# Enable Home Assistant API
api:
  encryption:
    key: "secret"
  on_client_connected:
    - esp32_ble_tracker.start_scan:
        continuous: true
  on_client_disconnected:
    - esp32_ble_tracker.stop_scan:

wifi:
  ssid: "secret"
  password: "secret"

esp32_ble_tracker:
  id: ble_tracker
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
    continuous: false

bluetooth_proxy:
  active: true 
```

Some notes:
- using the latest esp-idf and platform versions is important
- to avoid compilation errors check this PR https://github.com/esphome/esphome/pull/7580
- the `continuous` mode of `esp32_ble_tracker` at start-up may lead to Wi-Fi connectivity issues. I didn't face that personally, but it happened for other ESP32 chips. So I prefer to use a solution to enable BLE only after the Wi-Fi is ready and HA is connected.

mx5gr, 2024-10-11:
@rvalitov with your settings and framework versions, I get the following error, using a Xiao ESP32-C3 (4MB, not 8MB flash):

Compiling .pioenvs/sens-1st-bedroom-ac/src/esphome/components/esp32_ble/ble_uuid.cpp.o
In file included from src/esphome/components/esp32_ble/ble.cpp:6:
src/esphome/components/esp32_ble/const_esp32c6.h:62:1: error: 'const esp_bt_controller_config_t' has no non-static data member named 'enable_uart_hci'
   62 | };
      | ^
Compiling .pioenvs/sens-1st-bedroom-ac/src/esphome/components/esp32_ble_client/ble_characteristic.cpp.o
Compiling .pioenvs/sens-1st-bedroom-ac/src/esphome/components/esp32_ble_client/ble_client_base.cpp.o
Compiling .pioenvs/sens-1st-bedroom-ac/src/esphome/components/esp32_ble_client/ble_service.cpp.o
*** [.pioenvs/sens-1st-bedroom-ac/src/esphome/components/esp32_ble/ble.cpp.o] Error 1

rvalitov, 2024-10-11:
@mx5gr My config is for ESP32-C6 only, it will not work for C3.

mx5gr, 2024-10-11:
> @mx5gr My config is for ESP32-C6 only, it will not work for C3.

My mistake.. I wanted to write C6. ESP32-C6 is the correct model I wanted to write.

I also saw a push you made to esphome dev branch, I hope it makes it to the stable... :)

rvalitov, 2024-10-11:
Yes, the PR fixes the compilation errors. Either you need to wait for it to be merged, or apply the code modifications locally.

closed 2024-10-17 by jesserockz

pyrker, 2024-12-18:
Trying to compile for a ESP32-C6 with the latest esphome 2024.12.0 as the chip is now supported with esp-idf 5.1.5. Unfortunatly the BLE issue reappears:

`INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/esp32-c6-kloluft.yaml...
WARNING GPIO8 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing esp32-c6-kloluft (board: esp32-c6-devkitm-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
Compiling .pioenvs/esp32-c6-kloluft/src/esphome/components/esp32_ble/ble.cpp.o
Compiling .pioenvs/esp32-c6-kloluft/src/esphome/components/fan/fan.cpp.o
Compiling .pioenvs/esp32-c6-kloluft/src/esphome/components/fan/fan_state.cpp.o
Compiling .pioenvs/esp32-c6-kloluft/src/esphome/components/gpio/output/gpio_binary_output.cpp.o
Compiling .pioenvs/esp32-c6-kloluft/src/esphome/components/homeassistant/binary_sensor/homeassistant_binary_sensor.cpp.o
In file included from src/esphome/components/esp32_ble/ble.cpp:6:
src/esphome/components/esp32_ble/const_esp32c6.h:69:1: error: 'const esp_bt_controller_config_t' has no non-static data member named 'enable_uart_hci'
   69 | };
      | ^
Compiling .pioenvs/esp32-c6-kloluft/src/esphome/components/homeassistant/switch/homeassistant_switch.cpp.o
Compiling .pioenvs/esp32-c6-kloluft/src/esphome/components/homeassistant/text_sensor/homeassistant_text_sensor.cpp.o
*** [.pioenvs/esp32-c6-kloluft/src/esphome/components/esp32_ble/ble.cpp.o] Error 1
========================= [FAILED] Took 15.03 seconds =========================`

cjc7373, 2024-12-18:
@pyrker try use esp-idf 5.3.1

pyrker, 2024-12-18:
> @pyrker try esp-idf 5.3.1

That's what I've been using before and can confirm it worked until esphome update 2012.12.0. From now on it seems to require esp-idf 5.1.5 which would be fine for me.. if BLE worked again.

rvalitov, 2024-12-18:
The ESPHome does not forbid you to use version 5.3.1. So why not to use this version then?

pyrker, 2024-12-18:
> The ESPHome does not forbid you to use version 5.3.1. So why not to use this version then?

actually I was using 5.1.4 as I2C did not work on the C6 with the 5.3.1. And from the latest esphome 2024.12.0 the 5.1.4 is not compiling anymore. Anyways, the only thing not working with the latest 5.1.5 is BLE.

marcor2196, 2025-01-16:
@pyrker Hi, have you resolved in some ways?

This is my conf:

```
esp32:
  board: esp32-c6-devkitc-1
  flash_size: 16MB
  variant: esp32c6
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0

esp32_ble_tracker:
  id: ble_tracker
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
    continuous: false

bluetooth_proxy:
  active: true 
```

During the compile i see only this warning:

```
Compiling .pioenvs/esp32-bluetooth-proxy/src/esphome/components/esp32_ble/ble_advertising.cpp.o
In file included from src/esphome/components/esp32_ble/ble.cpp:6:
src/esphome/components/esp32_ble/const_esp32c6.h:69:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_drop_mode' [-Wmissing-field-initializers]
   69 | };
      | ^
src/esphome/components/esp32_ble/const_esp32c6.h:69:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_low_tx_pwr' [-Wmissing-field-initializers]
```