esphome/issues #4910: Error in Waveshare 7.5v2 after 2023.9.0 upgrade

primeroz, 2023-09-27:
### The problem

after upgrading to  2023.9.0  , so i could get https://github.com/esphome/esphome/pull/5283 , i am seeing some errors in the logs of my display

```
[09:01:00][D][main:102]: Sensor data updated and activity in home detected: Refreshing display...
[09:01:00][I][waveshare_epaper:1628]: Power on the display and hat
[09:01:10][E][waveshare_epaper:1573]: Timeout while displaying image!
[09:01:21][E][waveshare_epaper:1573]: Timeout while displaying image!
```

The display referesh **works** so i just thought i would bring this up to see if is an issue . pinging @phoenixswiss as the auehtor of that PR 

### Which version of ESPHome has the issue?

 2023.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.2

### What platform are you using?

ESP32

phoenixswiss, 2023-09-27:
Hi @primeroz 

Unfortunately, the documentation hasn't been approved yet. You need to invert your busy pin to get the correct busy status:

https://github.com/esphome/esphome/pull/5283#:~:text=Fixed%20the%20wait_until_idle_()%20function%20to%20work%20properly%20with%20Waveshare%207.5v2%20screens%20(Warning%3A%20You%20also%20need%20to%20invert%20the%20busy%20pin%20in%20the%20yaml%20config!)

closed 2023-09-27 by primeroz

phoenixswiss, 2023-09-27:
Great to hear. By the way, the documentation was approved just a minute ago:

https://esphome.io/components/display/waveshare_epaper.html?highlight=waveshare#:~:text=Warning,in%20the%20config.