esphome/issues #4911: Stop working MAX7219 7-Segment Display after update 2023.09 (HA can't connect to API)

DieMetRik, 2023-09-27:
### The problem

After update from ESPHome 2023.08.3
One of my device goes down

Device Unknow in HA at now
I can update ESP throw OTA wo any problem.
Then try to log wirelessly:
````
WARNING Can't connect to ESPHome API for 192.168.1.116: Error connecting to ('192.168.1.116', 6053): [Errno 111] Connect call failed ('192.168.1.116', 6053) (SocketAPIError)
````
After small investigation:
If comment this section:
````
display:
    platform: max7219
    id: max7219_display
    cs_pin: D8
    num_chips: 1
    intensity: 15
    update_interval: 1s
    lambda: |-
      it.print(0, id(display_data).state.c_str());
````
Afterthat ESP Api work good

### Which version of ESPHome has the issue?

2023.09

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.09.3

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

max7219

### Example YAML snippet

```yaml
substitutions:
  device_name: temperature-display
  device_ip: 192.168.1.116
  platform: ESP8266
  board: d1_mini
  device_description: "Дисплей температуры и ppm в гостиной"

esphome:
  name: ${device_name}
  platform: ${platform}
  board: ${board}
  comment: ${device_description}

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "J2+3oqEO6eDX0EleMvkLcTooGO0x2MaNFk+C/nFkRbU="
# Password for OTA update

ota:
  password: !secret ota_password
# Wifi settings
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: ${device_ip}
    gateway: 192.168.1.1
    subnet: 255.255.255.0
# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name} ap
    password: !secret ap_password

captive_portal:
spi:
  clk_pin: D5
  mosi_pin: D7

display:
    platform: max7219
    id: max7219_display
    cs_pin: D8
    num_chips: 1
    intensity: 15
    update_interval: 1s
    lambda: |-
      it.print(0, id(display_data).state.c_str());

# # SENSOR AREA
text_sensor:
  - platform: homeassistant
    id: display_data
    entity_id: sensor.livingroom_screen_data
    internal: true
    disabled_by_default: true
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.9.0
INFO Reading configuration /config/esphome/temperature-display.yaml...
INFO Starting log output from 192.168.1.116 using esphome API
WARNING Can't connect to ESPHome API for 192.168.1.116: Error connecting to ('192.168.1.116', 6053): [Errno 111] Connect call failed ('192.168.1.116', 6053) (SocketAPIError)
INFO Trying to connect to 192.168.1.116 in the background
```

jscotka, 2023-09-27:
With the mentioned version I've met same issue, with serial line there is traceback:
```
[13:42:56]ELF file SHA256: 0000000000000000
[13:42:56]
[13:42:56]Rebooting...
[13:42:56]ets Jun  8 2016 00:22:57
[13:42:56]
[13:42:56]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:42:56]configsip: 0, SPIWP:0xee
[13:42:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:42:56]mode:DIO, clock div:2
[13:42:56]load:0x3fff0030,len:1184
[13:42:56]load:0x40078000,len:13132
[13:42:56]load:0x40080400,len:3036
[13:42:56]entry 0x400805e4
[13:42:57][     6][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[13:42:57][I][logger:268]: Log initialized
[13:42:57][VV][esp32.preferences:071]: nvs_get_blob: key: 233825507, len: 4
[13:42:57][C][ota:473]: There have been 9 suspected unsuccessful boot attempts.
[13:42:57][VV][esp32.preferences:040]: s_pending_save: key: 233825507, len: 4
[13:42:57][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:42:57][VV][esp32.preferences:123]: Checking if NVS data 233825507 has changed
[13:42:57][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[13:42:57][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[13:42:57][E][spi:078]: SPIDevice not initialised - did you call spi_setup()?
[13:42:57]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[13:42:57]
[13:42:57]Core  1 register dump:
[13:42:57]PC      : 0x4017614f  PS      : 0x00060c30  A0      : 0x800db84c  A1      : 0x3ffb26e0  
WARNING Decoded 0x4017614f: esphome::spi::SPIDelegate::end_transaction() at /home/jscotka/.esphome/build/displey-obyvak/src/esphome/components/spi/spi.h:189
[13:42:57]A2      : 0x3ffba708  A3      : 0x0000000a  A4      : 0x00000000  A5      : 0x00000000  
[13:42:57]A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800db76a  A9      : 0x3ffb26b0  
[13:42:57]A10     : 0x00000000  A11     : 0x00000001  A12     : 0x0000004e  A13     : 0x3f403e08  
[13:42:57]A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
[13:42:57]EXCVADDR: 0x00000000  LBEG    : 0x4008b2ed  LEND    : 0x4008b2fd  LCOUNT  : 0xffffffec  
[13:42:57]
[13:42:57]
[13:42:57]Backtrace:0x4017614c:0x3ffb26e00x400db849:0x3ffb2700 0x400db8dd:0x3ffb2720 0x400e4432:0x3ffb2740 0x400f182e:0x3ffb2820 
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4017614c: esphome::spi::SPIDelegate::end_transaction() at /home/jscotka/.esphome/build/displey-obyvak/src/esphome/components/spi/spi.h:189
WARNING Decoded 0x400db849: esphome::spi::SPIDevice<(esphome::spi::SPIBitOrder)1, (esphome::spi::SPIClockPolarity)0, (esphome::spi::SPIClockPhase)0, (esphome::spi::SPIDataRate)1000000>::disable() at /home/jscotka/.esphome/build/displey-obyvak/src/esphome/components/spi/spi.h:424
 (inlined by) esphome::max7219::MAX7219Component::send_to_all_(unsigned char, unsigned char) at /home/jscotka/.esphome/build/displey-obyvak/src/esphome/components/max7219/max7219.cpp:164
WARNING Decoded 0x400db8dd: esphome::max7219::MAX7219Component::set_intensity(unsigned char) at /home/jscotka/.esphome/build/displey-obyvak/src/esphome/components/max7219/max7219.cpp:222
WARNING Decoded 0x400e4432: setup() at /home/jscotka/.esphome/build/displey-obyvak/src/main.cpp:260
WARNING Decoded 0x400f182e: loopTask(void*) at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

RoelHermus, 2023-09-27:
I've the same problem with MAX7219, the error I get:

[17:27:55][E][spi:078]: SPIDevice not initialised - did you call spi_setup()?

zwgmy, 2023-09-28:
also, cann't connected to HA.  ota is nomal.
it displayed: Error connecting to ('192.168.2.176', 6053): [Errno 111] Connect call failed ('192.168.2.176', 6053) (SocketAPIError)

emilianogetino, 2023-09-28:
hi,
same problem here: 

WARNING Can't connect to ESPHome API for invernadero.local: Error connecting to ('172.26.254.2', 6053): [Errno 111] Connect call failed ('172.26.254.2', 6053) (SocketAPIError)
INFO Trying to connect to invernadero.local in the background

my code:

`display:
  - platform: max7219
    cs_pin: ${gpio_cs_pin}
    num_chips: 1
    lambda: |-
      it.printf(0,"        ");       
      if (id(puerta_${devicename}).state) {
        it.printf(5, "H.%.0f", id(humedad_${devicename}).state);
        it.printf(0, "%.1f~", id(temperatura_${devicename}).state);
      } `

If I remove this code everything works fine.

kb2ear, 2023-09-29:
me too, same issue on nodemcu esp8266

Adrian57, 2023-09-29:
I also have this same problem. A working setup has stopped working since 2023.09.0 installed. Still same problem at 2023.09.2.

I get no log returned from the device although I can still flash it ota.

I get 10 flashes on the D1-Mini onboard LED on startup and periodically after

Grumpfbe2, 2023-09-30:
Le dot.
Metoo...
Same issue, same "solution". FYI, it seems Max7219digit does not prevent the API to connect. I did put just that to switch off the 7SEG temporarily as this ESP has a 20X4 LCD for other purposes that works fine.

My code :

`substitutions:
  boardname : esp01_1m
  outsideprobetemp : sensor.tunex_6102_temp
  outsideprobehum : sensor.tunex_6102_hum
  insideprobetemp : sensor.kitchen_probe_temperature
  insideprobehum : sensor.kitchen_probe_humidity
  windspeedprobe: sensor.openweathermap_forecast_wind_speed
  winddirectionprobe: sensor.narvi_wind_sensor
  ProbePressTrends:  sensor.narvi_cups3_local_pressure_tendency
  uvprobe: sensor.narvi_cups3_local_uv_index
  Precip: sensor.narvi_cups3_local_precipitation
  Forecast: weather.narvi_cups3_local

  SevSEGCLKPIN : GPIO12
  SevSEGMOSIPIN: GPIO14
  SevSEGCSPIN: GPIO13
  I2CSDAPIN: GPIO04
  I2CSCLPIN: GPIO05
  frequency: 400kHz

esphome:
  name: bombclock
  friendly_name: Bombclock

esp8266:
  board: $boardname

# Enable logging
logger:

# Enable Home Assistant API
api:
#  encryption:
#    key: "WNQZFOEoE9yj5EyJSqL/oKklFZSZ2mzE/q5VxoDKGq8="

ota:
  password: "19bfe704a952d84c946e6ac188bbd75f"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bombclock Fallback Hotspot"
    password: "Sex1VFCDGzYT"

captive_portal:
time:
  - platform: sntp
    id: esptime
    update_interval: 60s

spi:
  clk_pin: $SevSEGCLKPIN
  mosi_pin: $SevSEGMOSIPIN
i2c:
  sda: $I2CSDAPIN
  scl: $I2CSCLPIN

sensor:
  - platform: homeassistant
    id : outtemp
    entity_id: $outsideprobetemp

  - platform: homeassistant
    id : outhum
    entity_id: $outsideprobehum

  - platform: homeassistant
    id : intemp
    entity_id: $insideprobetemp

  - platform: homeassistant
    id : inhum
    entity_id: $insideprobehum

  - platform: homeassistant
    id : Windspeed
    entity_id: $windspeedprobe

  - platform: homeassistant
    id : UVindex
    entity_id: $uvprobe

  - platform: homeassistant
    id : Rainfall
    entity_id: $Precip

text_sensor:

  - platform: homeassistant
    id : PressTrends
    entity_id: $ProbePressTrends

  - platform: homeassistant
    id : WindDirection
    entity_id: $winddirectionprobe

  - platform: homeassistant
    id : WFC
    entity_id: $Forecast

display:
#  - platform: max7219
#    cs_pin: $SevSEGCSPIN
#    num_chips: 1
#    update_interval: 500ms
#    lambda: |-
#      static int i = 0;
#      i++;
#
#      if ((id(PressTrends).state == "falling") && ((i % 2) == 0))
#          it.strftime("%H-%M-%S.", id(esptime).now());
#      else
#          it.strftime("%H-%M-%S", id(esptime).now());
  
  - platform: lcd_pcf8574
    user_characters:
      - position: 3
        data: 
          - 0x04
          - 0x0E
          - 0x1F
          - 0x04
          - 0x04
          - 0x1F
          - 0x0E
          - 0x04
      - position: 4
        data: 
          - 0x08
          - 0x14
          - 0x08
          - 0x03
          - 0x04
          - 0x04
          - 0x04
          - 0x03
      - position: 5
        data: 
          - 0x08
          - 0x14
          - 0x08
          - 0x05
          - 0x05
          - 0x07
          - 0x05
          - 0x05
      - position: 6
        data: 
         - 0x14
         - 0x18
         - 0x14
         - 0x00
         - 0x00
         - 0x04
         - 0x07
         - 0x05
          
      - position: 7
        data:
         - 0x04
         - 0x04
         - 0x0E
         - 0x0E
         - 0x17
         - 0x17
         - 0x1F
         - 0x0E
    dimensions: 20x4
    address: 0x27
    update_interval: 1000ms
    lambda: |-
         it.strftime(0,0,"%A %d/%m", id(esptime).now()); 
         it.printf(17,0,"\xCE%02.0f", id(UVindex).state);
         it.printf(0,1, "O:%02.1f\x04 ",id(outtemp).state);
         it.printf(8,1, "%.0f\x05", id(outhum).state);
         it.printf(15,1, "%\x07%02.1f", id(Rainfall).state);
         it.printf(0,2, "I:%02.1f\x04 ",id(intemp).state);
         it.printf(8,2, "%.0f\x05", id(inhum).state);
         it.printf(12,2, "%02.1f\x06", id(Windspeed).state);
         it.printf(17,2, "%s", id(WindDirection).state.c_str());
         it.printf(0,3, "\xAB%s", id(WFC).state.c_str());
         it.printf(13,3, "%s", id(PressTrends).state.c_str());`

DieMetRik, 2023-10-03:
After update to 9.2 all work. Thanks

closed 2023-10-03 by DieMetRik

Grumpfbe2, 2023-10-03:
Same here, thanks !