esphome/issues #4912: Dashboard no longer displays "Update Available"

3ative, 2023-09-27:
### The problem

Post ESPHome 2023.9.0 (Add-on) Dashboard no longer displays blue "Update Available" on devices. 

Yet, HA Settings Icon displays number of devices with updates though.

I.E.
[image]

### Which version of ESPHome has the issue?

2023.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP8266

### Board

N/A

### Component causing the issue

N/A

### Anything in the logs that might be useful for us?

```txt
None
```

3ative, 2023-09-27:
Forgot to add:
Tried HA Restart and Supervisor (VM) Reboot

robgough1970, 2023-09-27:
same here - looking at device log - shows version 2023.9.0 although devices have not been updated

3ative, 2023-09-27:
Additional... If it helps:
[image]

DieMetRik, 2023-09-27:
Try to install again.

KrX3D, 2023-09-27:
i got the same problem. i noticed that the /config/esphome/.esphome folder is missing

when i hit install on i.e. one of my esp32 this folder is not generated anymore

im not sure  if this was always the folder or if it changed:

Building **.pioenvs**/flexispot-e7/firmware.bin

DieMetRik, 2023-09-27:
Folder move to add-on/data folder

3ative, 2023-09-27:
> Try to install again.

Tried that too... Made no difference

3ative, 2023-09-27:
> > Folder move to addon/data folder
> 
> via windows smb this folder is empty, also when i look at it with studio code server

Yes, it was listed as one of the "Breaking Changes" - It's a real pain to debug cpp files now.

jesserockz, 2023-09-27:
Unfortunately I cannot automatically fix this for people that have already upgraded to 2023.9.0, but I will try to put out 2023.9.1 today that will migrate the json files used for the card information so people that skip 2023.9.0 wont have this issue.

ssieb, 2023-09-27:
> It's a real pain to debug cpp files now.

The only cpp file that you don't already have is the generated `main.cpp`.  But really, if you're debugging you should be running esphome locally.  It's much easier.

andreas-bulling, 2023-09-27:
> Unfortunately I cannot automatically fix this for people that have already upgraded to 2023.9.0, but I will try to put out 2023.9.1 today that will migrate the json files used for the card information so people that skip 2023.9.0 wont have this issue.

Would be good to at least have instructions for how to fix this for these people (I am one of them and I have 45+ ESPHome devices running)

jesserockz, 2023-09-27:
If you have a recent HA core backup, you could pull the json files from there and put them into the right place. This could be back into the `/config/esphome/.esphome` folder which should then be moved into the `/data` folder when upgrading to 2023.9.1 later

andreas-bulling, 2023-09-27:
And how do I copy the files from the backup to the hassio directory - samba?

ssieb, 2023-09-27:
Yes, or any of the other file manager addons.

3ative, 2023-09-27:
Thanks @jesserockz, that's great. 🙏
Luckily I copied my .esphome folder before I upgraded. 😎

andreas-bulling, 2023-09-27:
that's not great, this is a desaster. I also don't get why it doesn't create the files in /data apparently - or does it, i.e. I can just upgrade all devices and don't have to  upload the json files manually one by one?

jesserockz, 2023-09-27:
@andreas-bulling yes you can just install/upgrade them and the json files will be regenerated again one-by-one

jesserockz, 2023-09-27:
@andreas-bulling If you have 2023.9.0 already, it wont be looking at the .esphome directory, it will only look in the /data directory for the files

closed 2023-09-27 by jesserockz

KrX3D, 2023-09-27:
/addon 
/data

do not contain any .esphome folder...

[image]

jesserockz, 2023-09-27:
`/data` is individual per each addon (the ssh addon has its own `/data` folder). 
You can only see the contents either from the host OS, or by exec into the esphome addon container.

3ative, 2023-09-28:
For those with back-ups (prior to 2023.9.0), wait for this update, copy back your JSONs and start the add-on.

https://github.com/esphome/esphome/pull/5441#issuecomment-1738537547

andreas-bulling, 2023-09-28:
FYI - I had upgraded to 2023.9.0 already without saving the JSONs. `.esphome` was gone and no available updates were indicated in the web interface then as many others have reported.

I upgraded to 2023.9.1 this morning without copying over the JSONs manually before and now the update status is still shown correctly again.

3ative, 2023-09-28:
> FYI - I had upgraded to 2023.9.0 already without saving the JSONs. `.esphome` was gone and no available updates were indicated in the web interface then as many others have reported.
>
> I upgraded to [...]

I noticed that too... I thought I comment, just in case.

KrX3D, 2023-09-28:
> For those with back-ups (prior to 2023.9.0), wait for this update, copy back your JSONs and start the add-on.
> 
> [esphome/esphome#5441 (comment)](https://github.com/esphome/esphome/pull/5441#issuecomment-1738537547)

- update esphome
- stop esphome addon
- copy back .esphome folder from backup
- start esphome addon

that worked for me, now i see the UPDATE button again

JackTalisker, 2023-10-02:
I upgraded my esphome (docker container) from 2023.8.3 to 2023.9.2 and my .esphome config directory seems to be unharmed in its old position "/home/pi/esphome:/config"
Starting the container I do not see the update button.
What can I do?
Thank you