esphome/issues #4913: Breaking change from .esphome to /data

Fluffkin, 2023-09-27:
### The problem

Just to clarify the situation here.
There's a breaking change in the latest release https://github.com/esphome/esphome/commit/fe81bcc00343ba8d99a437762087717d45a20946 that moves the .esphome contents to /data.

It's not documented.

It contains code that will delete my existing .esphome folder without asking when I restart and load esphome.

At the moment it's still compiling the updated version apparently using the .esphome folder (?!).
There's nothing in /data after test upgrading one of my ESP8266 units.

There's no easy way for the user to install fonts etc to /data because it's not exposed by the Samba server.

When does this time bomb go off exactly and delete all my fonts etc from .esphome without moving them for me first? When I reboot HA?

### Which version of ESPHome has the issue?

2023.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.09.2

### What platform are you using?

ESP8266

### Board

d1 mini

its-me-jake, 2023-09-27:
Following because I just nuked my installation this morning.  Similar issue - I cannot find the /data directory at all, and my configuration was all set up using packages.  I did a test installation of a new Wemos D1 to see if that helped, but no.  All my devices are still working, but I can't update them or figure out where to put my package files as I rebuild them.

3ative, 2023-09-27:
Chatting with one of the DEVs this morning. His advice was to shell in to: cd /mnt/data/supervisor/addons/data/5c53de3b_esphome/

And: "if you need to do a lot of debugging with the files in the .esphome directory, you should install it locally". Then he pointed me to this link: https://esphome.io/guides/installing_esphome#windows

Personally, this "Breaking Change" is a real kick in the NADs for us tinkerers and tutorial makers - _I'm really annoyed_

ssieb, 2023-09-27:
I think you're all misunderstanding.  You should never have put anything in `.esphome`.  You don't need to access the `/data` directory either.  These are temporary storage areas for esphome to use while compiling.  Any files you need to use are in `/config/esphome` or normal subdirectories under that..  That's where you put your yaml, images, fonts, etc.

Fluffkin, 2023-09-27:
Ahh.
This maybe where documentation would have been an idea?
IRCC when I first got displays working in esphome I put the fonts in .esphome

ssieb, 2023-09-27:
I'm curious why you would do that.  `.` directories are usually "hidden" and reserved for application use.
Everything in the docs says to put extra files in the same directory as the yaml files or a directory under it *that you create*.  "installing .esphome locally" is misunderstanding what was said.  The comment was that if you need to look at the build files in `.esphome` for debugging purposes, then you should install esphome locally so that you have easy access to it.  But you should never be putting any files in there yourself.

its-me-jake, 2023-09-27:
@ssieb how would one implement a package-based configuration with the new version?  I followed this: https://github.com/jcallaghan/esphome-config and after I updated, all of my files were just gone.  I realize that I have to rebuild everything one way or the other, but I also have another installation that I have not upgraded yet, and I would like to know how to migrate that if it is possible.

ssieb, 2023-09-27:
I don't see how anything there would be affected by this change.  Maybe come to discord, so it's easier to discuss.

leoshusar, 2023-09-28:
I am currently experiencing this issue:
```
-- Configuring incomplete, errors occurred!

CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/targets.cmake:108 (message):
   Target 'esp32' in sdkconfig '/data/build/4heat-controller/sdkconfig.4heat-controller' does not match currently selected IDF_TARGET 'esp32c6'. To change the target, clear the build directory and sdkconfig file, and build the project again.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:24 (__target_init)
  CMakeLists.txt:2 (include)
```

Since the UI "new device" dialog doesn't support C6 yet, I selected ESP32 device type and then wanted to just change the board.
But just changing the board apparently doesn't clean the sdkconfig, so I needed to delete it using `docker exec` in advanced terminal.

IMO there are legitimate use cases for accessing build directory - for example for cleaning things that "clean build files" doesn't clean.

DjordjeMandic, 2023-09-28:
Honestly i would like to have access to the data. Sometimes reading generated code helps better understanding it, copying some files for debugging and etc. It's fine for everything to stay in isolated /data but please make a mount/link to .esphome directory for easy access.

ssieb, 2023-09-28:
The clean build command clears everything out for that specific device.  There is nothing more that you could do with file access.  It literally deletes the entire build directory for that device.
There is no way to link to the `/data` directory.
And as I mentioned before, the only generated file is `main.cpp`.  Everything else you already have easy access to.

ssieb, 2023-09-28:
https://github.com/esphome/esphome/pull/5443

Dilbert66, 2023-11-07:
Unfortunately, another issue that occured is that there is no way to do a full on the build files when recompiling.  I'm in the process of migrating some custom_components to the external components mechanism because of course esphome home is deprecating custom components.  The issue that occurs is that the included files from the old custom components are never deleted and now conflict with the new external_component files which are in a different directory.  Deleting the .pioenv files doesnt accomplish anything here.  Wasnt there a real clean option before?  At least with the .esphome directory, it was easy enough to tell users of the compoents to go and delete the directory. I'm not going to start asking people to go to terminal, bash into the addon esphome docker container and follow complex path to have them do it.  Absolutely not.

So please either make the Clean option delete the complete build directory or at the very least add another option to the menu to add an option to also clean the source files as well as the compiled objects.

camsaway, 2024-10-20:
I know this is an old thread, but I'm wondering if it might be what I'm looking for as well.

I'm trying to debug some errors in my esphome logs:
[13:30:16][W][component:237]: Component interval took a long time for an operation (149 ms).
[13:30:16][W][component:238]: Components should block for at most 30 ms.

If I recall, those numbers 237 & 238 refer to the line numbers in the generated interim code that gets compiled. How am I meant to work out which component is causing the error without access to those files?

DjordjeMandic, 2024-10-20:
> I know this is an old thread, but I'm wondering if it might be what I'm looking for as well.
>
> I'm trying to debug some errors in my esphome logs:
> [13:30:16][W][component:237]: Component interval [...]

I'd for start enable very verbose logging and see what's going on via uart. What's your current log level?

ssieb, 2024-10-20:
That's a generic log message and those lines aren't going to help you.
Those numbers are the line numbers from the source code, not the generated code.  You can look in github for the files.
It tells you which component is taking too long, so see what you're doing in the `interval:` to cause that.

camsaway, 2024-10-21:
Thanks guys - I thought from previous investigations that it was the lien number of the generated code. If not, then I'm off base. Thanks for the suggestions - I suspect it's the SPI display, will enable more verbose logging and work from there.