esphome/issues #4914: ESPHome 2023.9.0 broke SPI (with st7789v display)

RC-Thoughts, 2023-09-27:
### The problem

Have been happy user of ESP32-board with st7789v display, no issues. Now did an update to 2023.9.0 (from previous, I think it was 2023.8.3) and suddenly the display stays blank. I can see a few flickers on backlight but that's it.

I would think this is an SPI issue since using "interface: software" I can get the attached display to work, albeit really slow.:
[W][component:204]: Component st7789v.display took a long time for an operation (1.85 s).

Before update it was more functional with:
[W][component:204]: Component st7789v.display took a long time for an operation (0.27 s).

### Which version of ESPHome has the issue?

2023.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4

### Component causing the issue

SPI st7789v

### Example YAML snippet

```yaml
esp32:
  board: esp32dev # Also tried az-delivery-devkit-v4 same result
  framework:
    type: arduino
--
--
# Display
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  interface: software # This is only option that works, hardware, spi, spi1, spi2 do not

display:
  - platform: st7789v
    model: Custom
    height: 240
    width: 240
    offset_height: 0
    offset_width: 0
    rotation: 90
    eightbitcolor : yes
    #update_interval: 20ms
    backlight_pin: GPIO16
    cs_pin: GPIO25
    dc_pin: GPIO26
    reset_pin: GPIO27
    lambda: .....
```

RC-Thoughts, 2023-09-27:
Just as a test, tried esp-idf to rule out framework, no difference. Hardware SPI seems to be broken.

RC-Thoughts, 2023-09-28:
Just tested 2023.9.1, still broken.

ssieb, 2023-09-28:
Can you get serial logs?

DrBando, 2023-09-28:
I believe I have a related problem: my TTGO-T1 ESP32 board with integrated ST7789V TFT LCD is working, however, I can't change the backlight brightness anymore. It stays at 100% and is not changable. Other components with other display types do work, even the brightness properties.

RC-Thoughts, 2023-09-28:
> Can you get serial logs?

This is from OTA and restart after that:
`[D][ota:151]: Starting OTA Update from 123.123.123.6...68.1.6): Connection closed
[V][ota:182]: OTA features is 0x01to ESPHome Logs 2023.9.1 (123.123.123.6)
[V][ota:202]: Auth: Nonce is 6b04820963b9c22a5ca83990f3b2b1e7e for an operation (0.27 s).
[V][ota:222]: Auth: CNonce is a611c697eff041ca9ee914d71767355es.
[V][ota:229]: Auth: Result is 381883e0e6b4ebdd009e0d3e9a70940e
[V][ota:237]: Auth: Response is 381883e0e6b4ebdd009e0d3e9a70940ecess=0, value=53.3)
[V][ota:265]: OTA size is 1012128 bytes
[239401][D][Updater.cpp:133] begin(): OTA Partition: app0
[V][ota:282]: Update: Binary MD5 is cf83af65e38629292d7defd60ad6a1b7
[D][ota:320]: OTA in progress: 0.1%
[D][ota:320]: OTA in progress: 11.7%
[D][ota:320]: OTA in progress: 23.5%
[D][ota:320]: OTA in progress: 33.5%
[D][ota:320]: OTA in progress: 45.8%
[D][ota:320]: OTA in progress: 58.7%
[D][ota:320]: OTA in progress: 71.6%
[D][ota:320]: OTA in progress: 84.6%
[D][ota:320]: OTA in progress: 98.7%
[I][ota:353]: OTA update finished!
[I][app:134]: Rebooting safely...
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[V][esp32.preferences:126]: sync: key: 233825507, len: 4
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[248624][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: SSIDname, BSSID: 00:00:00:00:00:00, Reason: 8
[248626][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[248628][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Stopped
[248633][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[W][wifi_esp32:465]: Event: Disconnected ssid='SSIDname' bssid=00:00:00:00:00:00 reason='Association Leave'
[248666][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
[V][wifi_esp32:440]: Event: WiFi STAets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[     6][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[I][logger:268]: Log initialized
[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[V][esp32.preferences:126]: sync: key: 233825507, len: 4
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[D][spi:039]: Setting up SPI bus...
[C][st7789v:011]: Setting up SPI ST7789V...
[D][spi_device:354]: mode 0, data_rate 20000kHz
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: 00:00:00:00:00:00
[V][esp32.preferences:059]: nvs_get_blob('3570421264'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][wifi_esp32:039]: Enabling STA.
[   657][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[V][wifi_esp32:424]: Event: WiFi ready
[   752][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[   753][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[V][wifi_esp32:435]: Event: WiFi STA start
[V][internal_temperature:029]: Raw temperature value: 204
[V][sensor:043]: 'Internal Temp': Received new state 95.555557
[D][sensor:094]: 'Internal Temp': Sending state 95.55556 °C with 1 decimals of accuracy
[W][component:204]: Component st7789v.display took a long time for an operation (0.27 s).
[W][component:205]: Components should block for at most 20-30ms.
[V][esp-idf:000]: E (3436) wifi:
[... 232 lines cut ...]
[C][homeassistant.sensor:033]:   Attribute: 'volume_level'
[C][homeassistant.sensor:030]: Homeassistant Sensor 'lights'
[C][homeassistant.sensor:030]:   State Class: ''
[C][homeassistant.sensor:030]:   Unit of Measurement: ''
[C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[V][homeassistant.sensor:030]:   Force Update: YES
[C][homeassistant.sensor:031]:   Entity ID: 'light.cd_led'
[C][homeassistant.sensor:033]:   Attribute: 'brightness'
[C][homeassistant.text_sensor:023]: Homeassistant Text Sensor 'power'
[C][homeassistant.text_sensor:024]:   Entity ID: 'media_player.living_room'
[C][homeassistant.text_sensor:023]: Homeassistant Text Sensor 'sound'
[C][homeassistant.text_sensor:024]:   Entity ID: 'media_player.living_room'
[C][homeassistant.text_sensor:026]:   Attribute: 'sound_mode'
[C][homeassistant.text_sensor:023]: Homeassistant Text Sensor 'mute'
[C][homeassistant.text_sensor:024]:   Entity ID: 'media_player.marantz_sr7015_2'
[C][homeassistant.text_sensor:026]:   Attribute: 'is_volume_muted'
[C][homeassistant.text_sensor:023]: Homeassistant Text Sensor 'source'
[C][homeassistant.text_sensor:024]:   Entity ID: 'media_player.living_room'
[C][homeassistant.text_sensor:026]:   Attribute: 'source'
[C][homeassistant.text_sensor:023]: Homeassistant Text Sensor 'heos'
[C][homeassistant.text_sensor:024]:   Entity ID: 'media_player.marantz_sr7015_2'
[C][homeassistant.text_sensor:026]:   Attribute: 'source'
[W][component:204]: Component st7789v.display took a long time for an operation (0.27 s).
[W][component:205]: Components should block for at most 20-30ms.
`

RC-Thoughts, 2023-10-02:
Unfortunately still broken in 2023.9.2, only "interface: software" works, slow as watching paint dry.

clydebarrow, 2023-10-02:
Just saw this issue - will check it tomorrow. I have an st7789v display to test on.

clydebarrow, 2023-10-02:
Tested with my st7789v display - no problem. There are a few possibilities, I think it's less likely to be an SPI problem, more likely to be related to changes to the st7789v driver. These changes were tested on several boards, but one possibility is that the data rate was changed from 10MHz to 20MHz. The chip is specified at the higher rate, but may fail depending on the electrical characteristics of the connection.

Currently the data rate is not configurable for the st7789v so I have raised a PR to test this. Add this to your yaml:

```
external_components:
  source: github://pr#5466
  components: [ spi, st7789v ]
```

And add `data_rate` to your display config:

```
display:
  - platform: st7789v
    data_rate: 10MHz
```

Test and report back.

RC-Thoughts, 2023-10-03:
Using the "external_components" gives no issues, using "data_rate: 10MHz" is not allowed.

clydebarrow, 2023-10-03:
That's strange. What was the exact error message? Also check you have the right PR number. - I originally posted the wrong one and later edited it. You could try this instead:

```
external_components:
  source: github://clydebarrow/esphome@st7789v
  components: [ spi, st7789v ]
```

RC-Thoughts, 2023-10-03:
This is getting interesting. I thought that maybe the "data_rate" is not accepted since driver is not loaded correctly. So I did a clean build and then the "data_rate" was accepted. However, unless I do "interface: software" the display stays black.

Current yaml:
`# Display
external_components:
  source: github://clydebarrow/esphome@st7789v
  components: [ spi, st7789v ]

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  #interface: software

display:
  - platform: st7789v
    data_rate: 10MHz
    model: Custom
    height: 240
    width: 240
    offset_height: 0
    offset_width: 0
    rotation: 90
    eightbitcolor : yes
    #update_interval: 200ms
    backlight_pin: GPIO16
    cs_pin: GPIO25
    dc_pin: GPIO26
    reset_pin: GPIO27
    lambda: |-`

clydebarrow, 2023-10-03:
Is the backlight on? I.e. can you see the LED behind the display? And in the log do you see messages about the display taking a long time? And which display exactly are you using? I assume it's not built into the board.

You could also try a much lower data_rate, like say 1MHz. Will be slow of course but if that still fails it should rule out data rate as an issue. The software SPI runs at around 200kHz.

RC-Thoughts, 2023-10-03:
Yes, I can clearly see the backlight so there is signs of life. I have no idea what the exact display it is, it has st7789v chip and have been working as such up to 2023.9.0. 

I have tried various speeds from 1MHz and up, nothing with hw-spi.

And yes, these exist:
[14:59:50][W][component:204]: Component st7789v.display took a long time for an operation (1.14 s).
[14:59:50][W][component:205]: Components should block for at most 20-30ms.

clydebarrow, 2023-10-03:
Ok, so next step is to isolate the st7789v and spi changes. I'll make a couple of branches for you to try, one with the old display driver and new SPI, and one the reverse. Will be tomorrow though, it's late here.

RC-Thoughts, 2023-10-03:
No worries, your help is highly appreciated!

RC-Thoughts, 2023-10-03:
Updated to 2023.9.3, no change.

clydebarrow, 2023-10-03:
Ok, I've updated the PR - you will probably want to do clean _and_ delete the `.esphome/external_components` folder to make sure you get the latest version.

With the change you can now add a line like `spi_mode: MODE3` to the display config. There are four possible modes (0-4) and on my display only MODE0 works. I know there are other boards that work with both MODE0 and MODE3. So I suggest trying the different modes (start with 3) and see if it has any effect.

The ST7789V datasheet clearly shows it as expecting mode 0, but I wonder if there are other "st7789v-like" chips that might behave differently. The mode was changed from 0 to 3 in the recent update.

clydebarrow, 2023-10-03:
And best add a `refresh` line to the external components config:

```
external_components:
  source: github://clydebarrow/esphome@st7789v
  components: [ spi, st7789v ]
  refresh: always
```

RC-Thoughts, 2023-10-04:
I did changes one at a time with clean build every time, the adding of "spi_mode: MODE3" is the key, that made spi work with default "interface: hardware" and display speed is now back to normal.

clydebarrow, 2023-10-04:
Great! I have found mention elsewhere on the net that some st7789v's require mode 3, so it's just going to have to be a user-configured option.

closed 2023-10-04 by kbx81

DrBando, 2023-10-05:
I'm not sure if this is related, but seeing my post above, I'm still not able to change the brightness of the backlight, while it is working with other display types and was working with st7789v before 2023.9.0.
I've tried the possible solution with spi_mode: MODE3, where I can confirm that MODE0 and 3 are showing content in the display, while 1 and 2 are not working, however, display brightness can't be changed in either 0 or 3. Should I open a new issue?

kbx81, 2023-10-05:
@DrBando the issue with the backlight is fixed by https://github.com/esphome/esphome/pull/5476 and it will either be in the next patch release or next major release. No need to open a new issue at this point.

besanur, 2023-10-11:
same problem here with the Adafruit Library and the example code it is working. But in esp-home not anymore since the update. I have this display https://de.aliexpress.com/item/1005005039702168.html.  Until recently it worked great in esphome.

```
esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: arduino

# Enable logging
logger:

time:
  - platform: homeassistant
    id: esptime

### Display
font: 
  - file: "fonts/Roboto-Regular.ttf"
    id: font_clock
    size: 34
    glyphs: ':0123456789'

spi:
  clk_pin: GPIO7 # SCL
  mosi_pin: GPIO11 # SDA

display:
  - platform: st7789v
    id: arbeitstisch_display
    model: Adafruit RR 280x240
    backlight_pin: GPIO39 # BLK
    cs_pin: GPIO10 # CS
    dc_pin: GPIO33 # DC
    update_interval: 30s
    reset_pin: GPIO15 # RES
    lambda: |-
      it.printf(it.get_width()/2, 0, id(font_clock), TextAlign::TOP_CENTER, id(esptime).now().strftime("%H:%M").c_str());
```

besanur, 2023-10-11:
> And best add a `refresh` line to the external components config:
> 
> ```
> external_components:
>   source: github://clydebarrow/esphome@st7789v
>   components: [ spi, st7789v ]
>   refresh: always
> ```

with this and spi_mode: 'mode3' it works