esphome/issues #4915: SPIDevice not initialised - did you call spi setup()?

erik3rik, 2023-09-27:
### The problem

After update to 2023.9.0 (from 2023.8.3), all my units that has a display with SPI interface are bricked.
Not the screen itself, the entire ESP32 refuses to boot. Neither power cycling nor pressing the reset button solves the issue, and the unit is no longer available for OTA updates.

Upon connecting a device using USB, I was able to retrieve the logs.

Rolling back to ESP Home 2023.8.3 and upload the software by USB fixes the issue (with working display). Removing the display: block (see below) also fixes the issue and works with ESP Home 2023.9.0 but for obvious reasons the display does not work in this case.

Logs:
[image]

The device reboots every second and the logs looks like in the screenshot above.

### Which version of ESPHome has the issue?

2023.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4

### Component causing the issue

display (SPI?)

### Example YAML snippet

```yaml
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23

display:
  - platform: ssd1306_spi
    model: "SH1106 128x64"
    cs_pin: GPIO05
    dc_pin: GPIO22
    reset_pin: GPIO19
    id: oled_display
    flip_y: true
    pages:
    # Long list of pages that are not relevant to the case
```

### Anything in the logs that might be useful for us?

```txt
[E][spi:078]: SPIDevice not initialised - did you call spi setup()?
Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.

Core 1 register dump:
...

Backtrace:
...
```

samuelolteanu, 2023-09-27:
I was in the diagnosing a bunch of ip esphome conflicts with eof orors, ping timeouts and of cursed api key changes then got "[Errno 111] Connect call failed ('<esp-ip>', 6053) (SocketAPIError)" on an not easily reachable device with a 7 digit display...so yeah, it was not the wifi, not the display hardware nor its lambda because any other device bootloops with spi configured with max7219 example and no actual display connected. Tested on esp32s2 and vanilla esp32.

RudiP, 2023-09-28:
Same issue here with ESPHome 2023.9.1, SPI display st7789v

[image]

clydebarrow, 2023-09-28:
Looking into it...

clydebarrow, 2023-09-28:
Can you share the whole yaml? The display on its own does not seem to cause an issue - I took your sample and can build and run that no problem. Any other SPI devices in there? Anything that might be trying to access the display before everything is set up?

clydebarrow, 2023-09-28:
> Same issue here with ESPHome 2023.9.1, SPI display st7789v

I just tested on an ESP32S3 with st7789v (LilyGo T-Embed) with no issues.

How much RAM is free? The new SPI code does use a DMA buffer, it's possible that RAM usage is higher than before.

samuelolteanu, 2023-09-28:
`esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

esphome:
  name: esp32-4
  friendly_name: esp32-4

logger:
  level: very_verbose

api:
  encryption:
    key: "ahvDDjMvo+CtqnrbjYg7PCr2aGZ/BRkdX9nNp5jyg0E="

ota:
  password: "56b68f4eb517a5a1123c39441e9899a3"

wifi:
  ssid: 
  password:
  manual_ip:
        static_ip: 192.168.0.138
        gateway: 192.168.0.1
        subnet: 255.255.255.0

  ap:
    ssid: "Esp32-4 Fallback Hotspot"
    password: "RRxZqKsCQdZ4"

captive_portal:

mdns:

  
    

spi:
  clk_pin: 22
  mosi_pin: 23

display:
  - platform: max7219
    cs_pin: 21
    num_chips: 1
    lambda: |-
      it.print("01234567");`

clydebarrow, 2023-09-28:
You want three backticks, on a line by themselves, before and after the yaml.

samuelolteanu, 2023-09-28:
```
esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

esphome:
  name: esp32-4
  friendly_name: esp32-4

logger:
  level: very_verbose

api:
  encryption:
    key: "ahvDDjMvo+CtqnrbjYg7PCr2aGZ/BRkdX9nNp5jyg0E="

ota:
  password: "56b68f4eb517a5a1123c39441e9899a3"

wifi:
  ssid: 
  password:
  manual_ip:
        static_ip: 192.168.0.138
        gateway: 192.168.0.1
        subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-4 Fallback Hotspot"
    password: "RRxZqKsCQdZ4"

captive_portal:

mdns:

  
    

spi:
  clk_pin: 22
  mosi_pin: 23

display:
  - platform: max7219
    cs_pin: 21
    num_chips: 1
    lambda: |-
      it.print("01234567");

```

clydebarrow, 2023-09-28:
Ok, I have reproduced it. Stand by...

clydebarrow, 2023-09-28:
Ok, for the MAX7219 adding this to the yaml will fix it:

```
external_components:
  source: github://pr#5446
  components: [ max7219 ]
```

It's a bug in the MAX7219 component, was not exposed until the new SPI code landed which is a bit stricter.

Don't know about the other reports until I get a full YAML to reproduce.

clydebarrow, 2023-09-28:
I figured out why the problem caused a crash, and have pushed to the PR. So adding this to the YAML should fix any other instances of the problem:

```
external_components:
  source: github://pr#5446
  components: [ spi ]

```

However, please report here if you find this necessary and/or if you see the "SPI Device not initialised" error message.

The underlying cause will be an error in another component and it will be nice to fix those rather than just papering over the cracks.

RoelHermus, 2023-09-28:
I had a problem with the max7219 dot display. Adding ` 

> external_components:
>   source: github://pr#5446
> components: [ spi ]`

 fixed the problem. The first solution, with max7219 between the brackets did not fix the problem. The ESP8266 restarted again and again. Thanks so far for this solution

closed 2023-09-29 by jesserockz

erik3rik, 2023-09-29:
Is that (5446) really fixing all issues? It seems to fix problems with max7219 as per the example that [samuelolteanu](https://github.com/samuelolteanu) showed ( platform: max7219
), but my original problem arises from  platform: ssd1306_spi, is that also fixed by 5446? It only mentions max7219. 

If ssd1306_spi is not fixed by this, then this bug report should not be closed.

clydebarrow, 2023-09-29:
It should stop the ESP crashing. You may still see the "SPI not initialised error" which should be reported, but is technically a different bug.

Can you try it and see?

erik3rik, 2023-09-29:
I will try after work today. Can't really test it remotely as if it does not work the device becomes bricked and needs cable connected to it to update.

kbx81, 2023-09-29:
@erik3rik the issue was auto-closed because it was mentioned in the PR (which was merged). Still, please test latest dev (or you can add the PR directly, as mentioned above) and let us know if the issue persists.

samuelolteanu, 2023-09-29:
max7219 7digit not crashed anymore on esp32s2 wih both pr#5446 max7219 or spi separately.

erik3rik, 2023-09-29:
Using:
```
external_components:
  source: github://pr#5446
  components: [ spi ]
```

While also having the newer ESPHome 2023.9.1, I still get the `[E][spi:078]: SPIDevice not initialised - did you call spi_setup()?` error, twice. See the log below. But it is at least improving in the sense that the ESP32 no longer panics during boot and thus is no longer bricked after update.

Yet some kind of problem still exist, and I am unsure what to do about it.

[image]

clydebarrow, 2023-09-29:
Can you provide the entire YAML?

erik3rik, 2023-09-29:
Yes, but its quite long and contain **a lot** of stuff. I'm sorry anyone else but me have to see this. Anyhow, here goes 1300 rows:

```
substitutions:
  friendly_name: ESP Clock Radio
  fixed_ip: "192.168.0.123"

esphome:
  name: esp-clock-radio   

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

preferences:
  flash_write_interval: 5min

# Enable logging
logger:

external_components:
  source: github://pr#5446
  components: [ spi ]
  

font:
  # gfonts://family[@weight]
  - file: "gfonts://Roboto"
    id: roboto_large
    size: 28

  - file: "gfonts://Roboto@500"
    id: roboto_xlarge
    size: 50

  - file: "fonts/fira-sans.two.ttf" #"gfonts://Roboto@300"
    id: roboto_xlargeThin
    size: 55 # 50
    
  - file: "gfonts://Roboto@100"
    id: roboto_small
    size: 18

  - file: "gfonts://Roboto@100"
    id: roboto_xsmall
    size: 12

  - file: "gfonts://Roboto@100"
    id: roboto_xxsmall
    size: 9

#  - file: "gfonts://Carrois Gothic"
#    id: big_value_font
#   size: 27

    #https://pictogrammers.com/library/mdi/
  - file: "fonts/mdi.ttf"
    id: mdi
    size: 38
    glyphs: 
      - "󰀦" # Warning
      - "󰖍" # Drop off
      - "󰖌" # Drop on
      - "󰔏" # Thermometer
      - "󱯶" # Cloud off
      - "󰅠" # Cloud on
      - "󰇵" # Happy
      - "󰇸" # Sad
      - "󰖨" # Sun
[... 1156 lines cut ...]
    id: version_string
    name: $friendly_name Version
  - platform: wifi_info
    ip_address:
      name: $friendly_name IP
    ssid:
      name: $friendly_name SSID    
  - platform: homeassistant
    id: current_time
    entity_id: sensor.clock_radio_time
  - platform: homeassistant
    id: wake_time
    entity_id: sensor.clock_radio_alarm
  - platform: homeassistant
    id: room_fan
    entity_id: sensor.clock_radio_fan
  - platform: homeassistant
    id: room_temperature
    entity_id: sensor.clock_radio_temperature
  - platform: homeassistant
    id: room_scene
    entity_id: sensor.clock_radio_scene

    
   ```

clydebarrow, 2023-09-29:
Ok, found the problem. In your YAML:
```
switch:
  - platform: template
    name: "$friendly_name Night Mode"
    id: night_mode_switch
    restore_mode: ALWAYS_ON
    lambda: |-
      return id(is_night_mode);
    turn_on_action:
      - globals.set:
          id: is_night_mode
          value: 'true'
      - lambda: id(oled_display).set_contrast(0.01);           <-------------
      - component.update: oled_display
 ```

The switch gets turned on as part of after-boot restore, so `set_contrast` call is done _before_ the display has been setup, so the SPI is not initialised at that time. 

It might be argued that the ssd1306 driver is structured incorrectly, as it probably should not write directly to the SPI when values are changed, but simply store the value and update it when its `update()` method is called (which will be after setup.)

For your YAML you might want to defer setting contrast or any other direct call to the display in lambdas until after setup is complete.

clydebarrow, 2023-09-29:
I would note that the initial on-restore `set_contrast()` probably never worked anyway, it just silently failed.

erik3rik, 2023-09-30:
I tried commenting out that line with set_contrast() and the error disappeared. 

> I would note that the initial on-restore set_contrast() probably never worked anyway, it just silently failed.

Perhaps.

Yet, I want the behaviour of that switch to adjust the contrast of the display. Both the turn_on_action and the turn_off_action adjusts the contrast of the display. And that is how I want it to be. And since the state is binary (it is either on or off) the button will be restored to any of those two states, and both of them will set the contrast --> error message.

 I could of course wrap that row in an additional if-clause to prevent that row to happen too early, yet be there in all other situations. It would not really have an impact on the behaviour of the device as it is essentially always connected to power (it typically only reboots to bump the ESPHome software). But it would make an already messy code even messier. 
 
 > It might be argued that the ssd1306 driver is structured incorrectly, as it probably should not write directly to the SPI when values are changed, but simply store the value and update it when its update() method is called (which will be after setup.)
 
 I totally agree. And since that is unfortunately not the current behaviour then I guess I would have to go for that if-clause...

clydebarrow, 2023-09-30:
Or you can just leave things as they are and ignore the messages in the log. Other than that message, nothing now behaves differently to what it did before (now that the crashing, which was totally my fault, has been fixed.)

That's probably what I'd do for now at least.

clydebarrow, 2023-09-30:
Aha, I found another solution - add a line to your yaml to make the display have higher setup priority than the switch.

```
display:
  - platform: ssd1306_spi
    setup_priority: 950
```

The display by default gets set up _after_ the switches. Template switches have priority `HARDWARE` which seems odd since they are not hardware (if I were King I would call them "virtual" switches, not templates :-) Displays, or at least the ssd1306, have priority `PROCESSOR` which also makes little sense to me.

Anyway, in your case you want the display to have a higher priority than anything that touches it.