esphome/issues #4916: OTA with esp-idf not working

TimelessNL, 2023-09-27:
### The problem

The OTA functionality does not seem to work, the update itself is successful but after reboot it is still running the previous firmware.

### Which version of ESPHome has the issue?

2023.9.0b2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

Shelly Plus 1PM

### Component causing the issue

OTA

### Example YAML snippet

```yaml
esphome:
  name: ${hostname}
  comment: "Shelly Plus 1PM"
  project:
    name: "ESPHome.Shelly"
    version: "Shelly Plus 1PM"
  platformio_options:
    board_build.f_cpu: 160000000L

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_160: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "160"

preferences:
  flash_write_interval: 1d

ota:
  password: !secret ota_password
```

### Anything in the logs that might be useful for us?

```txt
INFO Uploading /config/.esphome/build/shelly/.pioenvs/shelly/firmware.bin (859616 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
```

### Additional information

The OTA section on the webportal is also not present, I cannot upload a image via the webinterface.
[image]

TimelessNL, 2023-09-27:
The OTA log output from the device:
[image]

nerdyjan, 2024-09-18:
same here with 2024.8.3 and Shelly PM mini Gen3 (https://devices.esphome.io/devices/Shelly-PM-Mini-Gen3)

hannesb, 2024-10-28:
same with 2024.10.2 and firebeetle esp32-c3

hannesb, 2024-10-29:
Got it working now by specifing the newest version of the ESP-IDF framework:
```
esp32:
  board: dfrobot_beetle_esp32c3
  flash_size: 4MB
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
```

muhlba91, 2024-12-05:
any updates here? i have the same issue with a esp32-s3-devkitc-1 board.
ota using the CLI work but i don't see any section in the web interface.

@hannesb did you set any specific configuration other than the framework? (i have the same values)

randybb, 2024-12-05:
@muhlba91 there are two things in the original issue:
- in some cases after OTA the device boots from the same OTA partition - mostly on devices with a wrong partition scheme, so the device needs to be flashed via UART first and then it will work. Unfortunately, partition scheme cannot be changed via OTA yet (there is a PR for that...).
- and that OTA is not possible via the web interface - this is known limitation of IDF, so this is not an issue (more like a feature request).

muhlba91, 2024-12-05:
thanks for the explanation! this clarifies why it's not showing up in the web interface but works via USB. 👍