esphome/issues #4917: Cannot get ESP32-C6-WROOM-1U working

leoshusar, 2023-09-27:
### The problem

I am trying to build ESPHome for a custom board with ESP32-C6-WROOM-1U module and all I am getting after flash is this:
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0x68 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002949e
0x4002949e: uart_hal_get_txfifo_count in ROM

invalid header: 0xffffffff
invalid header: 0xffffffff
...
```

So I tried hello world from ESP-IDF, just to make sure my board is working fine, and it worked:
[image]

Then I tried to flash bootloader and partition table from that hello world project and I got *somewhere*:
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0x69 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
0x4001974a: software_reset in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd10
load:0x4086e610,len:0x2dc4
load:0x40875728,len:0x17d0
entry 0x4086c410
I (22) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (23) boot: compile time Sep 27 2023 22:27:31
I (24) boot: chip revision: v0.0
I (25) boot.esp32c6: SPI Speed      : 80MHz
I (30) boot.esp32c6: SPI Mode       : DIO
I (35) boot.esp32c6: SPI Flash Size : 8MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
E (83) boot_comm: mismatch chip ID, expected 13, found 0
E (89) boot: Factory app partition is not bootable
E (94) boot: No bootable app partitions in the partition table
```

but I guessed it won't be that easy.

I also thought the `board: esp32dev` might be the problem, but I don't see any other board I can use with my C6 module, since the devkit wasn't merged into stable.

And the web flasher apparently doesn't support C6 too:
[image]

Am I doing something wrong?

### Which version of ESPHome has the issue?

2023.9.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
esphome:
  name: 4heat-controller
  friendly_name: 4Heat controller
  platformio_options:
    board_upload.flash_size: 8MB
    build_flags:
      - -Wno-format

esp32:
  board: esp32dev
  variant: esp32c6
  framework:
    type: esp-idf
    platform_version: 6.4.0
    version: 5.1.1
```

leoshusar, 2023-09-28:
I was finally able to get `esp32-c6-devkitc-1` board working. After solving this issue:
```
-- Configuring incomplete, errors occurred!

CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/targets.cmake:108 (message):
   Target 'esp32' in sdkconfig '/data/build/4heat-controller/sdkconfig.4heat-controller' does not match currently selected IDF_TARGET 'esp32c6'. To change the target, clear the build directory and sdkconfig file, and build the project again.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:24 (__target_init)
  CMakeLists.txt:2 (include)
```

I ran into this:
```
Linking .pioenvs/4heat-controller/firmware.elf
RAM:   [=         ]   6.2% (used 32584 bytes from 524288 bytes)
Flash: [=====     ]  51.0% (used 936200 bytes from 1835008 bytes)
Building .pioenvs/4heat-controller/firmware.bin
Creating esp32c6 image...

A fatal error occurred: Contents of segment at SHA256 digest offset 0xb0 are not all zero. Refusing to overwrite.
*** [.pioenvs/4heat-controller/firmware.bin] Error 2
========================= [FAILED] Took 115.27 seconds =========================
```

cjc7373, 2023-09-28:
See https://github.com/stintel/platform-espressif32/commit/00df4d9e9beeb899cb7aa350947441221b2b5949

You can comment out the checksum part in the platform-espressif32 code. In my test this would not cause any problem.

closed 2023-09-28 by leoshusar

EEC70, 2024-01-22:
I am facing the same issue with the CMake error. Could you please share how you have solved it?

Cybermb, 2024-06-19:
Specifying esp-idf version to `5.2.2` solved issue for me
```
esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  framework:
    type: esp-idf
    version: 5.2.2 # this part
```

rvalitov, 2024-10-10:
Apart from proper ESP-IDF version, I had to specify a correct platform version and flash size, example:
```yaml
esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  flash_size: 8MB
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
```