esphome/issues #4918: No BK72xx / RTL87xx option to select when creating new Device.

mxbchr, 2023-09-27:
### The problem

When creating a new device from esphome interface, the option to select BK72xx / RTL87xx chips is missing.
They can still be configured by selecting another chip type and then editing the .yaml file to match the BK72xx / RTL87xx chips. But that is counterintuitive and it would be nice if esphome would spit out a working standard configuration .yaml file, as it does for the other chipsets when selected.

[image]

PS. Its not possible to select BK72xx / RTL87xx / Libretiny as plattform when creating new issues.

### Which version of ESPHome has the issue?

2023.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.2

### What platform are you using?

ESP8266

jesserockz, 2023-09-28:
This is not a bug and is not there by design. 
We will have something in a future release but are still not sure exactly what it will be.

closed 2023-09-28 by jesserockz

donburch888, 2023-12-13:
I agree this is not a bug, and is not by design ... however almost 3 months later it is still telling users that their BK72xx / RTL87xx devices cannot be added to ESPHome. 
I also agree that the current ESPHome selection of device type is not intuitive or very helpful to less knowledgable users, and that a better approach needs to be developed ... 

however ... while we are waiting for this, please advise what the correct or best method is to add new devices.  Do we need to install LibreTiny Integration ? Looking at the [ESPHome/LibreTiny + Tuya-Cloudcutter - tutorial video](https://www.youtube.com/watch?v=sSj8f-HCHQ0) the procedure looks so complicated that I am amazed I ever managed to get a device converted.