esphome/issues #4922: Compile Error after Upgrade SPI Touch Related

corbyschmitz, 2023-09-29:
### The problem

After upgrading to ESPHome 2023.9.1 (from 2023.8.3), I am receiving a compilation error for one of my devices. The configuration and code worked properly under 2023.8.3. 

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP32

### Board

ESP32-2432S028R - i.e. Cheap Yellow Display

### Component causing the issue

SPI

### Example YAML snippet

```yaml
esphome:
  name: entryway-display
  friendly_name: entryway-display

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: <REDACTED>

ota:
  password: <REDACTED>

wifi:
  ssid: !secret wifi_iot
  password: !secret iot_password

# ============================================================ 
# ESPHome Display related setup
#
# Create a font to use, add and remove glyphs as needed. 
font:
  - file: '/config/fonts/Arimo-Regular.ttf'
    id: arimo24
    size: 24
    glyphs: "!\"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZÅÄÖ abcdefghijklmnopqrstuvwxyzåäö"
  - file: '/config/fonts/Arimo-Regular.ttf'
    id: arimo96
    size: 96
    glyphs: ":0123456789"

# Create a Home Assistant blue color
[... 217 lines cut ...]
Compiling .pioenvs/entryway-display/src/esphome/components/wifi/wifi_component_libretiny.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/components/wifi/wifi_component_pico_w.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/application.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/component.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/controller.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/entity_base.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/log.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/scheduler.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/string_ref.cpp.o
Compiling .pioenvs/entryway-display/src/esphome/core/util.cpp.o
Compiling .pioenvs/entryway-display/src/main.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:288:24: error: expected primary-expression before 'return'
   touch->set_interface(return new SPIClass(HSPI));
                        ^~~~~~
*** [.pioenvs/entryway-display/src/main.cpp.o] Error 1
========================= [FAILED] Took 15.57 seconds =========================
```

### Additional information

Commenting out the touchscreen and related binary sensor still results in the compile errors. Only commenting out the SPI touch interface allows compilation (without the touchscreen and binary sensor). Happy to provide any other details as needed. The configuration is based on the example yaml for the board noted here:
https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/ESPHome

angelnu, 2023-09-29:
This problem is even simpler to reproduce as it only requires defining 2 SPI buses: https://github.com/esphome/issues/issues/4927

angelnu, 2023-09-30:
With https://github.com/esphome/issues/issues/4927 fixed, this should be solved as well.