esphome/issues #4923: Waveshare long delay before refresh

cbhiii, 2023-09-29:
### The problem

Since 2023.9.1 and likely change [#5283](https://github.com/esphome/esphome/pull/5283) my 7.5" screens (all 4 of them) are taking a long time to refresh when triggered to do so. No longer do they update within a couple seconds after being triggered. Now there is a long wait. See log detail.

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP32

### Board

waveshare Universal e-Paper Driver Board

### Component causing the issue

Waveshare E-Paper Display

### Anything in the logs that might be useful for us?

```txt
[12:38:04][D][main:126]: Skipping display refresh...
[12:38:06][D][main:126]: Skipping display refresh...
[12:38:08][D][main:126]: Skipping display refresh...
[12:38:10][D][main:126]: Skipping display refresh...
[12:38:10][D][homeassistant.binary_sensor:026]: 'binary_sensor.office_e_ink_frame_motion': Got state ON
[12:38:10][D][binary_sensor:036]: 'motion_detected': Sending state ON
[12:38:12][D][main:120]: Refreshing display...
[12:38:12][I][waveshare_epaper:1628]: Power on the display and hat
[12:38:22][E][waveshare_epaper:1573]: Timeout while displaying image!
[12:38:33][E][waveshare_epaper:1573]: Timeout while displaying image!
[12:38:33][D][sensor:094]: 'Last Update': Sending state 1696005504.00000  with 1 decimals of accuracy
[12:38:33][W][component:204]: Component time took a long time for an operation (21.48 s).
[12:38:33][W][component:205]: Components should block for at most 20-30ms.
[12:38:33][D][esp32.preferences:114]: Saving 1 preferences to flash...
[12:38:33][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[12:38:33][D][sensor:094]: 'Recorded Display Refresh': Sending state 10541.00000 Refreshes with 0 decimals of accuracy
[12:38:33][D][main:126]: Skipping display refresh...
[12:38:33][D][main:126]: Skipping display refresh...
[12:38:33][D][main:126]: Skipping display refresh...
[12:38:33][D][main:126]: Skipping display refresh...
[12:38:33][D][main:126]: Skipping display refresh...
[12:38:33][D][main:126]: Skipping display refresh...
```

cbhiii, 2023-09-29:
Didn't realize that 2023.9.1 had a breaking change just for the 7.5"V2 Waveshare e-paper screens. 

Had to invert busy pin per docs now. 

```
    busy_pin:
      number: GPIO25
      inverted: true 
```

closed 2023-09-29 by cbhiii

haiirogray, 2023-10-19:
Hi, can you elaborate on how you fixed this, i am facing the same issue. Where exactly in the yaml did you put those lines of code?