esphome/issues #4924: Waveshare ssd1351_spi, set_brightness bootloop

3ative, 2023-09-29:
### The problem

Using a lambda to call "set_brightness "in a "Button" works fine.

However, calling the same from "Number" prevents proper boot and panics the device

### Which version of ESPHome has the issue?

2023.9.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

ESP32

### Board

esp32dev

### Example YAML snippet

```yaml
This works fine:

button:
  - platform: template
    id: b_button
    name: Brightness
    on_press:
      - lambda: |-
          id(oled).set_brightness(0.8);

This panics it:
```yaml
number:
  - platform: template
    id: oled_brightness
    name: Brightness
    optimistic: true
    min_value: 0
    max_value: 1
    step: 0.1
    initial_value: 0.5
    on_value:
      - lambda: |-
          id(oled).set_brightness(id(oled_brightness).state);
```
```

### Anything in the logs that might be useful for us?

```txt
Serial logs:
ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
[I][logger:268]: Log initialized
[C][ota:473]: There have been 1 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[D][spi:039]: Setting up SPI bus...
[D][number:012]: 'Number1': Sending state 65.000000
[D][number:012]: 'Number2': Sending state 22.000000
[D][number:012]: 'Brightness': Sending state 0.500000
[E][spi:078]: SPIDevice not initialised - did you call spi_setup()?
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40184fef  PS      : 0x00060230  A0      : 0x801852ac  A1      : 0x3ffb2510  
A2      : 0x3ffba694  A3      : 0x00000001  A4      : 0x3ffb252d  A5      : 0x00000001  
A6      : 0x007bdc80  A7      : 0x003fffff  A8      : 0x00000010  A9      : 0x3ff44000  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000004  A13     : 0x3f401ca0  
A14     : 0x00060620  A15     : 0x00000001  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008a8c9  LEND    : 0x4008a8d9  LCOUNT  : 0xffffffff  

Backtrace:0x40184fec:0x3ffb25100x401852a9:0x3ffb2530 0x400dcb06:0x3ffb2550 0x400dc595:0x3ffb2570 0x400e1fc2:0x3ffb2590 0x400e1faa:0x3ffb25b0 0x40185737:0x3ffb25e0 0x40185793:0x3ffb2610 0x400daa29:0x3ffb2630 0x400dce0a:0x3ffb2680 0x400dce16:0x3ffb26b0 0x400e0fa6:0x3ffb26d0 0x40185529:0x3ffb2720 0x400e0b5c:0x3ffb2740 0x400e34ae:0x3ffb2770 0x400e827a:0x3ffb2820
```

xdecock, 2023-09-29:
This seems linked to an [spi refactor](https://github.com/esphome/esphome/pull/5311), from what i see there is a work-around that could go quite fast for you (and looks like the "correct way" to do it:

set_brightness write on the spi bus (which is not yet started (which explain your problem))

you should change your lambda for:

```yaml
    on_value:
      - lambda: |-
          if (id(oled).is_ready()) { id(oled).set_brightness(id(oled_brightness).state); } else { id(oled).init_brightness(id(oled_brightness).state); }
```

it's probably possible to patch up the set_brightness to do that natively.

let me know if this works for you.

3ative, 2023-09-29:
@xdecock BOO-YA! there you go, nice one... thanks for the info

I re-wrote my code a bit (I wanted a 0-100% slider).
Here's what I have now and it works a treat:
```yaml
number:
  - platform: template
    id: oled_brightness
    name: Brightness
    optimistic: true
    min_value: 0
    max_value: 100
    step: 5
    initial_value: 50
    on_value:
      - lambda: |-
          if (id(oled).is_ready()) {
            id(oled).set_brightness(id(oled_brightness).state/100);
            } else {
              id(oled).init_brightness(id(oled_brightness).state/100);
            }
```

Also, any idea why other OLEDs were changed to use "set_contrast", a few ESPHomes ago, and yet this still uses "set_brightness" ??

xdecock, 2023-09-29:
I'm just discovering the code inside esphome, and as i can help a bit, i do provide support, i currently have on PR open with 20-30 lines, so no expert, i'll try to fix the root cause of the issue for your screen. if you could test it once done, i'll be happy (as i don't have the screen at hand)

3ative, 2023-09-29:
Oh, soz... I thought you were a DEV - you replied so quickly with something that worked. 

I'm still testing stuff on my screen here (for a tutorial/project). But no worries if you want to send code for me to test on it. 
Here is not the best option to continue on that. Come join my DISCORD server:  https://discord.com/invite/3RbZgvHeJ9

xdecock, 2023-09-29:
Could you give a try to this?

```yaml
external_components:
  # equivalent shorthand for GitHub pull request
  - source: github://pr#5454
    components: [ ssd1351_base, ssd1351_spi ]
```

3ative, 2023-09-29:
> Could you give a try to this?
> 
> ```yaml
> external_components:
>   # equivalent shorthand for GitHub pull request
>   - source: github://pr#5454
>     components: [ ssd1351_base, ssd1351_spi ]
> ```

Tried. Allows my orig code:
```yaml
number:
  - platform: template
    id: oled_brightness
    name: Brightness
    optimistic: true
    min_value: 0
    max_value: 100
    step: 5
    initial_value: 50
    on_value:
      - lambda: |-
          id(oled).set_brightness(id(oled_brightness).state);
 ```
 However, brightness doesn't change on screen.

3ative, 2023-09-29:
Opps, forgot to divide by 100.... 
```yaml
id(oled).set_brightness(id(oled_brightness).state/100);
```

## Comfirmed: WORKS!

3ative, 2023-09-29:
Now all we need is 0 Brightness to turn off the OLED! LOL

xdecock, 2023-09-29:
That's a feature request ^^ and i'm not sure i have the patience to do that
kind of dev blind

closed 2023-09-29 by kbx81