esphome/issues #4925: stacktrace not decoded

stleusc, 2023-09-29:
### The problem

I am not necessarily looking for answers WHY my code crashes, I am only looking for advice how to decode the trace. I see many logs where it shows some form of decoded trace. I am not getting that!
I have the 'debug:' option in the yaml file. Do I need anything else?
Like I said, happy to debug it but with that trace it is hard.

It sounds like it should automatically do that.

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest, but unrelated

### What platform are you using?

ESP32

### Anything in the logs that might be useful for us?

```txt
ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:268]: Log initialized
[C][ota:473]: There have been 9 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[W][app:022]: Component <unknown> already registered! (0x3ffce0c4)
[W][app:022]: Component custom_component already registered! (0x3ffce0c4)
[I][app:029]: Running through setup()...
[D][binary_sensor:034]: 'Status': Sending initial state OFF
[D][text_sensor:064]: 'ESPHome Version': Sending state '2023.9.1 Sep 29 2023, 13:54:47'
[C][esp32_ble:027]: Setting up BLE...
[D][esp32_ble:043]: BLE setup complete
[D][esp32_ble_tracker:246]: Starting scan...
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: xx:xx:xx:xx:xx:xx
[I][wifi:277]: WiFi Connecting to 'ssid'...
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e7233  PS      : 0x00060130  A0      : 0x801fd880  A1      : 0x3ffcd780  
A2      : 0x3ffce0c4  A3      : 0x3ffcd7e4  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffb7fe0  A7      : 0x00000000  A8      : 0x40092e1c  A9      : 0x3ffcd760  
A10     : 0x00000362  A11     : 0x3f4042ff  A12     : 0x00000362  A13     : 0x00005580  
A14     : 0x00060120  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40092e8c  LBEG    : 0x400845ad  LEND    : 0x400845b5  LCOUNT  : 0x00000027  

Backtrace:0x400e7230:0x3ffcd7800x401fd87d:0x3ffcd7a0 0x400e541f:0x3ffcd7c0 0x400e4002:0x3ffcd810 0x400e6cd0:0x3ffcd840 0x400f135a:0x3ffcd970 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

....
```

ssieb, 2023-09-29:
You need to be watching the logs directly from esphome on the computer that is running esphome, not using web serial.  So either plugged directly into the server or running esphome locally on your computer.
There is another way using the stack trace data and the firmware elf file, but that's somewhat complicated and I can't explain how to do that.

stleusc, 2023-09-29:
@ssieb I have been watching it with usb serial.... I did not see anything.
I assumed there is a way to use the elf file. Do you happen to know where to find the instructions?

ssieb, 2023-09-29:
I assumed that's what you were doing, but it has to be the esphome program itself reading the logs.
Search "esp stack trace decoder".
I did find https://maximeborges.github.io/esp-stacktrace-decoder/ which might work.

ssieb, 2023-09-29:
But github issues are not for support requests.  If you have more questions, please ask on the esphome discord server.

closed 2023-09-29 by ssieb

stleusc, 2023-09-29:
Thank you!
I was not using esphome program for it though. But your link looks promising!