esphome/issues #4926: 2023.9.1 fresh install gets status code 500 when adding first device

aleksrozman, 2023-09-29:
### The problem

Fresh installation of the ESPHome addon is failing, the error message to the user is visible as received error 500 (presented as a red warning on the screen post device selection).

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP32

### Anything in the logs that might be useful for us?

```txt
Home Assistant Core Logs

[homeassistant.components.esphome.dashboard] Error requesting ESPHome Dashboard data: 500, message='Internal Server Error', url=URL('http://127.0.0.1:_XXXX_/devices')
[homeassistant.components.esphome.dashboard] Dashboard unavailable; skipping reauth: 500, message='Internal Server Error', url=URL('http://127.0.0.1:_XXXX_/devices')

ESPHome Logs 

  File "/esphome/esphome/util.py", line 265, in list_yaml_files
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
  File "/esphome/esphome/util.py", line 265, in <listcomp>
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome'
```

### Additional information

This is resolved by manually creating the /config/esphome directory. The install was done fresh on HASSIO 2023.9.1, and likely this will get resolved during the course of development so I am noting this in case someone comes across it and needs a workaround solution. Let me know if I should spend a cycle and apply a fix to handle the FileNotFoundError with a `mkdir` fallback.

Razzer09, 2023-09-30:
Im having the same problem

bertusras, 2023-09-30:
I feel like this part of your original post should be in bold and underlined as I read right across it the first time, but it did solve my issue, thank you so much!

**This is resolved by manually creating the /config/esphome directory.**

RJournee, 2023-09-30:
Wauw what a simple solution for this annoying problem that i was facing since I got my M5stack and first install of ESPhome yesterday. Thank you very much!

iztokR360, 2023-10-01:
Hello,
i am a complete beginner in this field but i have the same problem with connecting esp8266 to home assistant. It always shows me an error (500).
The home assistant is ver. 2023.9.3.
How do I change the directory manually??
Please help me !

RJournee, 2023-10-01:
I added the directory with the (add-on) file editor.

iztokR360, 2023-10-01:
Thank you very much for the answer.How do I enter a new directory in the file editor?Do I open a new directory (which address), or add a new line to the config directory?If possible, can you send me a screenshot of this directory.

Best Regards.
Iztok

Razzer09, 2023-10-01:
[image]
[image]

iztokR360, 2023-10-01:
Thanks for your help, you solved my problem. The thing works. I didn't know it was so simple.

closed 2023-10-02 by jesserockz