esphome/issues #4929: ESPHome 2023.9.1 Apparently Cripples SPI / Display on RP2040

TDStrasser, 2023-09-30:
### The problem

The 7789V display driven by the SPI bus takes upwards of 6 or 7 seconds to update.  I have tried using the software compatible 7735 display configuration with the same result.
This is indicative of a very slow SPI bus speed (or related problem).
As a consequence, the RP2040 is held up and does not update its connection to the Home Assistant server and all dashboard entities are shown as unavailable, even though the RP2040 is running (very slowly).
Removing the display code from the configuration restores normal speed of operation (minus the display).
[logs_esphome-hydroponics_run (2).txt](https://github.com/esphome/issues/files/12773928/logs_esphome-hydroponics_run.2.txt)
[esphome-hydroponics (1).zip](https://github.com/esphome/issues/files/12773938/esphome-hydroponics.1.zip)

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.1

### What platform are you using?

RP2040

### Board

Pico-W

### Component causing the issue

SPI

### Example YAML snippet

```yaml
See attached zipped yaml script
```

### Anything in the logs that might be useful for us?

```txt
[13:59:49][W][component:204]: Component st7789v.display took a long time for an operation (6.84 s).
[13:59:49][W][component:205]: Components should block for at most 20-30ms.

Full log is attached.
```

kbx81, 2023-09-30:
Do you know about how long it took to refresh the display prior to this release? I suspect it has always been somewhat slow on the RP2040 as, when re-working the SPI component, we realized that SPI is not being handled by hardware but instead by software (bit-banging) on the RP2040 platform. This is not optimal for displays as software SPI maxes out at a few hundred kHz (for comparison, hardware SPI can reach clock speeds as high as 120 MHz on the ESPs).

TDStrasser, 2023-09-30:
Previously this, which is tolerable:
[00:00:00][W][component:204]: Component st7789v.display took a long time for an operation (0.79 s).
Now 10 times as long which is not tolerable.
I tried setting the SPI to hardware but it isn't supported.  Using MicroPython directly on the RP2040 SPI runs at MHz speeds. The hardware SPI exists and is accessible - see https://docs.micropython.org/en/latest/rp2/quickref.html#hardware-spi-bus.  It could be the underlying arduino port code that isn't using the hardware SPI (an educated guess).

TDStrasser, 2023-09-30:
See also https://arduino-pico.readthedocs.io/en/latest/spi.html# which definitely uses the hardware SPI bus on the RP2040. https://arduino-pico.readthedocs.io/en/latest/spi.html#

kbx81, 2023-09-30:
Ah nice, that's promising. So I guess now we just need someone who has a Pico-W to test with and who can create a PR. 😄

TDStrasser, 2023-09-30:
I'm a novice at this, but looking at the ESPHome component repository (https://github.com/esphome/esphome/tree/dev/esphome/components/spi)
/__init__.py, the valid rp2040 SPI hardware pins appear to not be defined and therefore the YAML is forced to use the software SPI bus.
The Pico-W pinout is here https://www.raspberrypi.com/documentation/microcontrollers/images/picow-pinout.svg

clydebarrow, 2023-09-30:
Yeah right now there is only software SPI for the RP2040. I do have hardware to test on but was unable to get any ESPHome build to run on the RP2040 when I was doing the PR so put it in the Too Hard basket at the time.

Will take another look next week. I don't expect it to be too hard once I can actually get some code running.

TDStrasser, 2023-09-30:
Hardware SPI would be great. No explanation though as to why the software SPI seems to have radically slowed with this update.

clydebarrow, 2023-09-30:
I think the hardware SPI was working before, now it's not. At the time of the SPI rewrite I was unable to test RP2040 and since "Support for all aspects of ESPHome on the RP2040 is still in development" I decided, with @kbx81 's agreement (see how I shared the blame there :-) to defer that as we thought it would not impact too many people. The software SPI is platform-independent.

Do you have a basic working RP2040 YAML I can start with? The config in https://esphome.io/components/rp2040.html did not work for me. It'll be a couple of days before I can get to it though.

TDStrasser, 2023-10-01:
Hi Clyde. You'll get a working basic configuration by following the New Device dialogue in the ESPHome tabe of the Home Assistant web interface.  The encryption keys etc are unique to each instance.
I posted a working YAML in a zip file uploaded when I started this issue post. You can pare that back to what you need.
A basic YAML for the RP2040 with a st7789v display module looks like the yaml in the attachment but won't work unless you register the encryption key and password for your instance. The zip file also includes the Helvetica.ttf font file neeed to write text to the display.
For hardware I am using a Pico-W mounted on a DFRobot DFR0848 Pico breakout board https://www.dfrobot.com/product-2393.html and DFR0664 Fermion: 2.0" 320x240 IPS TFT LCD Display with MicroSD Card https://www.dfrobot.com/product-2071.html.  The GDI interface cable between the breakout board and the display uses hardware SPI0 on the Pico-W per the pins in the yaml script.
Interested to see how you get on.
[esphome-display-test package.zip](https://github.com/esphome/issues/files/12775822/esphome-display-test.package.zip)

TDStrasser, 2023-10-02:
@clydebarrow thanks for that. I'm not familiar with how to get the changed file onto my system to try out, so I guess I'll have to wait for the next release.

clydebarrow, 2023-10-02:
Easy-peasy - just add this to your Yaml:

```
external_components:
  source: github://pr#5466
  components: [ spi ]
```

See https://esphome-docs.web.app/components/external_components.html for more info, but the above is all you need.

closed 2023-10-02 by jesserockz