esphome/issues #4931: Cannot upgrade dimmer Firmware to 2023.9.1

tchup, 2023-09-30:
### The problem

After upgrading ESPHome to 2023.9.1 in Home assistant, I tried upgrading dimmer firmware without success.
In ESPHome dashboard, the update gives :
"src/RBDdimmer.h:11:11: fatal error: esp8266/RBDmcuESP8266.h: No such file or directory"

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

dimmer

### Example YAML snippet

```yaml
esphome:
  name: dimmer
  friendly_name: Dimmer
  includes: RBDdimmer.h
  

esp8266:
  board: nodemcuv2
  
  

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "hidden"

ota:
  password: "hidden"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.27
    gateway: 192.168.0.1
    subnet: 255.255.248.0
  power_save_mode: none
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dimmer Fallback Hotspot"
    password: "hidden"

captive_portal:

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: D2
    zero_cross_pin:
      number: D1
      mode:
        input: true
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer1
    name: Dimmerized Light
```

### Anything in the logs that might be useful for us?

```txt
Updating /config/esphome/dimmer.yaml
------------------------------------------------------------

INFO ESPHome 2023.9.1
INFO Reading configuration /config/esphome/dimmer.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing dimmer (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
Compiling .pioenvs/dimmer/src/main.cpp.o
Compiling .pioenvs/dimmer/libed9/ESPAsyncTCP-esphome/SyncClient.cpp.o
Compiling .pioenvs/dimmer/libed9/ESPAsyncTCP-esphome/tcp_axtls.c.o
Compiling .pioenvs/dimmer/libaf0/Hash/Hash.cpp.o
Compiling .pioenvs/dimmer/lib67b/ESP8266WiFi/BearSSLHelpers.cpp.o
In file included from src/main.cpp:31:
src/RBDdimmer.h:11:11: fatal error: esp8266/RBDmcuESP8266.h: No such file or directory
   11 |  #include "esp8266/RBDmcuESP8266.h"
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/dimmer/src/main.cpp.o] Error 1
```

leoshusar, 2023-10-02:
Didn't you perhaps have the RBDmcuESP8266 library in the `.esphome` directory? It's no longer exposed into HA.

tchup, 2023-10-02:
I don't recall putting anything in `.esphome` manually, and this directory does not exist anymore.
Should I put something in the `/data` folder?

leoshusar, 2023-10-02:
What do you have in your `esphome/libraries`?

ssieb, 2023-10-02:
This is not an esphome issue because it involves some sort of external custom component.
If you want help, come ask on the esphome discord server or one of the forums.

closed 2023-10-02 by ssieb