esphome/issues #4933: can't set brightness with st7789v display

hmax-72, 2023-10-01:
### The problem

After upgrading to 2023.9.1 it's not possible to set the brightness of the display. It stays on 100% and cannot be changed.

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP32

### Board

TTGO-T1 ESP32

### Component causing the issue

st7789v display

finnjet, 2023-10-02:
I am having the same issue.

output:
  - platform: ledc
    # D2 = GPIO4 Backlight LED for this device
    pin: GPIO4
    id: ledc_gpio4
    
light:
  - platform: monochromatic
    output: ledc_gpio4
    name: Display Backlight

Used to work with the backlight option in the display section not set
display:
  - platform: st7789v
    model: TTGO TDisplay 135x240
    # backlight_pin: GPIO4
[..]

Since 9.1 or maybe 9.0 it does not work anymore. 9.2 does not fix it.

If I configure a switch however:
switch:
   - platform: gpio
    pin: GPIO4
   name: ${name_prefix} Display Backlight Switch

It can be commanded on and off but without dimming obviously.

clydebarrow, 2023-10-03:
Working for me:
```
output:
  - platform: ledc
    pin: GPIO46
    id: ledc_gpio4

light:
  - platform: monochromatic
    output: ledc_gpio4
    name: Display Backlight
    id: light_1

display:
  - platform: st7789v
    model: LILYGO_T-EMBED_170X320
    height: 320
    width: 170
    offset_height: 35
    offset_width: 0
    dc_pin: GPIO13
    reset_pin: GPIO9
    lambda: |-
      auto rand_col = Color::random_color();
      it.filled_rectangle(0, 0, it.get_width() - 1, it.get_height() - 1, rand_col);
      it.print(it.get_width()/2, it.get_height()/2, id(roboto), TextAlign::CENTER, "Centered");

interval:
  - interval: 5s
    then:
      - light.turn_on:
          id: light_1
          brightness: 10%
          transition_length: 1s
      - delay: 2s
      - light.turn_on:
          id: light_1
          brightness: 80%
          transition_length: 1s
```

hmax-72, 2023-10-03:
Upgrade to 9.3.2 and tried the code from clydebarrow. I just changed the model specific settings (I'm using "model: TTGO TDisplay 135x240").
Brightness still not working (and was working before).

clydebarrow, 2023-10-03:
So is the LED on or not? The backlight has nothing to do with the display itself, it's just a GPIO pin driving a LED.

If you're using the `ledc` platform, try setting the frequency to 10Hz and you should see the backlight flashing at a 10Hz rate.

How are you setting the brightness?

hmax-72, 2023-10-03:
Hi!
The backlight of the display is at 100%. Maybe, it would be possible to turn it off (did not try), but I want to set the brightness, not just on and off.
That's the snippets out of my code which are relevant for backlight and brightness which worked before the update:

```
output:
  - platform: ledc
    pin: GPIO4
    id: gpio4
	
light:
  - platform: monochromatic
    output: gpio4
    id: backlight
    name: "Backlight"
	
sensor:
  - platform: adc
    pin: GPIO36
    id: bedwatch_brightness
    name: "BedWatch_Helligkeit"
    update_interval: 1s
    attenuation: 0db 
    raw: True
	
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO37
      mode:
        input: true
    name: BedWatch_Touch
    on_click:
      min_length: 50ms
      max_length: 3000ms
      then:
        - light.turn_on:
            id: backlight
            brightness: !lambda |-
              float brt = 0;
              // output value must be in range 0 - 1.0
              if (id(bedwatch_brightness).state > 1000) {
                brt = 1.0;
                }
              if (id(bedwatch_brightness).state < 1000) {
                brt = 0.8;
                }
              if (id(bedwatch_brightness).state < 500) {
                brt = 0.6;
                }
              if (id(bedwatch_brightness).state < 100) {
                brt = 0.4;
                }
              if (id(bedwatch_brightness).state < 50) {
                brt = 0.3;
                }
              //ESP_LOGD("main", "bright: %3.2f, light: %5.2f", brt, id(bedwatch_brightness).state );
              return brt;
        - delay: 10s
        - light.turn_off:
            id: backlight

display:
  - platform: st7789v
    model: TTGO TDisplay 135x240
    rotation: 90°
    update_interval: 1s
    cs_pin: GPIO5
    dc_pin: GPIO16
    reset_pin: GPIO23
    backlight_pin: GPIO4 _(added, after brightness was not working any more)_
    id: wecker_display
    pages:
      - id: mainpage
          ...........
```

clydebarrow, 2023-10-03:
If you specify `backlight_pin` in the display config it will always set it simply on.

hmax-72, 2023-10-03:
removed `backlight_pin` but the behaviour did not change.
Also turning the backlight completely off does not work (the interval is checking, if another interaction is in progress and turns off the backlight after ~20secs inactivity). As mentioned before, this was working like a charme with an older version of ESPHome.

```
interval:
  - interval: 20s
    id: displaypageinterval
    then:
      - if:
          condition:
            not:
              lambda: "return id(interval_pause);"
          then:
            - display.page.show: mainpage
            - component.update: wecker_display
            - light.turn_off:
                id: backlight
```

clydebarrow, 2023-10-03:
> Looks to me as if the behavior of the display changed such that GPIO4 is now the default and will be used when nothing else is specified.

I think you hit the nail on the head. Fix coming shortly. Curiously the board I've been testing with also has a backlight pin specified as default, but I'm not seeing the same behaviour.

clydebarrow, 2023-10-03:
PR raised to allow `backlight_pin: no`. For now just set it to another pin.

clydebarrow, 2023-10-03:
There is sadly no check done that a a pin is used in two different places :-(

closed 2023-10-04 by kbx81