esphome/issues #4934: Http_request on idf platform

Melkor82, 2023-10-01:
### The problem

Since pr [https://github.com/esphome/esphome/pull/3500] is closed web server is now working on idf platform but http_request still trow an error say that does not exist on esp32 idf platform 

> This feature is incompatible with esp-idf

### Which version of ESPHome has the issue?

2021.9.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

esp32 dev kit

### Component causing the issue

http_request

### Example YAML snippet

```yaml
esphome:
  name: test
  platformio_options:
    upload_speed: 512000

esp32:
  board: esp32dev
  framework: 
    type: esp-idf
    version: 5.1.1
    platform_version: 6.4.0

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
  port: 80

http_request:
  useragent: esphome/device
  timeout: 10s
  id: http_data
```

Melkor82, 2023-10-01:
i saw that there is already a pr for it

guillempages, 2024-07-21:
http_request available on ESP-IDF since the 2024.6.0 release

closed 2024-07-21 by guillempages