esphome/issues #4935: bk72xx - not supporting fastled_clockless

MnM001, 2023-10-01:
### The problem

Have a device with BK72xx (BK7231T chipset) - Globe Fairy Lights.
Doing the yaml for it it I get the error below:

[image]

Is this going to supported soon?

Also I cant select BK72xx for the platform below so ESP32 was selected instead so I can progress with this bug report.

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP8266

### Board

Tuya Globe Fairy Lights

### Component causing the issue

bk72xx:

### Example YAML snippet

```yaml
bk72xx:
  board: generic-bk7231t-qfn32-tuya

light:
  - platform: fastled_clockless
    chipset: WS2812b
    pin: D4
    num_leds: 115
    rgb_order: RGB
    name: "TestFairyLights"
```

ssieb, 2023-10-01:
fastled doesn't support that platform, so there's nothing esphome can do.  The most likely solution will be a new component like the `ESP32 RMT` or `RP2040 PIO` ones.

Fredaikis, 2023-11-03:
Are we certain about that? I don't see why it should not just compile if we were to add correct handling to require_framework_version