esphome/issues #4936: After upgrading to 2023.9.1 it's not possible to view the logs remote over wifi

larsdunemark, 2023-10-01:
### The problem

After upgrading to latest version of ESPHome it can't connect to any logs on any of my devices over wifi.

If I rebuild a device and flash it OTA it will flash and then boot and send the logs to ESPHome. But after disconnecting it is not possible to reconnect instead I get the following log:

```
INFO ESPHome 2023.9.1
INFO Reading configuration /config/esphome/heatpump.yaml...
INFO Starting log output from heatpump.local using esphome API
WARNING Can't connect to ESPHome API for heatpump.local: Unexpected error while connecting to heatpump.local: 'AsyncServiceInfo' object has no attribute 'addresses_by_version' (APIConnectionError)
INFO Trying to connect to heatpump.local in the background
```

Not sure if it related but in my network there is a IPv6 enabled but I haven't configured anything on the ESPHome devices to use IPv6 and disable IPv6 on HA don't change anything.

I got the problem before flashing the new firmware to the board.

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32

### Example YAML snippet

```yaml
esphome:
  name: "heatpump"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxx"

ota:
  password: "xxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heatpump Fallback Hotspot"
    password: "xxxxxxxx"

captive_portal:

dallas:
  - pin: 32
    update_interval: 10s

sensor:
  - platform: dallas
    address: 0x79xxxxxx
    name: "temp1"
  - platform: dallas
    address: 0x79xxxxxx
    name: "temp2"
  - platform: dallas
    address: 0x79xxxxxx
    name: "temp3"
  - platform: dallas
    address: 0x79xxxxxx
    name: "temp4"
  - platform: wifi_signal
    name: "Wi-Fi Signal"
    update_interval: 60s

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IP Address"
    ssid:
      name: "Wi-Fi SSID"
    bssid:
      name: "Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true

esp32_ble_tracker:
bluetooth_proxy:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.9.1
INFO Reading configuration /config/esphome/heatpump.yaml...
INFO Starting log output from heatpump.local using esphome API
WARNING Can't connect to ESPHome API for heatpump.local: Unexpected error while connecting to heatpump.local: 'AsyncServiceInfo' object has no attribute 'addresses_by_version' (APIConnectionError)
INFO Trying to connect to heatpump.local in the background
```

larsdunemark, 2023-10-01:
Looks like it can be related to https://github.com/esphome/aioesphomeapi/pull/559

bdraco, 2023-10-18:
Fixed by https://github.com/esphome/esphome/pull/5557