esphome/issues #4937: debug component fails when compiling withe 2023.9.1

lux4rd0, 2023-10-01:
### The problem

When providing for debugging I get this error on compiling:

```
src/esphome/components/debug/debug_component.cpp: In member function 'virtual void esphome::debug::DebugComponent::dump_config()':
src/esphome/components/debug/debug_component.cpp:152:23: error: 'CHIP_FEATURE_EMB_PSRAM' was not declared in this scope
   if (info.features & CHIP_FEATURE_EMB_PSRAM) {
                       ^~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/debug/debug_component.cpp:152:23: note: suggested alternative: 'CHIP_FEATURE_EMB_FLASH'
   if (info.features & CHIP_FEATURE_EMB_PSRAM) {
                       ^~~~~~~~~~~~~~~~~~~~~~
                       CHIP_FEATURE_EMB_FLASH
*** [.pioenvs/esp22/src/esphome/components/debug/debug_component.cpp.o] Error 1
```

Was working okay with 2023.9.0

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

LILYGO T5-4.7

### Example YAML snippet

```yaml
substitutions:
  devicename: esp22
esphome:
  name: '${devicename}'
  platformio_options:
    board_upload.flash_size: 16MB
esp32:
  board: m5stack-core2
  framework:
    type: arduino
    version: 2.0.2
    platform_version: '4.3.0'
debug:
  update_interval: 60s

sensor:
  - platform: debug
    free:
      name: '${devicename} Heap Free'
    block:
      name: '${devicename} Heap Max Block'
    loop_time:
      name: '${devicename} Loop Time'
text_sensor:
  - platform: debug
    device:
      name: '${devicename} Device Info'
      id: '${devicename}_device_info'
```

brixharlem, 2024-01-03:
Have you found any solution? I'm facing the same issue.

lux4rd0, 2024-01-03:
If I remove all my debugging components it works fine. But I like seeing them... :(