esphome/issues #4938: docker based lint does not work

trixpan, 2023-10-01:
### The problem

The documentation points new contributors to run

```
docker run --rm -v "${PWD}/":/esphome -it ghcr.io/esphome/esphome-lint script/fulltest
```

To test changes locally, however the script contains a number of issues

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

N/A

### Component causing the issue

N/A

### Anything in the logs that might be useful for us?

```txt
$ docker run --rm -v "${PWD}/":/esphome -it ghcr.io/esphome/esphome-lint script/fulltest
+ script/ci-custom.py
fatal: detected dubious ownership in repository at '/esphome'
To add an exception for this directory, call:

        git config --global --add safe.directory /esphome
+ script/lint-python
fatal: detected dubious ownership in repository at '/esphome'
To add an exception for this directory, call:

        git config --global --add safe.directory /esphome
+ script/lint-cpp
+ script/clang-tidy --fix --all-headers

        Oops. It looks like clang-tidy-14 is not installed.

        Please check you can run "clang-tidy-14 -version" in your terminal and install
        clang-tidy (v14) if necessary.

        Note you can also upload your code as a pull request on GitHub and see the CI check
        output to apply clang-tidy.
        
+ script/clang-format -i
fatal: detected dubious ownership in repository at '/esphome'
To add an exception for this directory, call:

        git config --global --add safe.directory /esphome
  [##############################]  100%
+ script/unit_test
+ pytest tests/unit_tests

...
==================================================================== 21 passed in 1.97s =====================================================================
+ script/test
+ esphome compile tests/test1.yaml
script/test: line 9: esphome: command not found
```
```

xdecock, 2023-10-02:
If you use vs code the devcontainer runs quite well, but i indeed never managed to run the docker. Might be worth updating the doc

functionpointer, 2023-11-07:
Try using the `dev` tag:
````
docker run --rm -v "${PWD}/":/esphome -it ghcr.io/esphome/esphome-lint:dev script/fulltest
````

pebblebed-tech, 2024-01-04:
This now also broken in the devcontainer using vscode. The clang task fails as it can not find clang 14