esphome/issues #4940: Dashboard no longer displays "Update Available" in esphome docker container installation

JackTalisker, 2023-10-02:
### The problem

After updating my docker container from 2023.8.3 to 2023.9.2 the "update available" option is missing.
files in config directory (/home/pi/esphome) are all there.

### Which version of ESPHome has the issue?

2023.9.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.9.3 core

### What platform are you using?

ESP8266

### Board

nodemcu, various sonoffs, esp32-wroom, etc.

### Component causing the issue

n/a

### Example YAML snippet

```yaml
none
```

### Anything in the logs that might be useful for us?

```txt
none
```

### Additional information

my docker compose esphome snippet:

```
  esphome:
    image: esphome/esphome
    volumes:
      - "/home/pi/esphome:/config"
      # Use local time for logging timestamps
      - /etc/localtime:/etc/localtime:ro
    network_mode: host
```
files in config directory (/home/pi/esphome) are all there.

Roemer, 2023-10-08:
Did you do anything else? I got the same problem after updating from 2023.8.3 directly to 2023.9.3.

JackTalisker, 2023-11-03:
> Did you do anything else? I got the same problem after updating from 2023.8.3 directly to 2023.9.3.

Sorry for the late reply.
This time update with the "install" option (click on the three dots).
Wait for the next esphome update and the "update available" will appear again.