esphome/issues #4941: Pulse Meter Sensor does not set Value

Oleg-Sob, 2023-10-02:
### The problem

Pulse Meter Sensor does not set Value. Everything works in ESPHome (dev).

### Which version of ESPHome has the issue?

2023.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2023.9.3

### What platform are you using?

ESP8266

### Board

nodemcu

### Example YAML snippet

```yaml
- platform: template
    name: "Reset 0"
    on_press:
      - pulse_meter.set_total_pulses:
         id: pulse_meter_id
         value: "0"
```

ssieb, 2023-10-02:
What is the problem?  You even say "everything works".  What about logs?

Oleg-Sob, 2023-10-03:
Runs in ESPHome (dev). ESPHome is not working.

ssieb, 2023-10-03:
That doesn't make sense and you haven't provided enough information to be useful.

Oleg-Sob, 2023-10-03:
sensor:
  - platform: pulse_meter
    id: pulse_meter_id
    pin: D1
    name: 'Tracker in min'
    internal_filter: 100ms
    total:
      unit_of_measurement: 'tr'
      name: 'Tracker'

button:
  - platform: template
    name: "Tracker in  0"
    on_press:
      - pulse_meter.set_total_pulses:
         id: pulse_meter_id
         value: "0"

The value is not set to 0,  in previous versions of ESPHome it was set to 0.

ssieb, 2023-10-03:
And logs?

Oleg-Sob, 2023-10-03:
[08:02:41][D][button:013]: 'Tracker in 0' Pressed.
[08:02:41][D][sensor:110]: 'Tracker': Sending state 0.00000 tr with 0 decimals of accuracy
This is in the ESPHome (dev) version.

[08:05:45][D][button:013]: 'Tracker in 0' Pressed.
This is in the ESPHome version.

There is no line "[08:02:41][D][sensor:110]: 'Tracker': Sending state 0.00000 tr with 0 decimals of accuracy" in the ESPHome version.

ssieb, 2023-10-03:
What version is the thing you're calling "dev"?

Oleg-Sob, 2023-10-03:
https://github.com/esphome/esphome/releases/tag/2023.9.0b4

ssieb, 2023-10-03:
That's not `dev`, that was one of the `beta` releases.  There was a pulse meter change just before that but nothing since then.
But also, it has never published the value either after setting it, so I don't know where that line came from.

ssieb, 2023-10-03:
Ok, I see what's going on, but I think you have the wrong release that worked.  The changes were before that.
@TrentHouliston https://github.com/esphome/esphome/pull/4231 changed the `loop()` so that the total was published only if the count changed, not if the total changed.

Oleg-Sob, 2023-10-03:
What do you recommend? I need the value to be published immediately after it is changed. It worked before.

ssieb, 2023-10-03:
If you give the total sensor an `id:`, you can publish the value yourself after changing it using a lambda.
`id(total_id).publish_state(0);`

TrentHouliston, 2023-10-03:
Now that I think about it, it does make sense to publish the total value when you set the total value.

Otherwise, you'd never actually see a sensor update with the value you set, only values greater than the value you set which for a total increasing sensor may be relevant for accuracy

Oleg-Sob, 2023-10-03:
- platform: pulse_meter
    id: pulse_meter_id
    pin: D1
    name: 'Tracker in min'
    internal_filter: 100ms
    total:
      unit_of_measurement: 'tr'
      id: total_pulse_id
      name: 'Tracker'

  - platform: template
    name: "Tracker in 0"
    on_press:
      - pulse_meter.set_total_pulses:
         id: pulse_meter_id
         value: "0"
      - lambda: id(total_pulse_id).publish_state(0);

I did this, after resetting the value is published. But after a reboot, ESP does not publish the values. How can I make the value 0 appear after a reboot?

Oleg-Sob, 2023-10-03:
```yaml
  on_boot:
    then:
    - lambda: id(total_pulse_id).publish_state(0);
```

Figured it out, added it when loading

closed 2023-10-19 by ssieb