esphome/issues #4944: LD2410 makes OTA updating almost impossible

DerEmder, 2023-10-03:
### The problem

OTA Update works one out of hundred times - it just times out normally. Sometimes it shoots up to 60%, sometimes it goes up to 12% and times out - OTA is impossible. Unfortunately this sensor sits inside a wall so I depend on WiFi connection. I assume I have to open the wall and connect a USB cable for flashing.

There is no webserver or captive portal. Still, the ESP8266 seems to be too busy to acknowledge the OTA transfer.

```
INFO ESPHome 2023.9.3
INFO Reading configuration /config/esphome/letta.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing letta (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
Compiling .pioenvs/letta/src/main.cpp.o
Linking .pioenvs/letta/firmware.elf
RAM:   [=====     ]  47.0% (used 38524 bytes from 81920 bytes)
Flash: [=====     ]  48.4% (used 505835 bytes from 1044464 bytes)
Building .pioenvs/letta/firmware.bin
esp8266_copy_factory_bin([".pioenvs/letta/firmware.bin"], [".pioenvs/letta/firmware.elf"])
========================= [SUCCESS] Took 24.27 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 10.42.0.48
INFO Uploading /data/build/letta/.pioenvs/letta/firmware.bin (509984 bytes)
INFO Compressed to 346923 bytes
Uploading: [=======                                                     ] 12% 
ERROR Error sending data: timed out
```

The yaml I use is identical with the one on the espHome website.

```
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  throttle: 1000ms
```

```
binary_sensor:
  - platform: ld2410
    has_target:
      name: "$upper_devicename Presence"
    has_moving_target:
      name: "$upper_devicename Moving Target"
    has_still_target:
      name: "$upper_devicename Still Target"
    out_pin_presence_status:
      name: "$upper_devicename Out Presence Status"
  - platform: gpio
    pin:
      number: D2
      inverted: false
      mode:
        input: true
        pullup: true
    name: "$upper_devicename Bewegung"
    device_class: motion
    filters:
      - delayed_off: 55s

```

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

LD2410

DerEmder, 2023-10-04:
Thank you for this idea. So I will get thet ESP out of my wall and re-program it via USB. I will let you know how it went later on.

Thanks again.

teamRibcage, 2023-10-04:
I'm not sure if this is the root cause of your issue or not, but have you also included
```
logger:
  baud_rate: 0
```
in your config? I had quite a few errors setting mine up initially - not realizing the module was trying to log over the same UART pins I was using for the LD2410 🙂

DerEmder, 2023-10-04:
Well, in fact I did not include it. But thinking about it - it might be a good idea (but I don't see the relevance according OTA updates and logging via WiFi, though). 

Nevertheless seems to be a good point to have the UART exclusively for the LD2410 instead of logging data. Thanks for pointing that out, I'll keep you updated about the outcome.

charles, 2023-10-04:
I had already included it:

```
logger:
  baud_rate: 0
```
  
and the OTA problem persist to. 
I don't know what is the cause of problem :-(
  
Safe mode is the way I've been able to do OTA updates so far.

DerEmder, 2023-10-04:
Darn, I can't even program via USB, the thing gets hot and does not flash anymore. I guess I have to build another board with the LD2410 and re-flash it. That might take a while - of course the root problem is still there. 'Safe mode' seems to be the way to go for now until the devs fix their code for the LD2410.

I'll keep you updated.

leoshusar, 2023-10-05:
@DerEmder how are you trying to program it? If you hold GPIO0 down after reset it should enter into download mode and you should be able to flash. If this doesn't work, you can also try to disconnect the UART device connected to it.

DerEmder, 2023-10-05:
> @DerEmder how are you trying to program it? If you hold GPIO0 down after reset it should enter into download mode and you should be able to flash. If this doesn't work, you can also try to disconnect the UART device connected to it.

? 

I am talking about OTA programming - not via cable. And I was describing a sensor behind (or more specifically inside) a wall. That's the whole point of using espHome flashing OTA with WiFi.

DerEmder, 2023-10-05:
I tried the 'safe mode' (as described above) without much success - still timeouts. Devs, please do something!

Anyway, the first LD2410 sensor board died with overheat because it seems to very sensitive regarding the Rx and Tx signal lines. 

I used the 8266_dma mode together with the neopixel library to drive one RGB led. I forgot about the fact that it uses the UART Rx for LED data **output**. I assume that might have destroyed the LD2410 chipset by short-circuiting the Tx line of the LD2410. My bad, my mistake.

I guess we have to wait until the LD2410 library gets fixed regarding OTA.

3ative, 2023-10-05:
Just updated mine to 2023.9.3 this morning without issue. 

D1 Mini V3

Here's some boiler-plate I'm using:
```yaml
esphome:
  name: radar-01
  platform: esp8266
  board: d1_mini
  includes:
    - uart_read_line_sensor_ld2410_2v3.h
  on_boot:
    priority: -100
    then:
      - script.execute: get_config

logger:
  baud_rate: 0
  logs:
    sensor: INFO # DEBUG level with uart_target_output = overload!

 uart:
  id: uart_bus
  tx_pin: TX #Yellow
  rx_pin: RX #White
 ```
 
 Everything else is just Binary Sensor & Sensors using the LD2410 serial data.

DerEmder, 2023-10-05:
> @DerEmder you said it does not flash anymore even via USB, so I was suggesting those pins, if you maybe happened to not know about them.

That's true but the d1_mini comes with standard 'Boot + Flash' logic. The reason for not being able to flash is the connected LD2410 which is using the same Rx/Tx lines. Again - my mistake, sometimes I do stuff without thinking :)

DerEmder, 2023-10-05:
> Just updated mine to 2023.9.3 this morning without issue.
> 
> D1 Mini V3
> 
> Here's some boiler-plate I'm using:

Thank you for this - that's a lot of stuff to ingest. I will try that with my setup but first I need to read all that information you linked to. I'll keep you posted.

Electronlibre2012, 2024-04-25:
Hello,

i have a lot of problem with OTA, only solution i found is to desolder the +5V of the LD2410...radical...but works.

I will try the safe_mode restart next time, thanks for the tip @charles

RubenKelevra, 2025-04-24:
While I'm not using the ESP8266 nor the LD2410, I have the same issue.

I'm using an ESP32-C3 in a location where the wifi quality isn't that great. There's quite often some long jitter, as wifi retransmits a package:

```  
--- 192.168.8.40 ping statistics ---
260 packets transmitted, 258 received, 0.769231% packet loss, time 259262ms
rtt min/avg/max/mdev = 4.556/165.341/1236.209/165.135 ms, pipe 2
``` 

This causes the upload to fail, due to a timeout of chunks:

``` 
RAM:   [=         ]  14.2% (used 46476 bytes from 327680 bytes)
Flash: [========  ]  81.8% (used 1500480 bytes from 1835008 bytes)
========================= [SUCCESS] Took 24.57 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.8.40 port 3232...
INFO Connected to 192.168.8.40
INFO Uploading /data/build/xxx/.pioenvs/xxx/firmware.bin (1504080 bytes)
Uploading: [=                                                           ] 1% ERROR Error receiving acknowledge chunk OK: timed out
```

So I think the overall OTA process could use some work, so chunks are retransmitted if no acknowledgement happened, and the timeout per chunk is longer, so a small hiccup isn't causing an abort here.