esphome/issues #4945: esp-idf framework problem?

def1149, 2023-10-03:
### The problem

I tried to set up an ESPHome Bluetooth proxy.

It compiled, linked, and flashed without error, but would not come online and was not visible on my network

I think the issue is related to specifying type: esp-idf framework for the Bluetooth proxy as is called out in the guide.

I tried two different esp32-s3 boards. Both of these were confirmed to work using the Arduino library for other ESPHome devices.

I switched the yaml to specify the Arduino framework, instead of the esp-idf framework, and the Bluetooth proxy came on-line and and appeared as a new device.

I'm not sure if t he BT proxy is working because it has not found any BT devices yet

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP32

### Board

Adafruit S3 Feather and Smart Bee S3

### Component causing the issue

???

def1149, 2023-10-03:
With help from the ESPHome Discord channel, I figured it out. When using the ESP32-S3 with the esp-idf it seems you have to specify DIO flash mode in the YAMAL  

E.g.
esphome:
  name: bluetooth-proxy-3
  friendly_name: Bluetooth Proxy 3
  platformio_options:
    board_build.flash_mode: dio

samuelolteanu, 2023-10-20:
I have an Weact esp32c3 wich does not boot anymore with esp-idf framework. Could be hardware problem because another similar board is online as bt_proxy+temp+ld2410.

fernplant, 2024-07-15:
> With help from the ESPHome Discord channel, I figured it out. When using the ESP32-S3 with the esp-idf it seems you have to specify DIO flash mode in the YAMAL
> 
> E.g.
> 
> ```
> esphome:
>   name: bluetooth-proxy-3
>   friendly_name: Bluetooth Proxy 3
>   platformio_options:
>     board_build.flash_mode: dio
> ```

Thank you! I have a C3 "super mini" board from Aliexpress and when compiled with esp-idf framework it would not connect to wifi (but using arduino, would connect). Specifying dio flash_mode as you posted made it work though! Thanks

closed 2025-05-12 by def1149