esphome/issues #4948: ILI9xxx: ILI9488 not working as expected

carslen, 2023-10-04:
### The problem

I'm using [this ILI9488 display](https://de.aliexpress.com/item/1005001999296476.html?spm=a2g0o.order_list.order_list_main.5.9b3b5c5fs4TTXy&gatewayAdapt=glo2deu#nav-specification) in the _IPS-ILI9488 No Touch_ variant from Aliexpress.

When using the TFT_eSPI Arduino library examples, the display is working with all examples I've tested. My `User_Setup.h` for TFT_eSPI lib looks like this:

```c
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// User defined information reported by "Read_User_Setup" test & diagnostics example
#define USER_SETUP_INFO "User_Setup"

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)

// ##################################################################################
//
// Section 2. Define the pins that are used to interface with the display here
//
// ##################################################################################

// If a backlight control signal is available then define the TFT_BL pin in Section 2
// below. The backlight will be turned ON when tft.begin() is called, but the library
// needs to know if the LEDs are ON with the pin HIGH or LOW. If the LEDs are to be
// driven with a PWM signal or turned OFF/ON then this must be handled by the user
// sketch. e.g. with digitalWrite(TFT_BL, LOW);

#define TFT_BL 32            // LED back-light control pin

// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)

// ##################################################################################
//
// Section 3. Define the fonts that are to be used here
//
// ##################################################################################

// Comment out the #defines below with // to stop that font being loaded
// The ESP8366 and ESP32 have plenty of memory so commenting out fonts is not
// normally necessary. If all fonts are loaded the extra FLASH space required is
// about 17Kbytes. To save FLASH space only enable the fonts you need!

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
[... 256 lines cut ...]
[22:43:59][C][ota:097]: Over-The-Air Updates:
[22:43:59][C][ota:098]:   Address: spielwiese.local:3232
[22:43:59][C][ota:101]:   Using Password.
[22:43:59][C][api:138]: API Server:
[22:43:59][C][api:139]:   Address: spielwiese.local:6053
[22:43:59][C][api:141]:   Using noise encryption: YES
[22:44:04][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=6611 (now=16612)
[22:44:14][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=16611 (now=26612)
[22:44:19][VV][api.service:540]: on_ping_request: PingRequest {}
[22:44:19][VV][api.service:043]: send_ping_response: PingResponse {}
[22:44:24][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=26611 (now=36618)
[22:44:34][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=36611 (now=46612)
[22:44:35][VV][api.service:540]: on_ping_request: PingRequest {}
[22:44:35][VV][api.service:043]: send_ping_response: PingResponse {}
[22:44:39][VV][api.service:540]: on_ping_request: PingRequest {}
[22:44:39][VV][api.service:043]: send_ping_response: PingResponse {}
[22:44:42][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=4294961960 (now=54665)
[22:44:44][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=46611 (now=56612)
[22:44:54][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=56611 (now=66611)
```
```

### Additional information

![23-10-04 22-44-26 6395](https://github.com/esphome/issues/assets/28647218/7c5bf906-8af4-4c84-b82c-bc04bf9341eb)

nielsnl68, 2023-10-05:
To use a ili9488 you need to use a esp32 with **PSRAM**.

carslen, 2023-10-05:
Is this a esphome limitation to use this display? As said, with the Arduino TFT_eSPI library examples, the display is working with the same ESP32 hardware.

nielsnl68, 2023-10-05:
Indeed, esphome uses a buffer to store all the display info.

carslen, 2023-10-06:
OK; tested with an esp32-S3-DevKitC-1 and now the display is working but with kinda inverted colors:

```yaml
display:
  - platform: ili9xxx
    model: ILI9488
    cs_pin: 4
    dc_pin: 6
    reset_pin: 5
    spi_id: test_disp
    lambda: |-
      it.fill(Color::WHITE);
      it.print(240, 0, id(roboto), id(my_white), TextAlign::TOP_CENTER, "Hello World!");
      it.line(20, 28, 460, 28, id(my_blue));
      it.image(40, 50, id(edna));

font:
  # gfonts://family[@weight]
  - file: "gfonts://Roboto"
    id: roboto
    size: 24

# Define a PWM output on the ESP32
output:
  - platform: ledc
    pin: 16
    id: gpio_16_backlight_pwm

# Define a monochromatic, dimmable light for the backlight
light:
  - platform: monochromatic
    output: gpio_16_backlight_pwm
    name: "Display Backlight"
    id: back_light
    restore_mode: ALWAYS_ON

image:
  - file: "fonts/05.jpg"
    id: "edna"
    resize: 200x200
    type: RGB24
```
The blue line looks yellow, and the pic looks like a negative. Any hint on that? I've also tried `ILI9488_A`, but

nielsnl68, 2023-10-07:
Hi, to validate your issue could you do us a favor and run the below script on that display:
``` yaml
display:
  - id: my_display
    platform: ili9xxx
    ...
    auto_clear_enabled: false
    lambda: |-
      int shift_x = (it.get_width()-310)/2;
      int shift_y = (it.get_height()-256)/2;
      for(auto i = 0; i<256; i++) {
        it.horizontal_line(shift_x+  0,i+shift_y,50, my_red.fade_to_white(i));
        it.horizontal_line(shift_x+ 50,i+shift_y,50, my_red.fade_to_black(i));

        it.horizontal_line(shift_x+105,i+shift_y,50, my_green.fade_to_white(i));
        it.horizontal_line(shift_x+155,i+shift_y,50, my_green.fade_to_black(i));

        it.horizontal_line(shift_x+210,i+shift_y,50, my_blue.fade_to_white(i));
        it.horizontal_line(shift_x+260,i+shift_y,50, my_blue.fade_to_black(i));
      }
      it.rectangle(shift_x+ 0, 0+shift_y, shift_x+ 310, 256+shift_y, my_yellow);

color:
  - id: my_blue
    blue: 100%
  - id: my_red
    red: 100%
  - id: my_green
    green: 100%
  - id: my_white
    red: 100%
    blue: 100%
    green: 100%
  - id: my_yellow
    hex: ffff00
```

thanks.

carslen, 2023-10-07:
Here we go:

```yaml
esphome:
  name: display-test
  friendly_name: Display Test

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "apiKey"

ota:
  password: "otaPwd"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Display-Test Fallback Hotspot"
    password: "superSecurePW"

captive_portal:

spi:
  - clk_pin: 15
    mosi_pin: 7
    miso_pin: 17
    id: test_disp

display:
  - platform: ili9xxx
    model: ILI9488
    cs_pin: 4
    dc_pin: 6
    reset_pin: 5
    spi_id: test_disp
    auto_clear_enabled: false
    lambda: |-
      int shift_x = (it.get_width()-310)/2;
      int shift_y = (it.get_height()-256)/2;
      for(auto i = 0; i<256; i++) {
        it.horizontal_line(shift_x+  0,i+shift_y,50, my_red.fade_to_white(i));
        it.horizontal_line(shift_x+ 50,i+shift_y,50, my_red.fade_to_black(i));

        it.horizontal_line(shift_x+105,i+shift_y,50, my_green.fade_to_white(i));
        it.horizontal_line(shift_x+155,i+shift_y,50, my_green.fade_to_black(i));

        it.horizontal_line(shift_x+210,i+shift_y,50, my_blue.fade_to_white(i));
        it.horizontal_line(shift_x+260,i+shift_y,50, my_blue.fade_to_black(i));
      }
      it.rectangle(shift_x+ 0, 0+shift_y, shift_x+ 310, 256+shift_y, my_yellow);

color:
  - id: my_blue
    blue: 100%
  - id: my_red
    red: 100%
  - id: my_green
[... 228 lines cut ...]
[VV][scheduler:226]: Running interval 'update' with interval=1000 last_execution=11205 (now=12211)
[V][ili9xxx:182]: Start display(xlow:85, ylow:32, xhigh:394, yhigh:287, width:310, heigth:256, start_pos:15445)
[W][component:204]: Component ili9xxx.display took a long time for an operation (0.46 s).
[W][component:205]: Components should block for at most 20-30ms.
[VV][scheduler:226]: Running interval 'update' with interval=1000 last_execution=12205 (now=13205)
[V][ili9xxx:182]: Start display(xlow:85, ylow:32, xhigh:394, yhigh:287, width:310, heigth:256, start_pos:15445)
[W][component:204]: Component ili9xxx.display took a long time for an operation (0.46 s).
[W][component:205]: Components should block for at most 20-30ms.
[VV][scheduler:226]: Running interval 'update' with interval=1000 last_execution=13205 (now=14205)
[V][ili9xxx:182]: Start display(xlow:85, ylow:32, xhigh:394, yhigh:287, width:310, heigth:256, start_pos:15445)
[W][component:204]: Component ili9xxx.display took a long time for an operation (0.46 s).
[W][component:205]: Components should block for at most 20-30ms.
[VV][scheduler:226]: Running interval 'update' with interval=1000 last_execution=14205 (now=15205)
[V][ili9xxx:182]: Start display(xlow:85, ylow:32, xhigh:394, yhigh:287, width:310, heigth:256, start_pos:15445)
[W][component:204]: Component ili9xxx.display took a long time for an operation (0.46 s).
[W][component:205]: Components should block for at most 20-30ms.
[VV][scheduler:226]: Running interval 'update' with interval=1000 last_execution=15205 (now=16212)
[V][ili9xxx:182]: Start display(xlow:85, ylow:32, xhigh:394, yhigh:287, width:310, heigth:256, start_pos:15445)
[W][component:204]: Component ili9xxx.display took a long time for an operation (0.46 s).
[W][component:205]: Components should block for at most 20-30ms.
[VV][scheduler:226]: Running interval 'update' with interval=1000 last_execution=16205 (now=17205)
[V][ili9xxx:182]: Start display(xlow:85, ylow:32, xhigh:394, yhigh:287, width:310, heigth:256, start_pos:15445)
[W][component:204]: Component ili9xxx.display took a long time for an operation (0.46 s).
[W][component:205]: Components should block for at most 20-30ms.
```

carslen, 2023-10-07:
[image]

nielsnl68, 2023-10-07:
That does not make sense to me at all. Did you try one of the other display types?
And can you give a link from where you brought this display ? 

Also are you using an esp32 with PSRAM on it?

nielsnl68, 2023-10-07:
Hi after some investigation i made a small PR: i would love you to test it out with the code below:

``` yaml
external_components:
  - source: github://pr#5494
    refresh: 10s
    components:
      - ili9xxx
```
And let us know of this will fix your issue with this PR.

carslen, 2023-10-09:
Sorry for not responding. I'll test your code today or latest tomorrow.

carslen, 2023-10-09:
Hm, I'm unable to install this config:

```
external_components:
  - source: github://pr#5494
    refresh: 10s
    components: 
      - ili9xxx
spi:
  - clk_pin: 15
    mosi_pin: 7
    miso_pin: 17
    id: test_disp

display:
  - platform: ili9xxx
    model: ILI9488_A
    cs_pin: 4
    dc_pin: 6
    reset_pin: 5
    spi_id: test_disp
    auto_clear_enabled: false
    lambda: |-
      int shift_x = (it.get_width()-310)/2;
      int shift_y = (it.get_height()-256)/2;
      for(auto i = 0; i<256; i++) {
        it.horizontal_line(shift_x+  0,i+shift_y,50, my_red.fade_to_white(i));
        it.horizontal_line(shift_x+ 50,i+shift_y,50, my_red.fade_to_black(i));

        it.horizontal_line(shift_x+105,i+shift_y,50, my_green.fade_to_white(i));
        it.horizontal_line(shift_x+155,i+shift_y,50, my_green.fade_to_black(i));

        it.horizontal_line(shift_x+210,i+shift_y,50, my_blue.fade_to_white(i));
        it.horizontal_line(shift_x+260,i+shift_y,50, my_blue.fade_to_black(i));
      }
      it.rectangle(shift_x+ 0, 0+shift_y, shift_x+ 310, 256+shift_y, my_yellow);
[...]
```
THis is the log output from trying to install:
```
INFO ESPHome 2023.9.3
INFO Reading configuration /config/esphome/psram-test.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5494/head
ERROR Unable to load component ili9xxx.display:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 165, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/data/external_components/631da664/esphome/components/ili9xxx/display.py", line 108, in <module>
    .extend(spi.spi_device_schema(False, "40MHz")),
TypeError: spi_device_schema() takes from 0 to 1 positional arguments but 2 were given
Failed config

display.ili9xxx: [source /config/esphome/psram-test.yaml:46]
  
  Platform not found: 'display.ili9xxx'.
  platform: ili9xxx
  model: ILI9488_A
  cs_pin: 4
  dc_pin: 6
  reset_pin: 5
  spi_id: test_disp
  auto_clear_enabled: False
  lambda: |-
    int shift_x = (it.get_width()-310)/2;
    int shift_y = (it.get_height()-256)/2;
    for(auto i = 0; i<256; i++) {
      it.horizontal_line(shift_x+  0,i+shift_y,50, my_red.fade_to_white(i));
      it.horizontal_line(shift_x+ 50,i+shift_y,50, my_red.fade_to_black(i));
    
      it.horizontal_line(shift_x+105,i+shift_y,50, my_green.fade_to_white(i));
      it.horizontal_line(shift_x+155,i+shift_y,50, my_green.fade_to_black(i));
    
      it.horizontal_line(shift_x+210,i+shift_y,50, my_blue.fade_to_white(i));
      it.horizontal_line(shift_x+260,i+shift_y,50, my_blue.fade_to_black(i));
    }
    it.rectangle(shift_x+ 0, 0+shift_y, shift_x+ 310, 256+shift_y, my_yellow);
```

carslen, 2023-10-09:
Due to the PSRAM thing, I'm using now a ESP32-S3-DevKitC-1 board.

closed 2023-10-12 by jesserockz

fsadak, 2023-10-26:
I have the same problem and I opened a discussion about it but I couldn't find any solution. I am using ESP32-S2 and it works fine with Arduino library. The screen I am using is ili9486 chip set.

carslen, 2023-10-27:
With recent ESP Update my issue was solved.

```yaml
spi:
  - clk_pin: 15
    mosi_pin: 7
    miso_pin: 17
    id: spi_display

display:
  - platform: ili9xxx
    id: 'mydisplay'
    model: ILI9488_A
    cs_pin: 4
    dc_pin: 6
    reset_pin: 5
    spi_id: spi_display
    auto_clear_enabled: True
    invert_display: true
    rotation: 270°
    pages: 
      [...]
```

fsadak, 2023-10-27:
Thanks for your answer but nothing changed... :(