esphome/issues #4950: CCT Bulb troubles.. temperature and brightness control shenanigans.

remlei, 2023-10-06:
### The problem

Basically its messed up

CCT bulb controls are buggy at best.

- brightness control is only possible when color temperature is set to cool (white) 
- brightness control is not possible when color temperature is set to warm (yellow)
- while in warm light, brightness control only adjust the temperature control from very yellow color to lightish yellow color
- when in warm color you basically cant turn off the bulb unless you set it to cool (white) temperature and turn it off

the pins are correct. I can control the bulb correctly using openbeken firmware. but not esphome.

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8

### What platform are you using?

ESP8266

### Board

generic-bk7231n-qfn32-tuya

### Example YAML snippet

```yaml
esphome:
  name: crabtek_a65_cct
  friendly_name: smartbulb_cct
  name_add_mac_suffix: false

bk72xx:
  board: generic-bk7231n-qfn32-tuya

logger:

web_server:
  port: 80
api:
ota:
captive_portal:
mdns:
tuya:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: ${name}AP
    password: "ESPNotHome"

button:
- platform: restart
  name: Restart

debug:
  update_interval: 30s

text_sensor:
- platform: debug
  reset_reason:
    name: Reset Reason
- platform: libretiny
  version:
    name: LibreTiny Version

sensor:
- platform: uptime
  name: Uptime

output:
  - platform: libretiny_pwm
    id: brightness_pwm
    pin: P7
  - platform: libretiny_pwm
    id: color_temp_pwm
    pin: P8

light:
  - platform: color_temperature
    name: "crabtek_a65_cct"
    color_temperature: color_temp_pwm
    brightness: brightness_pwm
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
```

agreenfield1, 2023-10-06:
I’ve been using the libretiny platform for a bit and I’m thinking there may be some issues with libretiny_pwm.  I have two different rgbww lights and the color selection works usually, but sometimes gets way off or wont set a chosen color at all.  For testing I set up the light as five independent lights in esphome and would sometimes get unexpected interactions between them.

patagonaa, 2023-12-19:
possibly related to https://github.com/libretiny-eu/libretiny/issues/200 (which is fixed in libretiny now), which causes PWM outputs to stop working when turning them off and on again

Until ESPHome comes with this new framework version, you can try replacing `bk72xx:` with
```yaml
bk72xx:
  framework:
    version: 0.0.0
    source: https://github.com/libretiny-eu/libretiny.git
```
in your device yaml to check if that is/was the issue

remlei, 2024-01-24:
thanks for the fix, yes using that specific framework did fix the issue on my side, no changes at all with the yaml file. I hope esphome pull the fixes soon.