esphome/issues #4951: restore possibility to edit content in .esphome folder

bkbartk, 2023-10-06:
### The problem

I am not able to edit content in the .esphome folder anymore
I know about this change https://esphome.io/changelog/2023.9.0.html#home-assistant-add-on
but I was trying to modify some code from an external component to see if the solution I had in mind worked.

I also use this sometimes to modify not merged PRs. 

### Which version of ESPHome has the issue?

2023.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.0

### What platform are you using?

ESP32

### Board

tuya light bulb; Sonoff ns panel

### Component causing the issue

esphome add on.

ssieb, 2023-10-07:
You should never modify things in there.  It is not a bug because it was never intended to work.  Any changes in there get overwritten when you compile.  If you need to test changes to an external component, then copy it, make the changes, and use your copy instead.

closed 2023-10-07 by ssieb

special-k-uk, 2023-10-31:
I can see your point for external components, but I'm struggling to find where to store the json for a custom board.

I've been following https://github.com/esphome/feature-requests/issues/1960#issuecomment-1540906617 , and I currently include the board definition here
```
  platformio_options:
    board_dir: boards
    #include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
    board: lilygo-t5-47-plus
```

with the 'boards' directory being located here `/config/esphome/.esphome/build/[your-esphome-device]/boards`

Now the .epshome folder has been removed, do you know where I should recreate the /boards folder?