esphome/issues #4952: Unable to find current revision in submodule path 'components/esp-adf-libs'

genebean, 2023-10-06:
### The problem

I am trying to update the firmware on an m5 stack echo and there seems to be an issue with the git submodules.

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.1

### What platform are you using?

ESP8266

### Board

m5 stack echo

### Component causing the issue

components/esp-adf-libs

### Example YAML snippet

```yaml
substitutions:
  name: m5stack-atom-echo-888210
  friendly_name: Bedroom Atom
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: foo

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.9.3
INFO Reading configuration /config/esphome/m5stack-atom-echo-888210.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5229/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 454, in command_run
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 190, in write_cpp
    return write_cpp_file()
  File "/esphome/esphome/__main__.py", line 208, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 342, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 295, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 592, in copy_files
    repo_dir, _ = git.clone_or_update(
  File "/esphome/esphome/git.py", line 114, in clone_or_update
    run_git_command(
  File "/esphome/esphome/git.py", line 32, in run_git_command
    raise cv.Invalid(err_str)
voluptuous.error.Invalid: fatal: Needed a single revision
Unable to find current revision in submodule path 'components/esp-adf-libs'
```

toby-griffiths, 2023-10-12:
@genebean I've just hit this, but retried a few times and it worked eventually.

genebean, 2023-10-12:
> @genebean I've just hit this, but retried a few times and it worked eventually.

I've retried several times over a few days with no luck... I'll try again tonight though.

toby-griffiths, 2023-10-13:
@genebean  I've realised I didn't retry in the same browser.  I was originally doing it in Firefox, but it worked in Chrome.

genebean, 2023-10-13:
@toby-griffiths Doubt it's browser related as it also fails in the mobile app, and the failure is server-side. I did retry, no luck.

lein1013, 2023-10-15:
I've the same issue on the 2023.10.0b2 ESPHome version

OzGav, 2023-10-19:
Restarting the ESPHome addon seemed to fix this for me. Im using ESPHome 2023.10.1

bkemmler, 2023-10-19:
After several retries i still get the same error message 

Unable to checkout '0c1eecaac259af95f76723b61d4e16cef5904fc7' in submodule path 'components/esp-sr'

This error is shown after adding this lines:

packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main

GrumpyMeow, 2023-10-24:
Same here:
```
INFO ESPHome 2023.10.3
INFO Reading configuration /config/esphome/echo-1.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 454, in command_run
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 190, in write_cpp
    return write_cpp_file()
  File "/esphome/esphome/__main__.py", line 208, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 342, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 295, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 593, in copy_files
    repo_dir, _ = git.clone_or_update(
  File "/esphome/esphome/git.py", line 114, in clone_or_update
    run_git_command(
  File "/esphome/esphome/git.py", line 32, in run_git_command
    raise cv.Invalid(err_str)
voluptuous.error.Invalid: fatal: Needed a single revision
Unable to find current revision in submodule path 'components/esp-adf-libs'
```
I did try to restart the ESPHome addon, which resulted in a "hanging process".
```
INFO ESPHome 2023.10.3
INFO Reading configuration /config/esphome/echo-1.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
```
No output comes after this..

genebean, 2023-11-07:
I am still having this issue on the latest build of Home Assistant and its corresponding ESPHome version. Restarts and the like do not seem to help.

phycoweirdo, 2023-11-10:
This was not working for me in edge, but _**is**_ working for me in chrome. specifically Version 119.0.6045.124 (Official Build) (64-bit)

genebean, 2023-11-10:
> This was not working for me in edge(Version 119.0.2151.58 (Official build) (64-bit)), but _**is**_ working for me in chrome. specifically Version 119.0.6045.124 (Official Build) (64-bit)

I feel quite certain this is luck - the issue is not browser-side, it is container side.

genebean, 2023-11-28:
Just noting that ESPHome 2023.11.6 still throws the same error.

jesserockz, 2023-11-28:
> I feel quite certain this is luck - the issue is not browser-side, it is container side.

Yes it has nothing to do with the browser used.

Have you tried clean building the file?

Also you could try uninstalling and reinstalling the ESPHome addon which will fully clear all caches.

paulvt, 2023-11-28:
I have ESPHome installed via Docker and even moving way the entire `.esphome` directory under `config` did not help.
The error did change from complaining about `esp-sr` to `esp-adf-libs`.

So, I went to `config/.esphome//idf_components/b0ad5728/.git/modules/components/` and removed the `esp-rs` and `esp-adf-libs` repositories, which where in weird states. Now it works on retry.

genebean, 2023-11-28:
> Have you tried clean building the file?

I have tried this many times
 
> Also you could try uninstalling and reinstalling the ESPHome addon which will fully clear all caches.

Just tried this, no help.

```log
INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/m5stack-atom-echo-888210.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1040, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1027, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 457, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 193, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 211, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 342, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 295, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 673, in copy_files
    repo_dir, _ = git.clone_or_update(
                  ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 114, in clone_or_update
    run_git_command(
  File "/esphome/esphome/git.py", line 31, in run_git_command
    raise cv.Invalid(lines[-1][len("fatal: ") :])
voluptuous.error.Invalid: Unable to find current revision in submodule path 'components/esp-adf-libs'
```

the-mentor, 2023-11-29:
I'm also seeing this issue running esphome 2023.11.6 in docker.

locomot1f, 2023-12-19:
I am also still getting this error message with the M5Stack Atom Echo device
I've been trying for a couple of days. I'll keep trying.

locomot1f, 2023-12-20:
> > I feel quite certain this is luck - the issue is not browser-side, it is container side.
> 
> Yes it has nothing to do with the browser used.
> 
> Have you tried clean building the file?
> 
> Also you could try uninstalling and reinstalling the ESPHome addon which will fully clear all caches.

Uninstalling and reinstalling the ESPHome Addon seems to have worked for me. thanks @jesserockz

DangerD1024, 2023-12-22:
Same here, HA supervised installed in docker Orange Pi PC2...

DangerD1024, 2023-12-23:
Same with DEV...

DangerD1024, 2023-12-24:
```
INFO ESPHome 2024.1.0-dev
INFO Reading configuration /config/esphome/m5stack-atom-echo-b8efdc.yaml...
INFO Updating https://github.com/esphome/firmware.git@main
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
```
After this whole system stucks, i even can't connect to ssh...

NathanaelA, 2023-12-29:
In my case hitting the "retry" several times would get if farther -- so something was causing "git" it appears to timeout or fail.   Once if finally got to the point where it had already downloaded all the git source code it appears to have worked to build the image.

devel-q, 2024-01-21:
Same here with 2023.12.8. I can build it once successfully after removing the folders but the next time I run it, it fails again.
Dropping every time the folders is nasty, so any hint on getting rid of the error completely is welcome!

monsieurhannes, 2024-05-09:
I'm having the same issue with 2024.5.2 (core), 2024.05.1 (supervisor), 12.3 (OS).
My ESPHome-version is 2024.4.2

Which folder do I need to rename when ESPhome is on a supervised system? How can I find it? 
I've tried multiple browsers (even though I believe the ones saying that it's not a browser issue) and clicked "retry" so many times... There needs to be another solution?! It doesn't seem to affect all installations either, right?

atomicpapa, 2024-07-08:
Still having this issue with ESPHome 2024.6.6 when trying to update my Box 3.  Retryed more times than I can count on all 12 fingers and all 12 toes

SGNegio, 2024-07-21:
A very similar problem also appears in the ESPHome 2024.7.0 version. Below is my attempt to update the M5Stack ATOM Echo.
```
INFO ESPHome 2024.7.0
INFO Reading configuration `/config/esphome/m5stack-atom-echo-31cad8.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1079, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1066, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 484, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 194, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 344, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 297, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 684, in copy_files
    repo_dir, _ = git.clone_or_update(
                  ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 111, in clone_or_update
    run_git_command(
  File "/esphome/esphome/git.py", line 31, in run_git_command
    raise cv.Invalid(lines[-1][len("fatal: ") :])
voluptuous.error.Invalid: Unable to find current revision in submodule path 'components/esp-adf-libs'
```
I tried the tips mentioned above, but to no avail.

atomicpapa, 2024-07-31:
Still happening with ESPhome 2024.7.2

monsieurhannes, 2024-07-31:
> [Tracking image][https://github.com/notifications/beacon/AE5R3ME7P6KP53ICB6FPQYLZPBZ25A5CNFSM6AAAAAA5WNLONGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUGWBXRW.gif]

Same here. My Atom Echo sit unused on my desk...

tclemos, 2024-08-05:
it worked fine yesterday, today I'm facing this issue.

tdy91, 2024-08-11:
I had the same problem with Esphome version 2024.7.3

I solved it using a developper-level procedure which I posted on Home Assistant community [M5Stack ATOM Echo adaptation problem](https://community.home-assistant.io/t/m5stack-atom-echo-adaptation-problem/751302/6).

Hope this can help a few people !

BrianMoos, 2024-10-25:
I also had this issue. It was actually because my server running esphome was out of disk space so it couldn't download from github. doh!

dr1rrb, 2024-11-14:
On my side, it failed once (also due to disk being full), then I had to manually delete the content of the `.esphome/idf_components` folder to fix the issue (a "clean build" was not enough).