esphome/issues #4954: Wrong power calculation HLW8012

KCarlsen, 2023-10-07:
### The problem

The power calculation of HLW8012 are wrong.
In the log notice that Voltage * current is not equal the returned power.

### Which version of ESPHome has the issue?

2023.6.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
sensor:
  - platform: hlw8012
    update_interval: 5s
    change_mode_every: 2
    sel_pin:
      number: GPIO12
      inverted: True
    cf_pin: GPIO4
    cf1_pin: GPIO5
    current:
      id: current
      name: ${friendly_name} Amperage
      device_class: current
      unit_of_measurement: A
    voltage:
      id: voltage
      name: ${friendly_name} Voltage
      device_class: voltage
      unit_of_measurement: V
    energy:
      id: energy
      name: ${friendly_name} Energy
      device_class: energy
      unit_of_measurement: Wh
    power:
      id: power
      name: ${friendly_name} Wattage
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
  - platform: total_daily_energy
    name: ${friendly_name} Total Daily Energy
    device_class: energy
    power_id: power
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
  - platform: template
    name: ${friendly_name} Apparent power
    lambda: return id(voltage).state * id(current).state;
    accuracy_decimals: 2
    unit_of_measurement: VA
    device_class: energy
  - platform: template
    name: ${friendly_name} Power factor
    lambda: return id(power).state / (id(voltage).state * id(current).state);
    accuracy_decimals: 2
    unit_of_measurement: ""
    device_class: power_factor
```

### Anything in the logs that might be useful for us?

```txt
[09:40:54][D][hlw8012:082]: Got power=1044.8W, voltage=228.2V
[09:40:54][D][sensor:093]: 'ladeteknik Voltage': Sending state 228.16623 V with 1 decimals of accuracy
[09:40:54][D][sensor:093]: 'ladeteknik Wattage': Sending state 1044.84314 W with 1 decimals of accuracy
[09:40:54][D][sensor:093]: 'ladeteknik Total Daily Energy': Sending state 0.88393 kWh with 3 decimals of accuracy
[09:40:54][D][sensor:093]: 'ladeteknik Energy': Sending state 118.55608 Wh with 1 decimals of accuracy
[09:40:59][D][sensor:093]: 'ladeteknik Wattage': Sending state 1042.60461 W with 1 decimals of accuracy
[09:40:59][D][sensor:093]: 'ladeteknik Total Daily Energy': Sending state 0.88537 kWh with 3 decimals of accuracy
[09:40:59][D][sensor:093]: 'ladeteknik Energy': Sending state 120.00415 Wh with 1 decimals of accuracy
[09:41:04][D][hlw8012:076]: Got power=1043.2W, current=2.6A
```

RoboMagus, 2023-10-09:
I never noticed this as I've [calibrated](https://frenck.dev/calibrating-an-esphome-flashed-power-plug/) all my power measuring plugs, but now that you mention it I also see a deviation from the raw power value to what I would expect.

A quick look at the implementation of the HLW8012 component vs other libraries relating to this chip does not show any inconsitencies though. So I'm not quite sure what causes this difference.

KimCarlsen, 2023-10-09:
ya, mine is also calibrated, down to one digit. 
And I've tested with a 600W infrared heater, so resistive load and the raw voltage and current is pretty close to that, but the reported power is like 60% off

RoboMagus, 2023-10-09:
With separate power calibration that should not be an issue. The reported power value from this sensor should then be corrected by the calibration.

KimCarlsen, 2023-10-09:
Well. it doesn't. 
I did not calibrate as Frenck, but set the current_resistor and voltage_divider, so i get accurate voltage and current readings. Verified with two multimeters. The current and voltage are correctly reported in log,

RoboMagus, 2023-10-09:
Well, thats why I mentioned `separate power calibration`, as that would nullify any computation error produced by the power sensor before it's published. I would recommend doing the same regardless of wether there's a bug to be found in this component anyway.

I hope to be able to further investigate this later this week, but looking at the computations in this component there are no anomalies when compared to other libraries around the HLW8012 chip and no obvious bugs that might cause these differences.

widewing, 2023-10-17:
I'm also having the same issue, power is always 60% higher than current*voltage. I use this KMC 4-outlet plug btw: https://a.co/d/1x194WD

KimCarlsen, 2023-10-23:
@RoboMagus did you had a chance to look at the code? 
I would like to avoid the power calibration. alternative i can create a custom sensor, that is just the product of voltage and current, but its seems like a wrong path to take.

RoboMagus, 2023-10-26:
@KimCarlsen, sorry no; life is getting in the way at the moment. Perhaps someone else is up for the challenge, as for the time being I don't have the opportunity to tinker with this.

bruceyv83, 2023-12-22:
Same issue here. I only noticed it after taking the time to calibrate the voltage and current.  Using the Gosund UP111