esphome/issues #4955: ESP32 S3 Touch sensor

silverchris, 2023-10-07:
### The problem

If the touchsensor value overflows it hangs the touch sensor hardware, resulting in no new readings.
See https://github.com/espressif/esp-idf/issues/12300

### Which version of ESPHome has the issue?

2023.10.0-dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Component causing the issue

esp32_touch

### Anything in the logs that might be useful for us?

```txt
Reported value from `setup_mode: true` will be stuck at a fixed value
```

### Additional information

decreasing `measurement_time` reduces the issue as it stops the sensor from overflowing and hanging

kfomichev, 2025-01-08:
Same for me
ESPHome 2024.10.3
ESP32-S3 Zero board
With setup_mode: true sensor reading in log gets stuck at 3207653 and never updates
Specifying measurement_time: 1ms makes values update. Btw, I'm not sure if the issue is because of overflow. When holding the gpio line firmly with my fingers I can see values above 4000000, but nothing hangs.

OniriCorpe, 2025-02-20:
i also had to put `measurement_duration: 1ms` to get variable values, otherwise the values are not updating

using a `lolin_s2_mini`